Koenkk/zigbee2mqtt #4332: Support for Philips Hue Play Lightbar

zenety, 2020-09-11:
## Information about the device + link
I assume it's the exact same device as the already supported lightbar. https://www.zigbee2mqtt.io/devices/915005733701.html
However this one has a different model number.

Name: Philips Hue Play Lightbar
Link: philips-hue.com/en-us/p/hue-white-and-color-ambiance-play-light-bar-double-pack/7820230U7
Model ID: 440400982841
Vendor: Philips
Zigbee ID: 0x00178801085b35cf

## data/database.db entry of the device
{"id":7,"type":"Router","ieeeAddr":"0x00178801085b35cf","nwkAddr":32087,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"440400982841","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,768,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"440400982841","manufacturerName":"Philips","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20190529","swBuildId":"1.56.7_r29434"}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20190529","swBuildId":"1.56.7_r29434","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1599817615280}

sjorge, 2020-09-11:
You can try with updating
```
   {
        zigbeeModel: ['LCT024'],
        model: '915005733701',
        vendor: 'Philips',
        description: 'Hue White and color ambiance Play Lightbar',
        meta: {turnsOffAtBrightness1: true},
        extend: hue.light_onoff_brightness_colortemp_colorxy,
        ota: ota.zigbeeOTA,
    },
```
to 

```
   {
        zigbeeModel: ['LCT024', '440400982841'],
        model: '915005733701', 
        vendor: 'Philips',
        description: 'Hue White and color ambiance Play Lightbar',
        meta: {turnsOffAtBrightness1: true},
        extend: hue.light_onoff_brightness_colortemp_colorxy,
        ota: ota.zigbeeOTA,
    },
```

in devices.js

zenety, 2020-09-11:
That works! On/off, brightness and colours work. Thanks for the quick response.

sjorge, 2020-09-11:
Good, I’ll do a PR so support should end up in the next release.

sjorge, 2020-09-11:
PR opened as koenkk/zigbee-herdsman-converters#1547

Koenkk, 2020-09-11:
Thanks for helping @sjorge 

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-09-11 by Koenkk