Koenkk/zigbee2mqtt #4333: HomeAssistant discovery and Xiaomi Aqara sensor causes entities with duplicate attributes

voltagex, 2020-09-11:
## What happened
I'm not sure if this is "as designed", but why do Z2M devices end up with duplicate attributes on every entity?

e.g. 

[image]

[image]

The battery entity also has temperature data (the same happens for each of the entities that are automatically created)

This causes issues if you're automatically pushing data to Influx/Grafana and the like as you end up with multiple copies of the same data (e.g. 0xfff_battery.temperature, 0xfff_temperature.temperature)

## What did you expect to happen
Either one entity with all attributes, or each entity only having the attributes that are related to it (battery only has battery level, temperature only has degrees C)

## How to reproduce it (minimal and precise)
* Follow the bare-metal instructions, enable home assistant integration and pair a Xiaomi Aqara sensor
## Debug info
Zigbee2MQTT version: 1.14.4 (commit #8129093)

Adapter hardware: CC2531
Adapter firmware version: zStack12

voltagex, 2020-09-11:
A quick fix for this is to remove https://github.com/Koenkk/zigbee2mqtt/blob/812909363b611b6eb5581f23dbae68e0cc5d8a3e/lib/extension/homeassistant.js#L2172 which makes the sensor in Home Assistant look like: 
[image] - I think I'd rather put this behind a config setting with the default being the current behaviour.

Koenkk, 2020-09-11:
You should be able to override this for all devices by setting the following in the Zigbee2mqtt `configuration.yaml`:

```yaml
device_options:
  homeassistant:
    json_attributes_topic: null
```

voltagex, 2020-09-12:
Thanks @Koenkk.

This should solve the issue, @82amp

closed 2020-09-12 by Koenkk