Koenkk/zigbee2mqtt #4334: SmartThings Multipurpose sensor (2018 model) IM6001-MPP01 - Tilt not working

MRobi1, 2020-09-11:
I have 3 of these paired, 1 on a man door with magnet and 2 that were used as tilt sensors on a garage door.
Just starting off with Zigbee2MQTT. Followed the pairing instructions with activating the sensors for 20s while pairing. They all show open/closed based on the magnet state, but none of them report tilt. Battery and temp are working fine. In Zigbee2MQTT Assistant all I see is battery, contact, link quality and temp. Also says manufacturer is Samjin. From reading a few other closed issues it looks like the manufacturer should be listed as Samjin-Smartthings or just Smartthings. So I think it's pairing as the wrong device.

Best way to go about fixing this so I can get tilt?

Koenkk, 2020-09-11:
Tilt is indeed not supported, can you provide the Zigbee2MQTT debug log when tilting the device?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

MRobi1, 2020-09-11:
Turned on debug logging and this is the except from raising the garage door. The only mention of the device in the log.

Zigbee2MQTT:info  2020-09-11 19:02:00: MQTT publish: topic 'zigbee2mqtt/all_thermostats_group', payload '{"brightness":0,"state":"OFF","system_mode":"off"}'
2020-09-11T22:02:00.048Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2020-09-11T22:02:00.097Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 89
2020-09-11T22:02:00.128Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 89 srcAddr: 0x794f destAddr: 0x0 profile id: 0x104 cluster id: 0x402
2020-09-11T22:02:00.128Z zigbee-herdsman:deconz:frameParser response payload: 8,6,10,0,0,41,119,6
2020-09-11T22:02:00.129Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":6,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":41,"attrData":1655}]},"address":31055,"endpoint":1,"linkquality":106,"groupID":null}'
Zigbee2MQTT:debug 2020-09-11 19:02:00: Received Zigbee message from 'hot_tub_door_sensor', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":1655}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2020-09-11 19:02:00: MQTT publish: topic 'zigbee2mqtt/hot_tub_door_sensor', payload '{"battery":86.5,"battery_low":false,"contact":false,"linkquality":106,"tamper":false,"temperature":16.55}'
2020-09-11T22:02:00.153Z zigbee-herdsman:controller:endpoint DefaultResponse 0x286d97000110db9f/1 1026(10, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
2020-09-11T22:02:00.153Z zigbee-herdsman:deconz:adapter zclFrame.payload:
2020-09-11T22:02:00.153Z zigbee-herdsman:deconz:adapter { cmdId: 10, statusCode: 0 }
2020-09-11T22:02:00.154Z zigbee-herdsman:deconz:adapter no response expected
2020-09-11T22:02:00.157Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x794f EP:1 SeqNr. 90 request id: 8
2020-09-11T22:02:00.160Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 8 status: 0
2020-09-11T22:02:00.166Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2020-09-11T22:02:00.207Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 91
2020-09-11T22:02:00.209Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x794f request id: 8 confirm status: 0
2020-09-11T22:02:00.210Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send
2020-09-11T22:02:04.945Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 92
2020-09-11T22:02:04.950Z zigbee-herdsman:deconz:frameParser device state: 10100010
2020-09-11T22:02:13.770Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2020-09-11T22:02:13.776Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2020-09-11T22:02:13.809Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 93
2020-09-11T22:02:13.811Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 93 srcAddr: 0xab21 destAddr: 0x0 profile id: 0x104 cluster id: 0x201
2020-09-11T22:02:13.812Z zigbee-herdsman:deconz:frameParser response payload: 24,91,10,8,0,32,0
2020-09-11T22:02:13.821Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":91,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":8,"dataType":32,"attrData":0}]},"address":43809,"endpoint":1,"linkquality":106,"groupID":null}'
Zigbee2MQTT:debug 2020-09-11 19:02:13: Received Zigbee message from 'thermostat_master_bedroom', type 'attributeReport', cluster 'hvacThermostat', data '{"pIHeatingDemand":0}' from endpoint 1 with groupID null

Koenkk, 2020-09-12:
@MRobi1 do you have a gateway were this sensor provides tilt values? If so we should sniff the traffic of that to figure out how the sensor sends the tilt values: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

MRobi1, 2020-09-12:
@Koenkk  by gateway I assume another hub that supports tilt for this device would be ok? I did have them hooked to smartthings and then hubitat and tilt worked there so I could move one back for testing fairly easily. I still have both hubs.

I checked out the instructions for sniffing traffic, I've only got a conbee II stick which isn't listed there. From looking at the instructions for the other 2 sticks, it looks like they need to be flashed so I'd probably lose my whole setup?

Koenkk, 2020-09-13:
@MRobi1 you need a separate CC2531 to do this indeed.

MRobi1, 2020-09-15:
@Koenkk damn, I'm not setup for that at all.

closed 2020-10-23 by github-actions[bot]

ssmall, 2021-02-28:
I am encountering the same issue. I have been using my IM6001-MPP01 sensors with the ZHA integration in HomeAssistant and I am trying to migrate to Zigbee2MQTT.

Getting them working in HomeAssistant was a bit hacky and required reading the cluster attributes directly to infer the position of the sensor based on the accelerometer. [Here](https://github.com/zigpy/zha-device-handlers/blob/0e3cb20aecdcf48c36d87572ef820abf4df2c9b9/zhaquirks/smartthings/__init__.py#L11) is the ZHA source that defines the cluster attributes used for this sensor.

If the accelerometer attributes can be exposed via Zigbee2MQTT, that would allow using the IM6001-MPP01 as a tilt sensor.

RobinCutshaw, 2021-03-27:
Looking for tilt data as well.  Debug shows:

> debug 2021-03-27 11:56:16: Received Zigbee message from 'Samjin multi Testbench', type 'attributeReport', cluster 'manuSpecificSamsungAccelerometer', data '{"acceleration":1}' from endpoint 1 with groupID null
info  2021-03-27 11:56:16: MQTT publish: topic 'zigbee2mqtt/Samjin multi Testbench', payload '{"battery":79.5,"battery_low":false,"contact":true,"linkquality":249,"moving":true,"tamper":false,"temperature":24.47}'
debug 2021-03-27 11:56:25: Received Zigbee message from 'Samjin multi Testbench', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":33}' from endpoint 1 with groupID null
info  2021-03-27 11:56:25: MQTT publish: topic 'zigbee2mqtt/Samjin multi Testbench', payload '{"battery":79.5,"battery_low":false,"contact":false,"linkquality":249,"moving":true,"tamper":false,"temperature":24.47}'
debug 2021-03-27 11:56:30: Received Zigbee message from 'Samjin multi Testbench', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":32}' from endpoint 1 with groupID null
info  2021-03-27 11:56:30: MQTT publish: topic 'zigbee2mqtt/Samjin multi Testbench', payload '{"battery":79.5,"battery_low":false,"contact":true,"linkquality":249,"moving":true,"tamper":false,"temperature":24.47}'
debug 2021-03-27 11:56:46: Received Zigbee message from 'Samjin multi Testbench', type 'attributeReport', cluster 'manuSpecificSamsungAccelerometer', data '{"acceleration":0}' from endpoint 1 with groupID null
info  2021-03-27 11:56:46: MQTT publish: topic 'zigbee2mqtt/Samjin multi Testbench', payload '{"battery":79.5,"battery_low":false,"contact":true,"linkquality":249,"moving":false,"tamper":false,"temperature":24.47}'