Koenkk/zigbee2mqtt #4336: MCCGQ11LM: how to fix losing connection (detailed post, tried a lot of things already)

prankousky, 2020-09-12:
Hi everybody,

```
{"commit":"edc2551","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20200417,"transportrev":2},"type":"zStack3x0"},"log_level":"info","network":{"channel":15,"extendedPanID":"yes","panID":yes},"permit_join":true,"version":"1.14.4"}

```

Adapter `Texas Instruments LAUNCHXL-CC1352P-2`

I have a rather large zigbee2mqtt network. Some devices seem to randomly (?) disconnect from the network. When this happens, it helps to simply re-pair them, then they work as expected. Example

[image]

This is a `Xiaomi MCCGQ11LM`. I just discovered [this part](https://www.zigbee2mqtt.io/devices/MCCGQ11LM.html#troubleshooting-device-stops-sending-messagesdisconnects-from-network) of the product page. So considering that the linkquality is only `12`, this might already be the issue. I have exchanged the battery without success. Also, the coordinator is pretty much **right below** this sensor one floor down. 

This is the route this device takes
[image]

So `coordinator` => ` Innr RB 185 C` =>  `Xiaomi MCCGQ11LM` (Zwischenzimmer/Tuer). Therer is another sensor of the same brand/model on that room's window (about 3 meters away) that has a linkquality of `69` and, as far as I can tell, does not lose connection. The device in between the coordinator and the disconnecting sensor is **not** of the following `Centralite, General Electric, Iris, Ledvance, OSRAM, Sylvania, SmartThings, Securifi` as mentioned on that page.

# Ideas

1. Should I install a repeater? 

1.1. I have countless `Texas Instruments CC2530` from my beginning days using `zigbee2mqtt`. *Last time* I tried using those as repeaters, though, it seemed like they made connectivity worse, not better.

1.2. I also have an unused `Texas Instruments LAUNCHXL-CC26X2R1`. Could this be used as a repeater for the first floor?

2. Different network layout?

2.1. There is a `Philips 7099860PH` in the very same room these Xiaomi sensor are in; afaik it should be able to act as a router as well, but the device(s) is/are not connected to it. Would it help if I removed the `Innr RB 185 C` from the room next door, "forcing" the Xiaomi sensors to connect to a different router? While this would be an inconvenience (as I kinda need that light in the room), I'd do it if it would fix this issue. I have many Xiaomi door / window sensors that work, but that one is one of the three most important ones in the house, and while the others work almost all of the time, the important one keeps disconnecting.

3. Xiaomi router?

3.1. I considered buying the Xiaomi router, connecting it to my `Home Assistant`, and using it to work with the devices that cause issues. However, (a) I don't know if that would fix the issue, and once I bought the router, I am stuck with it and (b) I really like `zigbee2mqtt`; if possible, I'd keep working with it as my one and only zigbee solution.

# From that forum post

[Link](https://community.hubitat.com/t/xiaomi-aqara-devices-pairing-keeping-them-connected/623)

* `Pair your Xiaomi / Device in its intended final location, not next to the hub.`

I might actually have done this when initially pairing the sensor (don't remember). **But** I have removed the battery and re-paired the device countless time since then. 

* `Interference from another nearby Zigbee hub and/or 2.4GHz WiFi router`

There is, indeed, a `UniFi AC AP Pro` on the ceiling of that room. However, shouldn't it also affect other zigbee devices, if it was causing the issue? All other sensors (in the room in question as well as the rooms connected) usually work fine. There are 6 more door / window sensors *right* next to the WiFi AP (as in "not further away than this one sensor that keeps disconnecting") that work without any noticeable issue.

* `Zigbee repeaters reported NOT to work with Xiaomi devices`

I do not own **any** of the devices stated as not working with Xiaomi devices in that forum post from the link above.

* `"Sandbox" your Xiaomi / Aqara devices on a different hub.`

I considered this, but then I doubt I'd reach the range I want. I have Xiaomi devices on the ground floor, the first floor, even in the garden and garage. Some of those would not have a decent signal if it weren't for other devices that act as routers.

* `Xiaomi Gateways have been reported to operate on ZigBee channel 25 24, channel 20 11 and channel 11 (here 17 and here 8), so the assumption is these are "safe" channels for Xiaomi / Aqara devices. Also, a number of Hubitat users have reported success with their Hubitat Hub's ZigBee radio set to channel 13.`

My channel is currently `15`. Again: all other Xiaomi devices work. **This** device works, but it frequently loses connection (okay, you might not call this "works", but it's not like it is incapable of operating on that channel all together).

# What do you suggest I do / try? 

I have had different issues before and people suggested sniffing my traffic. Unfortunately, I was not able to set this up, so hopefully there is a different way I can try.

Thank you for your ideas :)

Koenkk, 2020-09-12:
> I have had different issues before and people suggested sniffing my traffic.

Sniffing traffic is the best (and maybe only) way to get a good understanding what and why things are happening. Where do you get stuck in: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html ?

If you could provide a sniff from a point where the sensor works until it stopped working we should be able to identify the cause of this issue (and help us gather a better understanding why some xiaomis disconnect)

prankousky, 2020-09-12:
Thank you, Koen. 

I read through the link and have to admit that I did a *completely* different approach last time. Either there was a different method on your website, or I looked in the wrong place. I have now flashed a `CC2531` with the firmware and tried a capture. Attached is a `.pcap` file (as `.zip`, because github wouldn't let me upload it in the original format). Is the data correct?

[pcap.zip](https://github.com/Koenkk/zigbee2mqtt/files/5212647/pcap.zip)

I had the capture running for only a few seconds. As I don't know when the sensor will disconnect the next time, I'll probably be ending up with a huge file, so it'd be good to know if the data itself was correct before proceeding. 

I piped the output to a file directly, as piping to wireshark created permission issues.

Koenkk, 2020-09-12:
Sniff looks good, to decrypt I also need your network key, you can send it to me on telegram (@koenkk). 

When providing the final sniff, also post the networkAddress and ieeeAddress of the problematic device, you can find it in the `data/database.db`/

closed 2020-10-20 by github-actions[bot]

Mutagennn, 2021-04-10:
Just thanks to @prankousky ! I was turning around this problem for days to pair aqara MCCGQ11LM. That was creepy! just change the channel ans works fine! big thx!

ricargr, 2022-01-13:
I know this problems is closed, but i dont know where to ask:
Is the MCCGQ14LM model afected by this, it is not clear for me and i want to be sure before i buy them.

Thanks