Koenkk/zigbee2mqtt #4338: Sonoff SNZB-02 gets connected as SONOFF SNZB-01

nvivekananth, 2020-09-12:
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

Koenkk, 2020-09-12:
can you provide the data/databse.db entry of this device?

nvivekananth, 2020-09-12:
Here is the database.db entry:

Before my temporary fix:
{"id":2,"type":"EndDevice","ieeeAddr":"0x00124b001f876a56","nwkAddr":46486,"manufId":0,"manufName":"eWeLink","powerSource":"Battery","modelId":"WB01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,1026,1029,1],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"WB01","manufacturerName":"eWeLink","powerSource":3,"zclVersion":1,"appVersion":4,"hwVersion":1,"dateCode":"20200718"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001caa6bd5","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b001caa6bd5","endpointID":1}]}},"appVersion":4,"hwVersion":1,"dateCode":"20200718","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1599915376672}

after temporary fix
{"id":2,"type":"EndDevice","ieeeAddr":"0x00124b001f876a56","nwkAddr":9188,"manufId":0,"manufName":"eWeLink","powerSource":"Battery","modelId":"WB01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,1026,1029,1],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"WB01","manufacturerName":"eWeLink","powerSource":3,"zclVersion":1,"appVersion":4,"hwVersion":1,"dateCode":"20200718"}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2940}},"msRelativeHumidity":{"attributes":{"measuredValue":6038}},"genPowerCfg":{"attributes":{"batteryVoltage":29,"batteryPercentageRemaining":140}}},"binds":[{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b001caa6bd5","endpointID":1},{"cluster":1029,"type":"endpoint","deviceIeeeAddress":"0x00124b001caa6bd5","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b001caa6bd5","endpointID":1}]}},"appVersion":4,"hwVersion":1,"dateCode":"20200718","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1599900938584}

Also, for this to work, I did change from 'TH01' to 'WB01' in devices.js
The SNZB-02 device come in as WB01 for some reason.

Koenkk, 2020-09-12:
Should be fixed now.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

nvivekananth, 2020-09-12:
Thanks. It is fixed.

closed 2020-09-12 by nvivekananth

koenhendriks, 2020-11-29:
I'm currently having this exact same issue, but with the SNZB-03 getting connected as SNZB-02!

here is my pair log:
```

zigbee2mqtt    | Zigbee2MQTT:info  2020-11-29 13:03:41: Zigbee: allowing new devices to join.
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-29 13:03:41: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"6b32f30","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"debug","network":{"channel":12,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.16.1"}'
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-29 13:03:56: Received Zigbee message from '0x00124b002247b7b0', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-29 13:03:56: No converter available for 'SNZB-02' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"extendedstatus":0,"zonestatus":1}'
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-29 13:03:56: Received Zigbee message from '0x00124b002247b7b0', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-29 13:03:56: No converter available for 'SNZB-02' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"extendedstatus":0,"zonestatus":1}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-29 13:03:56: Configuring '0x00124b002247b7b0'
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-29 13:03:56: Received Zigbee message from '0x00124b002247b7b0', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-29 13:03:56: No converter available for 'SNZB-02' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"extendedstatus":0,"zonestatus":1}'
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-29 13:03:56: Received Zigbee message from '0x00124b002247b7b0', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-29 13:03:56: No converter available for 'SNZB-02' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"extendedstatus":0,"zonestatus":1}'
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-29 13:03:56: Received Zigbee message from '0x00124b002247b7b0', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-29 13:03:56: No converter available for 'SNZB-02' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"extendedstatus":0,"zonestatus":1}'
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-29 13:03:56: Received Zigbee message from '0x00124b002247b7b0', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-29 13:03:56: No converter available for 'SNZB-02' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"extendedstatus":0,"zonestatus":1}'
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-29 13:03:56: Device '0x00124b002247b7b0' announced itself
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-29 13:03:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00124b002247b7b0"},"type":"device_announced"}'
zigbee2mqtt    | Zigbee2MQTT:error 2020-11-29 13:04:00: Failed to configure '0x00124b002247b7b0', attempt 2 (Error: ConfigureReporting 0x00124b002247b7b0/1 msTemperatureMeasurement([{"attribute":"measuredValue","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":100}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
zigbee2mqtt    |     at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:173:23)
zigbee2mqtt    |     at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:386:26)
zigbee2mqtt    |     at Generator.next (<anonymous>)
zigbee2mqtt    |     at fulfilled (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58))
```