Koenkk/zigbee2mqtt #4340: Offline sensor appears online in home assistant

dedors, 2020-09-12:
## What happened
Battery driven sensor appears to be available in home assistant.
[image]

## What did you expect to happen
Show me that the device is not available. 

zigbee2mqttassistant does see that it is unavailable for weeks.
[image]

## How to reproduce it (minimal and precise)
Join a sensor to the network, take it offline again (e.g. put it out of range).

## Debug info
Zigbee2MQTT version: 1.14.4
Adapter hardware: CC1352P-2
Adapter firmware version: 20200805

Koenkk, 2020-09-12:
To have availability for battery powered devices see https://www.zigbee2mqtt.io/information/availability.html#non-pingable-devices

dedors, 2020-09-12:
This does not seem to work as my last restart of HA and z2m is over 50hours ago.

Koenkk, 2020-09-12:
@dedors can you share your `configuration.yaml`?

dedors, 2020-09-12:
I'm not sure if there needs to be a "availability_timeout" set in order to also have battery powered devices to show unavailable. 
The document suggests (at least for me with non-native English) that these are a own category and will be set unavailable even if availability_timeout is set to 0. 

`data_path: /config/zigbee2mqtt
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: ********
  user: ********
  password: ********
serial:
  port: /dev/ttyACM0
advanced:
  pan_id: 6755
  channel: 25
  network_key: ********
  availability_blacklist: []
  report: true
ban: []
whitelist: []
queue: {}
socat:
  enabled: false
  master: 'pty,raw,echo=0,link=/dev/ttyZ2M,mode=777'
  slave: 'tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5'
  restartdelay: 1
  initialdelay: 1
  options: '-d -d'
  log: false
`

Koenkk, 2020-09-12:
To enable availability `availability_timeout` != `0`, in case you only want it for battery powered devices I recommend setting this to a very high value e.g. `availability_timeout: 999999`. 5 minutes after starting Zigbee2MQTT the battery powered devices will be marked as offline (if their last seen is longer than 25 hours ago).

dedors, 2020-09-12:
Thanks! I can confirm the unavailable status now.

Can this information be included in the documentation? 
Both: that `availability_timeout ` need to be set (this can be confusing if I have to set a timeout that is not even used), 
but also that already after 5min it will get flagged unavailable if the "last seen" is older than 25 hours (I expected it needs 25 hours again, would made me test some more things before reporting).

Is there a reason availability is not active (at least with a high value) by default?

Koenkk, 2020-09-13:
Updated the docs.

By default the availability feature is not enabled as it is very heavy on the network and can make slower (e.g. CC2531) coordinators crash.

Having availability enabled by default for non-pingable devices might be worth considering indeed.

closed 2020-09-14 by dedors