Koenkk/zigbee2mqtt #4342: Unable to join LEDVANCE SMART+ Ceiling 33 Tunable White

kezmdor, 2020-09-12:
# Bug Report
## What happened

Trying to join LEDVANCE SMART+ Ceiling 33 Tunable White to latest version of zigbee2mqtt several times, Everytime it quit with the error "interview failed".

## What did you expect to happen

Successful join of the device.

## How to reproduce it (minimal and precise)

Reset the device, which starts the join process again.

## Debug Info
Zigbee2MQTT version: 1.14.4
Adapter hardware: CC1352P-2
Adapter firmware version: 20191106

## Logs
https://pastebin.com/PApPfrcF

Koenkk, 2020-09-12:
Make sure to check: https://www.zigbee2mqtt.io/information/FAQ.html#interview-fails

kezmdor, 2020-09-12:
I have to apologize. I tried nearly everything but not putting the device closer to the coordinator. :-(
Sorry for the inconvenience.
Near to the coordinatior it successfully joined within seconds in the first try.

closed 2020-09-12 by kezmdor