Koenkk/zigbee2mqtt #4353: Error while starting zigbee-herdsman after restoring backup (changing panID helps) - Error: AREQ - ZDO - stateChangeInd after 60000ms

mcdax, 2020-09-13:
Hi,
I'm running HASS.io with the Zigbee2MQTT-Addon on a Raspberry Pi 4 and a CC2652R1 as coordinator.
Yesterday I accidentally set the default configuration of zigbee2mqtt to default (with PanID 6754) and restored by backup after noticing my mistake. But now I'm not able to start zigbee2mqtt with my originally used PanID (6755). Only changing the PanID to another value helps here. (I don't want to repair all >25 devices.)

What I have tried so far:
- reflashing CC2652R1 (incl. erasing old data)
- reinstalling zigbee2mqtt
- removing zigbee2mqtt folder (just for testing)
- https://www.zigbee2mqtt.io/information/FAQ.html#cc1352p-2cc26x2r1-coordinators-only-press-the-reset-button-on-the-device

My config and logs:
https://hastebin.com/esapadokec.coffeescript

I hope you can give me some advice.

TheStigh, 2020-09-13:
Hi @mcdax , I had a similar problem the other day. After a good tip from @Koenkk I did the following;
- **Disconnect the adapter**
- Re-flash the adapter
- Restore the coordinator_backup.json
- restore the database.db
- restore the configuration.yaml
- OPTIONAL: I disconnected also my external antenna
- **Re-plug the adapter**

My adapter immediately discovered the network and z2m started working as before :)

mcdax, 2020-09-13:
Thank you very much! Z2M is starting and all my sensors are working again.

Unfortunately, there are some active devices that do not work:  
Zigbee2MQTT:error 2020-09-13 14:33:08: Failed to execute LQI for 'wohnzimmer_lampe'
Zigbee2MQTT:error 2020-09-13 14:33:13: Publish 'set' 'state' to 'arbeitszimmer_deckenlampe_1' failed: 'Error: Command 0x84182600000e30af/3 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'No network route' (205))'

closed 2020-09-13 by mcdax