Koenkk/zigbee2mqtt #4362: Hue Dimmer Switch Issue

SteveEdson, 2020-09-14:
## What happened

Hue dimmer switch used to work, then stopped. Device has no LED when pressing a button.

The switch used to work, then I used it to Touchlink reset some devices and it soon after became unresponsive. If I factory reset it using the button, I get the following in the logs, but it doesn't do anything afterwards:

```
zigbee2mqtt    | Zigbee2MQTT:info  2020-09-14 22:51:48: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{"block":false,"force":true,"id":"Bedroom Dimmer"},"status":"ok"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-09-14 22:52:14: Device '0x0017880102c37b16' joined
zigbee2mqtt    | Zigbee2MQTT:info  2020-09-14 22:52:14: Starting interview of '0x0017880102c37b16'
zigbee2mqtt    | Zigbee2MQTT:info  2020-09-14 22:52:14: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0017880102c37b16","ieee_address":"0x0017880102c37b16"},"type":"device_joined"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-09-14 22:52:14: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0017880102c37b16","ieee_address":"0x0017880102c37b16","status":"started"},"type":"device_interview"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-09-14 22:52:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x0017880102c37b16"},"type":"device_connected"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-09-14 22:52:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x0017880102c37b16"},"type":"pairing"}'
```

There's a strong chance I could have messed something up my side, I originally had a CC2531 stick, then swapped it for a different one with an antenna. I've then swapped it for a Conbee II (although I had the issue before with the CC2531 stick). The rest of my network is absolutely fine, consisting of Xiaomi door, motion and temperature sensors, Hue lights and Tradfri lights and switches.

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.14.4 (and latest-dev)
Adapter hardware: CC2531 + Conbee II
Adapter firmware version: Zigbee2MQTT 1.14.4-dev

Koenkk, 2020-09-15:
Did you try remove the dimmer trough https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove and repair it?

SteveEdson, 2020-09-16:
Yep, I'm using the new web dashboard. I've tried removing it, but get a timeout error. I've tried force removing it too with no luck.

I've also followed the steps at https://www.zigbee2mqtt.io/devices/324131092621.html#device-announces-itself-but-the-buttons-dont-work

Koenkk, 2020-09-16:
What do you see in the debug log when force removing and repairing?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

SteveEdson, 2020-09-16:
When trying to re-add the device, I get

```
zigbee2mqtt    | Zigbee2MQTT:info  2020-09-16 13:48:33: Device '0x0017880103caf67b' joined
zigbee2mqtt    | Zigbee2MQTT:info  2020-09-16 13:48:33: Starting interview of '0x0017880103caf67b'
zigbee2mqtt    | Zigbee2MQTT:info  2020-09-16 13:48:33: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0017880103caf67b","ieee_address":"0x0017880103caf67b"},"type":"device_joined"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-09-16 13:48:33: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0017880103caf67b","ieee_address":"0x0017880103caf67b","status":"started"},"type":"device_interview"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-09-16 13:48:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x0017880103caf67b"},"type":"device_connected"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-09-16 13:48:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x0017880103caf67b"},"type":"pairing"}'
```

Weirdly, I just did the same steps immediately after restarting the docker image and it worked straight away, could there be something causing an issue when it's been running for a while?

Koenkk, 2020-09-16:
@SteveEdson not sure, do you still have the logs of that? Otherwise it would be hard to investigate.

dchwilk, 2020-09-28:
Same problem since a frew days. 
Log while I try too readd device:
`Zigbee2MQTT:info  2020-09-28 18:49:59: Device '0x0017880102e2cd19' joined
Zigbee2MQTT:info  2020-09-28 18:49:59: Starting interview of '0x0017880102e2cd19'
Zigbee2MQTT:info  2020-09-28 18:49:59: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0017880102e2cd19","ieee_address":"0x0017880102e2cd19"},"type":"device_joined"}'
Zigbee2MQTT:info  2020-09-28 18:49:59: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0017880102e2cd19","ieee_address":"0x0017880102e2cd19","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2020-09-28 18:49:59: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x0017880102e2cd19"},"type":"device_connected"}'
Zigbee2MQTT:info  2020-09-28 18:49:59: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x0017880102e2cd19"},"type":"pairing"}'
Zigbee2MQTT:debug 2020-09-28 18:51:21: Received MQTT message on 'zigbee2mqtt/decke/get' with data ''
Zigbee2MQTT:error 2020-09-28 18:51:21: Invalid JSON '', skipping...
Zigbee2MQTT:error 2020-09-28 18:52:17: Failed to interview '0x0017880102e2cd19', device has not successfully been paired
Zigbee2MQTT:info  2020-09-28 18:52:17: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0017880102e2cd19","ieee_address":"0x0017880102e2cd19","status":"failed"},"type":"device_interview"}'
Zigbee2MQTT:info  2020-09-28 18:52:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x0017880102e2cd19"},"type":"pairing"}'`

closed 2020-11-06 by github-actions[bot]