Koenkk/zigbee2mqtt #4363: Lonsonho 2 gang switch module with neutral wire detected as Zemismart 3 gang in Home Assistant

McGiverGim, 2020-09-15:
## What happened
I added a `Lonsonho 2 gang switch module with neutral wire (QS-Zigbee-S04-2C-LN)` device to my network. I expected to see two switches (left/right) in Home Assistant, but I see it identified as a `Zemismart with three gang` (two inside the MQTT integration and the other as a switch not available????):

[image]

[image]

[image]

Here is the log, it seems correctly recognized, but it fails when trying to change the state:
```txt
info  2020-09-15 11:50:23: Zigbee: allowing new devices to join.
info  2020-09-15 11:50:23: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"unknown","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.14.4"}'
info  2020-09-15 11:50:36: Device '0x680ae2fffe096ce3' joined
info  2020-09-15 11:50:36: Starting interview of '0x680ae2fffe096ce3'
info  2020-09-15 11:50:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x680ae2fffe096ce3"},"type":"device_connected"}'
info  2020-09-15 11:50:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x680ae2fffe096ce3"},"type":"pairing"}'
info  2020-09-15 11:50:38: MQTT publish: topic 'homeassistant/switch/0x680ae2fffe096ce3/switch_left/config', payload '{"availability_topic":"zigbee2mqtt/bridge/state","command_topic":"zigbee2mqtt/0x680ae2fffe096ce3/left/set","device":{"identifiers":["zigbee2mqtt_0x680ae2fffe096ce3"],"manufacturer":"Zemismart","model":"Smart light switch - 3 gang with neutral wire (ZM-L03E-Z)","name":"0x680ae2fffe096ce3","sw_version":"Zigbee2MQTT 1.14.4"},"json_attributes_topic":"zigbee2mqtt/0x680ae2fffe096ce3","name":"0x680ae2fffe096ce3_switch_left","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x680ae2fffe096ce3","unique_id":"0x680ae2fffe096ce3_switch_left_zigbee2mqtt","value_template":"{{ value_json.state_left }}"}'
info  2020-09-15 11:50:38: MQTT publish: topic 'homeassistant/switch/0x680ae2fffe096ce3/switch_center/config', payload '{"availability_topic":"zigbee2mqtt/bridge/state","command_topic":"zigbee2mqtt/0x680ae2fffe096ce3/center/set","device":{"identifiers":["zigbee2mqtt_0x680ae2fffe096ce3"],"manufacturer":"Zemismart","model":"Smart light switch - 3 gang with neutral wire (ZM-L03E-Z)","name":"0x680ae2fffe096ce3","sw_version":"Zigbee2MQTT 1.14.4"},"json_attributes_topic":"zigbee2mqtt/0x680ae2fffe096ce3","name":"0x680ae2fffe096ce3_switch_center","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x680ae2fffe096ce3","unique_id":"0x680ae2fffe096ce3_switch_center_zigbee2mqtt","value_template":"{{ value_json.state_center }}"}'
info  2020-09-15 11:50:38: MQTT publish: topic 'homeassistant/switch/0x680ae2fffe096ce3/switch_right/config', payload '{"availability_topic":"zigbee2mqtt/bridge/state","command_topic":"zigbee2mqtt/0x680ae2fffe096ce3/right/set","device":{"identifiers":["zigbee2mqtt_0x680ae2fffe096ce3"],"manufacturer":"Zemismart","model":"Smart light switch - 3 gang with neutral wire (ZM-L03E-Z)","name":"0x680ae2fffe096ce3","sw_version":"Zigbee2MQTT 1.14.4"},"json_attributes_topic":"zigbee2mqtt/0x680ae2fffe096ce3","name":"0x680ae2fffe096ce3_switch_right","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x680ae2fffe096ce3","unique_id":"0x680ae2fffe096ce3_switch_right_zigbee2mqtt","value_template":"{{ value_json.state_right }}"}'
info  2020-09-15 11:50:38: MQTT publish: topic 'homeassistant/sensor/0x680ae2fffe096ce3/linkquality/config', payload '{"availability_topic":"zigbee2mqtt/bridge/state","device":{"identifiers":["zigbee2mqtt_0x680ae2fffe096ce3"],"manufacturer":"Zemismart","model":"Smart light switch - 3 gang with neutral wire (ZM-L03E-Z)","name":"0x680ae2fffe096ce3","sw_version":"Zigbee2MQTT 1.14.4"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x680ae2fffe096ce3","name":"0x680ae2fffe096ce3_linkquality","state_topic":"zigbee2mqtt/0x680ae2fffe096ce3","unique_id":"0x680ae2fffe096ce3_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info  2020-09-15 11:50:38: MQTT publish: topic 'homeassistant/switch/0x680ae2fffe096ce3/switch_right/config', payload 'null'
info  2020-09-15 11:50:39: Successfully interviewed '0x680ae2fffe096ce3', device has successfully been paired
info  2020-09-15 11:50:39: Device '0x680ae2fffe096ce3' is supported, identified as: Lonsonho 2 gang switch module with neutral wire (QS-Zigbee-S04-2C-LN)
info  2020-09-15 11:50:39: Configuring '0x680ae2fffe096ce3'
info  2020-09-15 11:50:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"2 gang switch module with neutral wire","friendly_name":"0x680ae2fffe096ce3","model":"QS-Zigbee-S04-2C-LN","supported":true,"vendor":"Lonsonho"},"type":"pairing"}'
info  2020-09-15 11:50:39: MQTT publish: topic 'zigbee2mqtt/0x680ae2fffe096ce3', payload '{"linkquality":105,"state_l1":"ON"}'
info  2020-09-15 11:50:39: MQTT publish: topic 'zigbee2mqtt/0x680ae2fffe096ce3', payload '{"linkquality":105,"state_l1":"ON","state_l2":"ON"}'
info  2020-09-15 11:50:39: Successfully configured '0x680ae2fffe096ce3'
info  2020-09-15 11:50:39: MQTT publish: topic 'zigbee2mqtt/rele_cocina', payload '{"consumption":0.05,"linkquality":49,"power":0.08,"state_l1":"OFF","state_l2":"OFF","temperature":42}'
info  2020-09-15 11:51:13: MQTT publish: topic 'zigbee2mqtt/presencia_despacho', payload '{"battery":100,"illuminance":1,"illuminance_lux":1,"linkquality":49,"occupancy":true,"voltage":3005}'
error 2020-09-15 11:51:22: Failed to call 'EntityPublish' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Endpoint name 'center' is given but device has no such endpoint
    at Zigbee.resolveEntity (/zigbee2mqtt-1.14.4/lib/zigbee.js:214:25)
    at EntityPublish.onMQTTMessage (/zigbee2mqtt-1.14.4/lib/extension/publish.js:64:44)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.14.4/lib/controller.js:365:44)
    at Controller.onMQTTMessage (/zigbee2mqtt-1.14.4/lib/controller.js:262:14)
    at MQTT.emit (events.js:315:20)
    at MQTT.onMessage (/zigbee2mqtt-1.14.4/lib/mqtt.js:95:14)
    at MqttClient.emit (events.js:315:20)
    at MqttClient._handlePublish (/zigbee2mqtt-1.14.4/node_modules/mqtt/lib/client.js:1271:12)
    at MqttClient._handlePacket (/zigbee2mqtt-1.14.4/node_modules/mqtt/lib/client.js:410:12)
    at work (/zigbee2mqtt-1.14.4/node_modules/mqtt/lib/client.js:321:12))
error 2020-09-15 11:51:23: Failed to call 'EntityPublish' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Endpoint name 'center' is given but device has no such endpoint
    at Zigbee.resolveEntity (/zigbee2mqtt-1.14.4/lib/zigbee.js:214:25)
    at EntityPublish.onMQTTMessage (/zigbee2mqtt-1.14.4/lib/extension/publish.js:64:44)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.14.4/lib/controller.js:365:44)
    at Controller.onMQTTMessage (/zigbee2mqtt-1.14.4/lib/controller.js:262:14)
    at MQTT.emit (events.js:315:20)
    at MQTT.onMessage (/zigbee2mqtt-1.14.4/lib/mqtt.js:95:14)
    at MqttClient.emit (events.js:315:20)
    at MqttClient._handlePublish (/zigbee2mqtt-1.14.4/node_modules/mqtt/lib/client.js:1271:12)
    at MqttClient._handlePacket (/zigbee2mqtt-1.14.4/node_modules/mqtt/lib/client.js:410:12)
    at work (/zigbee2mqtt-1.14.4/node_modules/mqtt/lib/client.js:321:12))
error 2020-09-15 11:51:27: Failed to call 'EntityPublish' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Endpoint name 'left' is given but device has no such endpoint
    at Zigbee.resolveEntity (/zigbee2mqtt-1.14.4/lib/zigbee.js:214:25)
    at EntityPublish.onMQTTMessage (/zigbee2mqtt-1.14.4/lib/extension/publish.js:64:44)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.14.4/lib/controller.js:365:44)
    at Controller.onMQTTMessage (/zigbee2mqtt-1.14.4/lib/controller.js:262:14)
    at MQTT.emit (events.js:315:20)
    at MQTT.onMessage (/zigbee2mqtt-1.14.4/lib/mqtt.js:95:14)
    at MqttClient.emit (events.js:315:20)
    at MqttClient._handlePublish (/zigbee2mqtt-1.14.4/node_modules/mqtt/lib/client.js:1271:12)
    at MqttClient._handlePacket (/zigbee2mqtt-1.14.4/node_modules/mqtt/lib/client.js:410:12)
    at work (/zigbee2mqtt-1.14.4/node_modules/mqtt/lib/client.js:321:12))
[... cut ...]
    at MqttClient._handlePacket (/zigbee2mqtt-1.14.4/node_modules/mqtt/lib/client.js:410:12)
    at work (/zigbee2mqtt-1.14.4/node_modules/mqtt/lib/client.js:321:12))
info  2020-09-15 11:52:23: Zigbee: disabling joining new devices.
```

## What did you expect to happen
To be recognized correctly and only have two gang/switches.

## How to reproduce it (minimal and precise)
Join the network with this device.

## Debug info
Zigbee2MQTT version: 1.14.4
Adapter hardware: CC2531
Adapter firmware version: zStack12.20190608

McGiverGim, 2020-09-15:
After restarting all, for some reason now it is correct:

[image]

The l1 and l2 switches have appeared.
[image]

After that, I have removed the two old entities that are unavailable and all is ok.

I will let this open, maybe is interesting to find the root cause of this problem, but if you think that is not, feel free to close this issue.

Koenkk, 2020-09-15:
Found the issue, should not happen anymore in latest dev.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-09-15 by Koenkk

mihaimdinca, 2022-11-01:
I have the same problem except restarting did not fix anything for me. I am using the Z2M Add-on for Home Assistant version 1.28.0-1

mihaimdinca, 2022-11-01:
I have another relay of the same type, paired a few months ago, that got recognized correctly from the first try Lonsonho QS-Zigbee-S04-2C-LN