Koenkk/zigbee2mqtt #4366: Interview failed because can not get node descriptor ('$HexString')

SimonReiger, 2020-09-15:
I am running Z2MQTT on a CC2531 with Mosquitto. Connected to a raspberry Pi with an extension cable
zigbee2mqtt version 1.14.2 (commit #b91b456)
It worked fine in the beginning and the 5 devices I had worked like a charm
But now I am absolutely not able to pair new devices. I tried now for about ~50 times and no success. 

I have 2 devices to add
-WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
-WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor (EndDevice)
  
Niether is able to pair. Both have new batteries

Both fail with 
**"Node descriptor request for '5541' failed (Error: AREQ - ZDO - nodeDescRsp after 10000ms), retry +1m"**
and
**"Interview - node descriptor request failed for '0x00158d0005404d68', attempt 1 +23s"**

I tried everything for days. Even in a completey shielded room. Device doesn't join with address 0x000000000000000, so I would say the CC2531 is fine. This is giving me staggering depression by now......

Here is a  _DEBUG=zigbee-herdsman:* npm start_

pi@ACCORDLogger:/opt/zigbee2mqtt $ sudo DEBUG=zigbee-herdsman:* npm start

> zigbee2mqtt@1.14.2 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-09-15 15:18:16: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-09-15.15-18-16' filename: log.txt
zigbee2mqtt:debug 2020-09-15 15:18:16: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-09-14.16-28-26'
zigbee2mqtt:debug 2020-09-15 15:18:17: Loaded state from file /opt/zigbee2mqtt/data/state.json
zigbee2mqtt:info  2020-09-15 15:18:17: Starting zigbee2mqtt version 1.14.2 (commit #b91b456)
zigbee2mqtt:info  2020-09-15 15:18:17: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-09-15 15:18:17: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"path":"/dev/ttyACM0"},"adapter":{"concurrent":null}}'
  zigbee-herdsman:adapter Path '/dev/ttyACM0' is valid for 'ZStackAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[160,80,17,86,14,205,183,217,56,42,188,168,134,246,147,98],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"concurrent":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":false,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +21ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,37,68,129,0,0,3,4,58,44,1,1,0,55,0,219,228,73,0,0,17,24,184,10,0,0,41,205,3,20,0,40,255,16,0,41,2,38,58,44,29,57] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,37,68,129,0,0,3,4,58,44,1,1,0,55,0,219,228,73,0,0,17,24,184,10,0,0,41,205,3,20,0,40,255,16,0,41,2,38,58,44,29,57] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 37 - 2 - 4 - 129 - [0,0,3,4,58,44,1,1,0,55,0,219,228,73,0,0,17,24,184,10,0,0,41,205,3,20,0,40,255,16,0,41,2,38,58,44,29] - 57 +3ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1027,"srcaddr":11322,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":55,"securityuse":0,"timestamp":4842715,"transseqnumber":0,"len":17,"data":{"type":"Buffer","data":[24,184,10,0,0,41,205,3,20,0,40,255,16,0,41,2,38]}} +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +22ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +500ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,121,1,26] +1s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,121,1,26] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [121,1] - 26 +3ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":377} +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +4ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {} +32ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35] +32ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,14,97,2,2,0,2,6,3,144,21,52,1,2,0,0,0,0,218] +19ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,97,2,2,0,2,6,3,144,21,52,1,2,0,0,0,0,218] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 14 - 3 - 1 - 2 - [2,0,2,6,3,144,21,52,1,2,0,0,0,0] - 218 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608} +24ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:adapter:zStack:adapter Adapter concurrent: 2 +0ms
  zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack12' ({"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}) +1ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":3840,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}} +23ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,0,15,0,37] +22ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,85,62] +19ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,85,62] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 8 - [0,1,85] - 62 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":1,"value":{"type":"Buffer","data":[85]}} +23ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +3ms
  zigbee-herdsman:adapter:zStack:startZnp Item 'hasConfigured' is valid +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":3840,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}} +24ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,0,15,0,37] +23ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,85,62] +16ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,85,62] +1ms
[... 561 lines cut ...]
  zigbee-herdsman:controller:log Permit joining +3m
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0} +3m
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228] +3m
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,54,0,82] +3m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [0] - 82 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0} +3m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":4,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,5,2,11,254,0]}} +24ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,26,36,2,2,253,255,0,0,0,0,0,0,242,0,0,242,33,0,4,0,30,6,0,25,5,2,11,254,0,234] +28ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,182,0,0,0,240] +11ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [0,0,0] - 240 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0} +3m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103] +9ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103 +1ms
[... cut ...]
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,242,4,49] +21ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,242,4,49] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,242,4] - 49 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":242,"transid":4} +34ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms

Koenkk, 2020-09-15:
Have you tried the steps in: https://www.zigbee2mqtt.io/information/FAQ.html#interview-fails ? If that doesn't help try reflashing the CC2531 (won't require repairing of existing devices)

rcpoison, 2020-09-23:
Also receiving 'Error: Interview failed because can not get node descriptor'

System:
- Linux/x86_64
- CC2531

Zigbee2MQTT versions tested:
- 1.14.2
- 1.14.4 (commit #0cd46078)

Sensors:
2x Konke motion sensor, new (0x14b457fffe46a84f, 0x14b457fffe46a85d)

Steps tried according to https://www.zigbee2mqtt.io/information/FAQ.html#interview-fails
- reboot :P
- directly next to stick
- directly plugged in without extension cable
- replaced a battery on one of the sensors
- tried resetting+re-pairing quite a few times
- reflashing the stick, multiple times

Log of pairing attempts of both devices:
[log-konke.txt](https://github.com/Koenkk/zigbee2mqtt/files/5272395/log-konke.txt)

closed 2020-11-01 by github-actions[bot]

rcpoison, 2020-11-04:
Just upgraded to zigbee2mqtt-1.16.0, issue still persists:
```
2020-11-04T12:04:31.246Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":53718,"extaddr":"0x14b457fffe46a84f","parentaddr":52907}
2020-11-04T12:04:31.246Z zigbee-herdsman:controller:log Device '0x14b457fffe46a84f' joined
2020-11-04T12:04:31.247Z zigbee-herdsman:controller:log Device '0x14b457fffe46a84f' accepted by handler
2020-11-04T12:04:31.247Z zigbee-herdsman:controller:log Device '0x14b457fffe46a84f' is already in database with different networkAddress, updating networkAddress
2020-11-04T12:04:31.250Z zigbee-herdsman:controller:log Interview '0x14b457fffe46a84f' start
2020-11-04T12:04:31.251Z zigbee-herdsman:controller:device:log Interview - start device '0x14b457fffe46a84f'
2020-11-04T12:04:54.309Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x14b457fffe46a84f', attempt 1
2020-11-04T12:05:17.352Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x14b457fffe46a84f', attempt 2
2020-11-04T12:05:40.396Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x14b457fffe46a84f', attempt 3
2020-11-04T12:06:03.438Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x14b457fffe46a84f', attempt 4
2020-11-04T12:06:26.489Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x14b457fffe46a84f', attempt 5
2020-11-04T12:06:49.538Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x14b457fffe46a84f', attempt 6
2020-11-04T12:06:49.539Z zigbee-herdsman:controller:device:log Interview - failed for device '0x14b457fffe46a84f' with error 'Error: Interview failed because can not get node descriptor ('0x14b457fffe46a84f')
2020-11-04T12:06:49.541Z zigbee-herdsman:controller:error Interview failed for '0x14b457fffe46a84f with error 'Error: Interview failed because can not get node descriptor ('0x14b457fffe46a84f')'
2020-11-04T12:07:43.883Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":49875,"extaddr":"0x14b457fffe46a85d","parentaddr":0}
2020-11-04T12:07:43.883Z zigbee-herdsman:controller:log Device '0x14b457fffe46a85d' joined
2020-11-04T12:07:43.883Z zigbee-herdsman:controller:log Device '0x14b457fffe46a85d' accepted by handler
2020-11-04T12:07:43.883Z zigbee-herdsman:controller:log New device '0x14b457fffe46a85d' joined
2020-11-04T12:07:43.883Z zigbee-herdsman:controller:log Creating device '0x14b457fffe46a85d'
2020-11-04T12:07:43.886Z zigbee-herdsman:controller:log Interview '0x14b457fffe46a85d' start
2020-11-04T12:07:43.886Z zigbee-herdsman:controller:device:log Interview - start device '0x14b457fffe46a85d'
2020-11-04T12:08:06.917Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x14b457fffe46a85d', attempt 1
2020-11-04T12:08:29.948Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x14b457fffe46a85d', attempt 2
2020-11-04T12:08:52.978Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x14b457fffe46a85d', attempt 3
2020-11-04T12:09:16.009Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x14b457fffe46a85d', attempt 4
2020-11-04T12:09:39.041Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x14b457fffe46a85d', attempt 5
2020-11-04T12:10:02.073Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x14b457fffe46a85d', attempt 6
2020-11-04T12:10:02.074Z zigbee-herdsman:controller:device:log Interview - failed for device '0x14b457fffe46a85d' with error 'Error: Interview failed because can not get node descriptor ('0x14b457fffe46a85d')
2020-11-04T12:10:02.075Z zigbee-herdsman:controller:error Interview failed for '0x14b457fffe46a85d with error 'Error: Interview failed because can not get node descriptor ('0x14b457fffe46a85d')'
```