Koenkk/zigbee2mqtt #4367: brightness_move won't set brightness

CharlsTrtanj, 2020-09-15:
## What happened
Basically the same what happened here: "[brightness_move won't return current brightness for Tradfri bulb #2667](https://github.com/Koenkk/zigbee2mqtt/issues/2667)"
Using "brightness_move" does not update the variable "brightness".

## What did you expect to happen
I expected the mqtt topic to reflect the current brightness value of the bulb.
I'm able to change the brightness with "brightness_move" and stop the process but unfortunatelly the change is not reflected by the variable "brightness" afterwads.
Therefore the change is lost after turning the light off and on again.

## How to reproduce it (minimal and precise)
1. Turn the light on
2. Change brightness with "brightness_move"
3. Turn the light off
4. Turn the light on again (brightness will be set to the state before you used "brightness_move")

My device is: ICPSHC24-30EU-IL-1 - IKEA TRADFRI driver for wireless control (30 watt) (Router)

## Debug info
Current version: 1.14.4-dev

I have a complete debug log here: [https://pastebin.com/twxCALuu](https://pastebin.com/twxCALuu)

Koenkk, 2020-09-15:
After brightness move, do you send `{"brightness_move": "stop"}`

CharlsTrtanj, 2020-09-15:
I'm sending {"brightness_move": 0}.
In the documentation it is stated that this should stop the brightness move. I guess I'm wrong?!
I will change it to {"brightness_move": "stop"}.

CharlsTrtanj, 2020-09-15:
Indeed, sending {"brightness_move": "stop"} saves the correct value to "brightness". Maybe the [documentation](https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html) should point that out. I was confused because I thought "0" and "stop" is equivalent.
#####
  "brightness_move": -40, // Starts moving brightness down at 40 units per second
  "brightness_move": 0, // Stop moving brightness
  "brightness_step": 40 // Increases brightness by 40
  "color_temp_move": 60, // Starts moving color temperature up at 60 units per second
  "color_temp_move": "stop", // Stop moving color temperature
#####

Thank You Koenkk for your help!
BR

Koenkk, 2020-09-15:
0 should indeed equal stop, but this was not the case! Fix this, thanks!

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-09-15 by Koenkk