Koenkk/zigbee2mqtt #4368: Repeating Issue: Failed to execute LQI for 'Coordinator'

gregbert42, 2020-09-15:
#What happened

After a day or two, when I scan the network, it fails to create a network map (Scan network on home assistant), and becomes a bit slower to convey messages. When I look into the logs, I see the following, with 'Failed to execute LQI" for a number of items.

```
Zigbee2MQTT:info  2020-09-15 14:23:03: Starting network scan (includeRoutes 'false')
Zigbee2MQTT:info  2020-09-15 14:23:12: MQTT publish: topic 'zigbee2mqtt/TempSensor2', payload '{"battery":80,"humidity":38.19,"linkquality":49,"temperature":32.13,"voltage":2965}'
Zigbee2MQTT:info  2020-09-15 14:23:12: MQTT publish: topic 'zigbee2mqtt/TempSensor2', payload '{"battery":80,"humidity":37.09,"linkquality":2,"temperature":32.13,"voltage":2965}'
Zigbee2MQTT:info  2020-09-15 14:23:12: MQTT publish: topic 'zigbee2mqtt/TempSensor2', payload '{"battery":80,"humidity":37.09,"linkquality":52,"temperature":32.13,"voltage":2965}'
Zigbee2MQTT:info  2020-09-15 14:23:12: MQTT publish: topic 'zigbee2mqtt/TempSensor2', payload '{"battery":80,"humidity":37.09,"linkquality":2,"temperature":32.13,"voltage":2965}'
Zigbee2MQTT:error 2020-09-15 14:23:13: Failed to execute LQI for 'Coordinator'
Zigbee2MQTT:error 2020-09-15 14:23:23: Failed to execute LQI for 'Fan_Power'
Zigbee2MQTT:error 2020-09-15 14:23:42: Failed to execute LQI for 'Power_Plug2'
Zigbee2MQTT:error 2020-09-15 14:23:52: Failed to execute LQI for 'Greg_Bed'
Zigbee2MQTT:error 2020-09-15 14:24:02: Failed to execute LQI for 'IKEA_Wardrobe_LED'
Zigbee2MQTT:info  2020-09-15 14:24:02: Network scan finished
```
## What did you expect to happen
Retain integrity

## How to reproduce it (minimal and precise)
Restart zigbee2mqtt (in docker). Wait 2 days.

## Debug info
Zigbee2MQTT version 1.14.4 (commit #654817a)
Product: TI CC2531 USB CDC
Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}'

syphernl, 2020-09-15:
This looks pretty similar to the issues reported in https://github.com/Koenkk/zigbee2mqtt/issues/4307. This also occurs after a few hours/days.

Koenkk, 2020-09-15:
Indeed a duplicate of #4307, some possible fixes are already in the dev branch: https://github.com/Koenkk/zigbee2mqtt/issues/4307#issuecomment-691682736

closed 2020-09-15 by Koenkk

gregbert42, 2020-09-15:
I looked over those comments, and while they do share the characteristic of happening after a couple of days, there are notable differences.

#4307: After some days of operation, my entire Zigbee system fails. No devices report and none can be controlled. Reconnecting the zzh coordinator (CC2652R) fixes the problem for a while.

But in my cases, all the devices report, and they can all still be controlled. The only noticeable change is that the system is less responsive, and I am sure it gets worse with time. A very quick unplug/replug/restart zigbee2mqtt solves the issue

This only started happening recently, and may have something to do with me moving one of my devices (a powerplug) from downstairs to upstairs.

I admit that the similarity in timing with #4307 is suspicious - and maybe I am just on the cusp of having the same issue.

Koenkk, 2020-09-15:
@gregbert42 I suspect the queue filling up, this is something that has been addressed in the latest dev, so please check that first.

gregbert42, 2020-09-15:
@Koenkk Ok, i've switched over to the latest-dev docker image.

 Starting Zigbee2MQTT version 1.14.4-dev (commit #a47217f)

gregbert42, 2020-09-16:
So far so good.  
**But of Note**:
I had a problem where all of the docker containers that have a health check were going unhealthy after ~1-2 days.
After switching to this -dev branch, that problem has also gone away.
I don't know how that is possible... but i was struggling with that one for weeks and had given up.. the zigbee2mqtt problem also appeared at the same time the docker healthcheck problems appeared.

may just be a coincidence.