Koenkk/zigbee2mqtt #4372: Smoothness timer for Symfonisk sound controller E1744

fliphess, 2020-09-15:
## Is your feature request related to a problem? 

I'm using the Tradfri E1744 symfonisk sound controller and I want to control either a dimmer or sound volume with it, but as there are only messages send when the rotary button starts or stops moving in the left or right direction, it's hard to create a smooth volume controller or dimmer with it. 

## Describe the solution you'd like

I'd like to have a "fader emulator" or internal timer that responds to the left and or right events that triggers periodic updates from the time the start event is send until the stop event.  

In other words: I'm looking for an internal mechanism that keeps sending the the `brightness_move_up` or `brightness_move_down` MQTT messages to the topic every N milliseconds until the `brightness_stop` event is received.

This behaviour is somewhat described in Koenkk/zigbee-herdsman-converters#658

## Describe alternatives you've considered

I've tried to do this in home-assistant using scripts that trigger each other in a loop until the `brightness_stop` event is received and this works, but there are so many parts moving that this results too much delay between starting to move and the action taken in home assistant which as a result turns my speakers up to 11 while level 2 or 3 was expected.

I've considered to write a small daemon that does this or use controllerX instead, but both require additional daemons that make the functionality for smoothly change a state a bit messy. 

Preferably it would be a feature flag enabled from the device type specific configuration that enables this timing functionality.

(Oh and ofc: Thanks so much for making this great tool)

ale1800, 2020-09-16:
Hi, you can try ControllerX. It uses the zigbee2mqtt integration and with few lines and no automations you can use it as a brightness or volume controller.
I have a E1744 too. 
One question. Do you have delay between when you start rotating and when your mqtt messages ar received? I opened this issue #4273 but still can't solve

fliphess, 2020-09-16:
Hey @ale1800! 

I am currently using ControllerX but I moved away from appdaemon a while ago, and having to reintroduce it felt a bit overkill (I did anyway for now, but it's not my favorite choice). 

Having all my automations on one place (I have several hundreds of automations declared in home assistant in yaml), was the main reason of not using appdaemon anymore (the dashboards could easily be replaced with the lovelace ui). Using appdaemon for a few automations while the others are all in yaml in home assistant is acceptable for now but not optimal as it introduces some complexity.

Preferably I skip using appdaemon and have this single functionality added to zigbee2mtt :)

fliphess, 2020-09-16:
Oh and to answer your question: about the delays @ale1800  

Getting ControllerX to work was a bit of a pita as there a several moving parts, but I got it to work without delay.

The most important thing: Make sure your zigbee network is stable and well covering your space so you don't have any connectivity of transmission issues. Add a router in each room and the zigbee coordinator in a central location and use a long (or multiple) usb extension cable(s) to place the coordinator stick some 2 meters away from your pi or computer. 

Good connectivity saves you most of your delay. 

When your network is stable, make sure appdaemon is listening directly on the MQTT broker topics instead of through the home assistant API, this saves you another few milliseconds on a raspberry pi.

My controllerx configuration:
```
sound_controller:
  module: controllerx
  class: E1744MediaPlayerController
  controller: sound_controller
  integration:
    name: z2m
    listen_to: mqtt
    action_key: action
  media_player: media_player.something
```

ale1800, 2020-09-16:
Ok, thank you! I'll try to stabilize my zigbee network. Crossed finger. I hope my remote isn't defective

closed 2020-10-12 by Koenkk