Koenkk/zigbee2mqtt #4374: Problem with Ikea 1603 connection

arnold-maderthaner, 2020-09-15:
Hi !

I bought recently a Ikea 1603 power supply and wanted to add it to my network but I fail to get this done. 
When I try to push the reset button of the plug it doesn't do anything while holding. When I release it after about 10 seconds I get the following device removal message:

warn  2020-09-15 13:52:53: Device '0xccccccfffeb30ddf' left the network
info  2020-09-15 13:52:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0xccccccfffeb30ddf"},"type":"device_removed"}'

The issue is that I never get a connected event - only device removals no matter how often I try to reset the plug I always only get the disconnection messages.

I'm on the newest zigbee2mqtt version and tried alreay different things like rebooting my Intel NUC and also disconnecting/connecting my CC2531.

anyone knows what I'm doing wrong ?

Koenkk, 2020-09-15:
Did you also try reflashing the CC2531?

arnold-maderthaner, 2020-09-16:
what would be the consequences of reflashing and with which FW ? would I need to reinclude every device ?

Koenkk, 2020-09-16:
https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing and no it doesn't require repairing.

arnold-maderthaner, 2020-09-17:
I reflashed now the source routing FW and it doesn't change anything - I only get the left network messages. I also try to pair a Ikea Tradfri E1743 button and get the same messages when I press 4 times the button  inside the case:
```
warn  2020-09-17 10:28:24: Device '0x680ae2fffe78b8cc' left the network
```

Koenkk, 2020-09-17:
Have you also checked the points of: https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair ?

arnold-maderthaner, 2020-09-17:
yes checked that - the devices are brand new and never connected to something else - I do the pairing 20 cm to the coordinator and nothing shows up - reflashed the source firmware - restarted and reconnected. 
I try 2 different devices (Ikea plug and button) and both don't work :(

Koenkk, 2020-09-17:
Could you provide the herdsman debug logging when pairing the device?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

arnold-maderthaner, 2020-09-19:
now it paired successfully when I start it in debug mode !?! I did the very same thing at the same location when trying to pair it as in normal start mode.

any idea why this happens ?

I can also pair now the ikea button but I get an error afterwards for OTA:
Zigbee2MQTT:debug 2020-09-19 10:06:40: Check if update available for '0xccccccfffeb30ddf' (TRADFRI control outlet)
Zigbee2MQTT:error 2020-09-19 10:06:40: Failed to call 'OTAUpdate' 'onZigbeeEvent' (Error: Request failed with status code 434
    at createError (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/node_modules/axios/lib/core/createError.js:16:15)
    at settle (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/node_modules/axios/lib/core/settle.js:17:12)
    at IncomingMessage.handleStreamEnd (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/node_modules/axios/lib/adapters/http.js:244:11)
    at IncomingMessage.emit (events.js:327:22)
    at endReadableNT (_stream_readable.js:1221:12)

Koenkk, 2020-09-19:
Good that you managed to pair, not sure what happend.

434 error has just been fixed: https://github.com/Koenkk/zigbee2mqtt/issues/4397

arnold-maderthaner, 2020-09-19:
the question is why can I only pair when in debug mode ?

Koenkk, 2020-09-19:
@arnold-maderthaner is this reproducible? Debug logging shouldn't change anything except logging.

arnold-maderthaner, 2020-09-19:
yes it was reproduceable - only if I had it in debug (with DEBUG=zigbee-herdsman* npm start) I could add devices instantly without issues. When I had it running with my normal way (systemctl start zigbee2mqtt.service) it didn't work at all

Koenkk, 2020-09-19:
And with just debug logging? So no `DEBUG=zigbee-herdsman*` but only the following in your `configuration.yaml`:

```yaml
advanced:
  log_level: debug
```

arnold-maderthaner, 2020-09-19:
thats what I always have turned on :)

Koenkk, 2020-09-19:
can you try with DEBUG=nothing ?

arnold-maderthaner, 2020-10-01:
with DEBUG=nothing I can also join the device

Koenkk, 2020-10-02:
Strange! Puzzled on what is going on here. Would you able to sniff the traffic of both situations (working vs non-working)? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

closed 2020-11-09 by github-actions[bot]

MixelF, 2022-10-02:
I have the same issue with this ikea control, in combination woth sonoff usb dongle (E variant).

Bulbs pair, control doesnt. Should trying in debug mode help?