Koenkk/zigbee2mqtt #4375: Touchlink-Scan fails to execute

pulsar256, 2020-09-15:
## What happened

Issuing a Touchlink-Scan triggers an unhandled exception rendering the Touchlink-Scan process to never complete:

```
zigbee2mqtt_1  | Zigbee2MQTT:debug 2020-09-15 19:36:47: Received MQTT message on 'zigbee2mqtt/bridge/request/touchlink/scan' with data '""'
zigbee2mqtt_1  | Zigbee2MQTT:info  2020-09-15 19:36:47: Start Touchlink scan
zigbee2mqtt_1  | Zigbee2MQTT:error 2020-09-15 19:36:47: Failed to call 'Bridge' 'onMQTTMessage' (TypeError: Cannot read property 'message' of undefined
zigbee2mqtt_1  |     at Bridge.onMQTTMessage (/app/lib/extension/bridge.js:73:78)
zigbee2mqtt_1  |     at processTicksAndRejections (internal/process/task_queues.js:97:5))
```

## What did you expect to happen

Touchlink Scan to be performed

## How to reproduce it (minimal and precise)

Issue a Touchlink-Scan using z2m admin or zigbee2mqttassistant

## Debug info

Zigbee2MQTT (dockerized) version:

`:latest (e4b207bd3c669e9bdbaa6690c7c3d86ab4ceed0d03af44f39d4638e9bd2497ff)` 
`:latest-dev (DIGEST:sha256:64de432b765790e1f3fe90e9ba6b121d9a1fdd966a3f14988e3bb2599dccc69f)` (1.14.4-dev)

Adapter hardware: Conbee II

Adapter firmware version:

```
"coordinator": {
        "meta": {
            "maintrel": 0,
            "majorrel": 38,
            "minorrel": 74,
            "product": 0,
            "revision": "0x264a0700",
            "transportrev": 0
        },
        "type": "ConBee2"
    },
```

Koenkk, 2020-09-15:
Touchlink is not supported yet for the Deconz adapter. @ChrisHae can this be supported?

ChrisHae, 2020-09-16:
It is not supported by the FW, yet. I think It is not supposed to come this year.

Koenkk, 2020-09-16:
Thanks @ChrisHae , I think this can be closed in that case.

closed 2020-09-16 by Koenkk

pulsar256, 2020-09-16:
Perhaps handling this edgecase gracefully is still worth looking at. Returning a scan result with "unsupported operation" to prevent management frontends to stay in the "scanning in progress" seems reasonable to me. At least a log message along these lines would help.

Koenkk, 2020-09-16:
@ChrisHae maybe change [`return Promise.reject();`](https://github.com/Koenkk/zigbee-herdsman/blob/c6d23d71a9c4cedee092ba78fdd695b0027f91f7/src/adapter/deconz/adapter/deconzAdapter.ts#L940)  to `throw new Error("Unsupported operation")`

ChrisHae, 2020-09-16:
Yes, thats a good idea.