Koenkk/zigbee2mqtt #4378: 4257050-RZHAC Incorrect Values

zikeji, 2020-09-16:
## What happened
Payload reports a large amount of power usage and current.
```
{"current":265,"linkquality":37,"power":2940,"state":"ON","voltage":119}
```

In the above example my UPS was reporting ~260W usage.

I expect some overhead over the UPS.

## What did you expect to happen
Accurate payload / data for power and current.

I believe #1278 might be related.

I don't know much about the Zigbee protocol but I am going to mess with the devices.js and see if I can sort it out.

## Debug info
Zigbee2MQTT version: 1.14.4
Adapter hardware: Conbee II
Adapter firmware version: 0x26580700 (Howlin’ Wolf ⋅ v2.5.76)

zikeji, 2020-09-16:
Couldn't for the life of me figure out what was going on until I noticed the fromZigbee electrical_measurement_power.

So my device hits the exception on line 7205 of devices.js (I switched to the dev branch), which sets some defaults instead of reading the values.

```js
            try {
                await readEletricalMeasurementPowerConverterAttributes(endpoint);
            } catch (exception) {
                console.log('hit exception');
                // For some this fails so set manually
                // https://github.com/Koenkk/zigbee2mqtt/issues/3575
                endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {
                    acCurrentDivisor: 10,
                    acCurrentMultiplier: 1,
                    powerMultiplier: 1,
                    powerDivisor: 10,
                });
            }
```

Note powerMultiplier and powerDivisor.

The electrical_measurement_power getFactor function uses the key to search for the attribute value - the key supplied in the lookup table is "acPower", not "power". Once I updated lines 7212 and 7213 to prepend with "ac" (e.g. "acPowerMultiplier") and reconfigured the device my power measurements were correct.

The current measurement is still wrong. I believe this device might be on older firmware which is why it's unable to obtain those values (?). I'm not knowledgeable on the Zigbee protocol though. But I get the impression that it functions correctly for some people, so changing the acCurrentDivisor to 1000 (which is what I had to do) is probably not ideal.

But now that all is said and done I get the correct reporting.

```
{"current":2.61,"linkquality":61,"power":287.7,"state":"ON","voltage":118}
```

zikeji, 2020-09-16:
Oh. #3575. So that the issues get properly linked.

@Koenkk I want to make a PR for the typo correction (however I won't be hurt if you change it without a PR), but I'm not sure what to do about the current value. Since I had to change the divisor to 1000 to get it to report correctly (assuming that value is supposed to report as amps?).

Koenkk, 2020-09-16:
@zikeji is indeed a typo, in #3575 current is shown in mA, however this should be in A please make a pr.

zikeji, 2020-09-16:
PR submitted. Thanks for the quick response! Glad I was able to figure it out.

closed 2020-10-24 by github-actions[bot]

nomadmd1, 2021-11-09:
The reported power and current are still 10 fold what are they supposed to be for Centralite 4257050-RZHAC. The new https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/centralite.js does not include corrections made by zikeji. I tried to update manually /mnt/data/docker/overlay2/50358e35a1ab0a60c6ddf38d2bd13b46a1224a5740ed33f8ebbdd69d7b1c7750/diff/app/node_modules/zigbee-herdsman-converters/devices.js to`acCurrentDivisor: 1000, acCurrentMultiplier: 1, acPowerMultiplier: 1, acPowerDivisor: 10` then removing and adding the outlet but the issue remains.
Currently using 1.22.0-dev

Zipties, 2022-06-01:
I am also still seeing this issue on the 4257050-RZHAC

ShadowVlican, 2022-12-04:
also still seeing this problem on my 4257050-RZHAC, but i'm on the main official branch (home assistant add-on)... it used to report correctly once upon a time...

Please-Send-Help, 2022-12-16:
This is still happening in 1.28.4 with 4257050-RZHAC outlets

Voltage is correct but power is off by factor of 10

Koenkk, 2022-12-17:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/b0d110314059fc4defc74620650580c5

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed
- if not, provide me the z2m log when clicking reconfigure in the z2m frontend (yellow refresh button)

Please-Send-Help, 2022-12-17:
Thanks Koenkk. I did change the following line:

acCurrentDivisor: 10, acCurrentMultiplier: 1, powerMultiplier: 1, powerDivisor: 100});

to

acCurrentDivisor: 1000, acCurrentMultiplier: 1, powerMultiplier: 1, powerDivisor: 10});

Now watts and amps are showing correctly. Thanks!

[image]

Koenkk, 2022-12-18:
Great, fix will be included in the next release!