Koenkk/zigbee2mqtt #4381: Publish device list after force remove

Hacker-CB, 2020-09-16:
## What happened

I force removed device: 

```
Zigbee2MQTT:info  2020-09-16 16:15:26: Force removing '0x000d6f0014f34ec9'
Zigbee2MQTT:debug 2020-09-16 16:15:26: Clearing Home Assistant discovery topic for '0x000d6f0014f34ec9'
Zigbee2MQTT:info  2020-09-16 16:15:26: Successfully force_removed 0x000d6f0014f34ec9
Zigbee2MQTT:info  2020-09-16 16:15:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"0x000d6f0014f34ec9","type":"device_force_removed"}'
```

## What did you expect to happen

New device list (without deleted one) should be published to mqtt

## How to reproduce it (minimal and precise)

Just force remove device.

## Debug info
Zigbee2MQTT version: 0.13.7
Adapter hardware: CC2538 + CC2592
Adapter firmware version: JetHome

Koenkk, 2020-09-16:
This is indeed not done for the legacy api (will not be updated anymore), when removing through the [new api]( https://github.com/Koenkk/zigbee2mqtt/issues/3281) this is already done

closed 2020-09-16 by Koenkk

Hacker-CB, 2020-09-16:
unfortunatelly, zigbee2mqttassistant uses old api :(

Koenkk, 2020-09-16:
You can try https://github.com/Koenkk/zigbee2mqtt/issues/4266 as an alternative