Koenkk/zigbee2mqtt #4382: Interview Fails on 3321-S & 3325-S

zikeji, 2020-09-16:
# Bug Report
## What happened

When trying to pair my door multi sensors and motion multisensor the interview fails.

These two devices:
https://www.zigbee2mqtt.io/devices/3321-S.html
https://www.zigbee2mqtt.io/devices/3325-S.html

## What did you expect to happen

Successfully pair.

## How to reproduce it (minimal and precise)

Conbee II USB adapter, both the above devices.

## Debug Info
Zigbee2MQTT version: 1.14.4
Adapter hardware: Conbee II
Adapter firmware version: 0x26580700 (Howlin’ Wolf ⋅ v2.5.76)

## Comments

Log file here: https://0x0.st/iI62.txt (had to upload it elsewhere as it was too large - the logs are spammed by my power adapter).

I've tried everything from the FAQ - what is relevant at least. I tried pairing while a few feet from the controller and no obstructions. Tried a new battery. Tried repairing multiple times. Just doesn't want to go through. I went through some of the other issues and saw advice like forcing it to stay awake - I tried that a few times with no success. Not sure how to proceed.

zikeji, 2020-09-16:
All three devices are still visible (though offline, as my ST Hub is offline and they've been reset numerous times) in the SmartThings IDE.  The only thing in there that seems relevant is the "raw description" `	01 0104 0402 00 08 0000 0001 0003 0020 0402 0500 0B05 FC02 01 0019`.

Koenkk, 2020-09-16:
Try to keep the device while pairing, it seems to fall asleep during it. You can do this by pressing a button on it.

zikeji, 2020-09-16:
> Try to keep the device while pairing, it seems to fall asleep during it. You can do this by pressing a button on it.

Unfortunately that didn't appear to help. Tried on all three, various approaches - reset, press button rapidly, press button slowly, press button even more slowly, don't press it at all, have it 5ft from the conbee 2, have it inches, have it 10ft, etc. Just unable to get it to finish. Plugged back in the ST hub and it paired first attempt.

zikeji, 2020-09-17:
```
Sep 17 10:08:02 r210ii npm[19104]: 2020-09-17T10:08:02.357Z zigbee-herdsman:controller:endpoint Read 0x000d6f000ba73908/1 genBasic(["appVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
Sep 17 10:08:02 r210ii npm[19104]: 2020-09-17T10:08:02.357Z zigbee-herdsman:controller:device:log Failed to read attribute 'applicationVersion' from endpoint '1' (Error: Read 0x000d6f000ba73908/1 genBasic(["appVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Status 'UNSUPPORTED_ATTRIBUTE'))
Sep 17 10:08:02 r210ii npm[19104]: 2020-09-17T10:08:02.357Z zigbee-herdsman:controller:endpoint Read 0x000d6f000ba73908/1 genBasic(["stackVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
```

Could unsupported attribute be related?

Koenkk, 2020-09-17:
- I don't see the interview failed message in your log
- Does it also give a `UNSUPPORTED_ATTRIBUTE` for the modelID?
- Would you be able to sniff the traffic when pairing? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html
- When providing debug logging, dont' use `DEBUG=*` but use `DEBUG=zigbee-herdsman*`

zikeji, 2020-09-17:
Looks like I can't sniff on my ConBee2. Might just have to buy a CC2531.

zikeji, 2020-09-17:
Interesting, this time around when debugging it was able to pick up the manufacturer.

I disconnected the smart plug to remove it's constant output from the logs and cut out all the MQTT received / published messages. I'm taking a look myself but I'm unfamiliar with the protocol.
https://hastebin.com/xiyazoyoqu.txt

zikeji, 2020-09-18:
I've been digging around in the spec and code and I see the device (or the library is misinterpreting) the command identifier as 255. I can't find anywhere in the spec what the identifier would be.

zikeji, 2020-09-18:
Hmm. I'm not really sure what to make of all this. Seems like being able to run a sniffer would be the best bet.

When I try to pair the device it essentially gets looped pairing it with the following types of failures:
```
2020-09-18T05:24:47.127Z zigbee-herdsman:deconz:adapter could not parse zclFrame: RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 7. Received 8
2020-09-18T05:24:47.174Z zigbee-herdsman:deconz:adapter could not parse zclFrame: Error: Unsupported frameType '2'
2020-09-18T05:24:48.979Z zigbee-herdsman:deconz:adapter could not parse zclFrame: Error: Cluster 'genOnOff' has no command response '1'
2020-09-18T05:24:58.894Z zigbee-herdsman:deconz:adapter could not parse zclFrame: Error: Unsupported frameType '3'
2020-09-18T05:22:49.814Z zigbee-herdsman:deconz:adapter could not parse zclFrame: TypeError: Cannot read property 'parseStrategy' of undefined
```

zikeji, 2020-09-18:
One of the issues I ran into was during the parsePayloadGlobal command - it was reading USE_DATA_TYPE and the data type being returned is "5" which maps to nothing (so undefined) in dataType.d.ts. I manually added a check to see if it's undefined and set it to type 255 (unknown) - that gets that frame past the zcfFrame.fromBuffer step. I don't know if that has made a difference though.

zikeji, 2020-09-18:
It is apparent now though that the device is being routed through my IKEA bulb when pairing, not directly to the coordinator. I had hoped it might be that so I turned off all the routers and tried again without them and with just the coordinator. No such luck!

Alot of frames fail because the command type is not in the cluster and the ZCL frame type is unknown. I'm starting to wonder if perhaps the latest FW on these devices renders them incompatible somehow. All in all I'm pretty mystified (granted this is all entirely new to me). I think the best route is to just throw them back on ST and use them there with the ST integration until they die and I replace them.

zikeji, 2020-09-18:
I take that back. I was inspired to try out the Deconz software instead of zigbee2mqtt and they were discovered without issue. Not sure what to make of that.

zikeji, 2020-09-18:
Okay well I setup my old network again and now they're paired and working best I can tell. I give up.

Koenkk, 2020-09-18:
@chrishae could you take a look at the errors in: https://github.com/Koenkk/zigbee2mqtt/issues/4382#issuecomment-694661785 ?

ChrisHae, 2020-09-22:
It is very likely these issues are related to some code I fixed last week. It was possible that the deconz adapter accepted the wrong incoming data indications as responses to data requests send to the devices. I have uploaded the code at sept. 15. to the zigbee-herdsman repository. I'm not sure if you already have the new code.

zikeji, 2020-09-22:
I was on the dev branch of zigbee2mqtt, running zigbee-herdsman 0.13.7 according to the package.json and shrinkwrap.

0.13.7 was released on the 16th and based on the commit chain I was already running your code from 9/15.

I assume you're referring to this commit: https://github.com/Koenkk/zigbee-herdsman/commit/120eb4e2c3e5b75a5b58b4ecbd7a68af3b8bf2ac

Further verifying the code in my node_modules I was running all the code from your 9/15 commit. Further verified by my logfile in my previous comment here you updated the debug line "sendZclFrameToEndpoint - message send" and added the transSeq Nr., doing a find in the log you'll find the lines using your new debug output.

https://github.com/Koenkk/zigbee2mqtt/issues/4382#issuecomment-694463235

ChrisHae, 2020-09-22:
You are right, it seems the code is not working as expected, yet. I will investigate further.

zikeji, 2020-09-22:
No problem, let me know if I can help in anyway. This sort of thing is to be expected and I'm happy to help iron out the kinks, even if in a small way. Thanks for taking a look and your work so far!

ChrisHae, 2020-09-22:
I have uploaded a hotfix to zigbee-herdsman. It seems to get rid of the error during my first tests. Please feel free to test it also.

Koenkk, 2020-09-22:
@ChrisHae just integrated this change into the dev branch

zikeji, 2020-09-22:
I'll test it later but I'm curious how it's supposed to help. All I see is logic for conditionally parsing the zcl frame. When the zcl frame parse fails if just shows a debug message but the frame is still null so the rest of the function handles it. So the end result doesn't change at all.

ChrisHae, 2020-09-22:
I added the parseZclFrame function last week to get the transSeqNr. that is in the header. But it does not work when the profileId is 0x00. These frames are handled differently. If a frame is received with profile 0x104 then the parsing should work.

closed 2020-10-31 by github-actions[bot]