Koenkk/zigbee2mqtt #4383: new device - Somgoms 2 gang switch  ['kjintbl']

hometheater13, 2020-09-16:
Hi 

need assistant to add new device - Somgoms 2gang switch 

zigbeeModel: ['kjintbl'],
        model: 'ZSTY-SM-21ZG-US-W',
        vendor: 'Somgoms',
        description: '2 gang switch',
        supports: 'on/off',

fromzigbee?
## Information about the device + link
..link for device from ALI

https://www.aliexpress.com/item/4000607858705.html?spm=a2g0s.9042311.0.0.16594c4d5ARgbV

## data/database.db entry of the device
..data from debug log:
type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer' ,dp 257 ,dp 258

Zigbee2MQTT:debug 2020-09-16 15:09:55: Received Zigbee message from 'kitchen_dinning', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[1],"type":"Buffer"},"dp":257,"fn":0,"status":0,"transid":148}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-09-16 15:09:55: No converter available for 'ZSTY-SM-21ZG-US-W' with cluster 'manuSpecificTuyaDimmer' and type 'commandSetDataResponse' and data '{"data":{"data":[1],"type":"Buffer"},"dp":257,"fn":0,"status":0,"transid":148}'
Zigbee2MQTT:debug 2020-09-16 15:09:58: Received Zigbee message from 'kitchen_dinning', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[0],"type":"Buffer"},"dp":257,"fn":0,"status":0,"transid":149}' from endpoint 1 with groupID 0

Zigbee2MQTT:debug 2020-09-16 15:12:09: Received Zigbee message from 'kitchen_dinning', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[1],"type":"Buffer"},"dp":258,"fn":0,"status":0,"transid":150}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-09-16 15:12:09: No converter available for 'ZSTY-SM-21ZG-US-W' with cluster 'manuSpecificTuyaDimmer' and type 'commandSetDataResponse' and data '{"data":{"data":[1],"type":"Buffer"},"dp":258,"fn":0,"status":0,"transid":150}'
Zigbee2MQTT:debug 2020-09-16 15:12:09: Received Zigbee message from 'kitchen_dinning', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[0],"type":"Buffer"},"dp":258,"fn":0,"status":0,"transid":151}' from endpoint 1 with groupID 0

switch","friendly_name":"kitchen_dinning","hardwareVersion":1,"ieeeAddr":"0xec1bbdfffe0c15d1","lastSeen":1599150465719,"manufacturerID":0,"manufacturerName":"_TYST11_nkjintbl","model":"ZSTY-SM-21ZG-US-W","modelID":"kjintbl\u0000","networkAddress":15805,"powerSource":"Mains (single phase)","type":"Router","vendor":"Somgoms"}],"type":"devices"}

please add support for this device  . update if need more details

Koenkk, 2020-09-16:
Does it work when changing https://github.com/Koenkk/zigbee-herdsman-converters/blob/ca8fd26e06ee0b3e1f2e1541ce3a23f903521242/devices.js#L1198 to `zigbeeModel: ['qnazj70','kjintbl'],`

hometheater13, 2020-09-17:
HI
yes it works  .
but inside devices.js have norklmes and little bit different conf 
changed to 

{
        zigbeeModel: ['qnazj70', 'kjintbl'],
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_nkjintbl'}],
        model: 'TS0601_switch',
        vendor: 'TuYa',
        description: '1, 2, 3 or 4 gang switch',
        supports: 'on/off',
        fromZigbee: [fz.tuya_switch, fz.ignore_basic_report, fz.tuya_switch2],
        toZigbee: [tz.tuya_switch_state],
        meta: {configureKey: 1, multiEndpoint: true},
        whiteLabel: [
            {vendor: 'Norklmes', model: 'MKS-CM-W5'},
            {vendor: 'Somgoms', model: 'ZSQB-SMB-ZB'},
        ],
        configure: async (device, coordinatorEndpoint) => {
            await bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        },
        endpoint: (device) => {
            // Endpoint selection is made in tuya_switch_state
            return {'l1': 1, 'l2': 1, 'l3': 1, 'l4': 1};
        },
    },

can u update  .thx

Koenkk, 2020-09-17:
Added!

Changes will be available in the latest dev branch tomorrow (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-09-17 by Koenkk

hometheater13, 2020-09-19:
HI

any updates about this ? when the dev brach will be updated ?

thx

Koenkk, 2020-09-19:
Will be done in a few hours.

hometheater13, 2020-09-19:
Great . thx

hometheater13, 2020-09-23:
Hi

Any update about that ,test it today still didn't change .

Thx
Br
Emil

Koenkk, 2020-09-25:
- Are you sure you updated to the latest dev branch?
- If yes, can you share your `data/database.db` entry of this device?

hometheater13, 2020-09-29:
Hi

reinstall it again and it working  :). sorry .

thx

BR
Emil