Koenkk/zigbee2mqtt #4386: CC2530+CC2591 issues

hspjanssen, 2020-09-16:
Hello,

I am using Zigbee2MQTT for about 6 months now, I only had one issue in May, at that time I reflashed my coordinator again with the firmware below (already using that firmware) and it worked until last week without any issues. I did not make any chances for the last two/three weeks, it suddenly stopped working.

I am using a CC2530+CC2591 with the following setup:
- connected to Raspberry Pi
- latest Zigbee2MQTT software
- USB extension cable between CC2530 and Raspberry Pi
- running on CC2530_CC2591_SOURCE_ROUTING_20190619 firmware
- 34 devices connected with about 10-15 routers (power plugs/light bulbs/CC2530 programmed as router)
- using Domoticz with Zigbee2Mqtt plugin

I am getting a lot of these kind of messages in the log file:
error 2020-09-16 21:48:35: Publish ‘set’ ‘state’ to ‘0x000d6ffffe550372’ failed: ‘Error: Command 0x000d6ffffe550372/1 genOnOff.off({}, {“timeout”:10000,“disableResponse”:false,“disableDefaultResponse”:false,“direction”:0,“srcEndpoint”:null,“reservedBits”:0,“manufacturerCode”:null,“transactionSequenceNumber”:null}) failed (Data request failed with error: ‘No network route’ (205))’
info 2020-09-16 21:48:35: MQTT publish: topic ‘zigbee2mqtt/bridge/log’, payload ‘{“message”:“Publish ‘set’ ‘state’ to ‘0x000d6ffffe550372’ failed: ‘Error: Command 0x000d6ffffe550372/1 genOnOff.off({}, {“timeout”:10000,“disableResponse”:false,“disableDefaultResponse”:false,“direction”:0,“srcEndpoint”:null,“reservedBits”:0,“manufacturerCode”:null,“transactionSequenceNumber”:null}) failed (Data request failed with error: ‘No network route’ (205))’”,“meta”:{“friendly_name”:“0x000d6ffffe550372”},“type”:“zigbee_publish_error”}’

I already trying reflashing the CC2530, without any positive result.

Some questions:
- what can I do to get it working (temporary) again without configuring Domoticz?
- on this page “https://www.zigbee2mqtt.io/information/supported_adapters.html” I saw that the capacity of the 2530 is 30/40 devices maximum. If i need to switch to an other adapter from that website, which one is the best to choose at this moment with 100+ capacity? I assume that I need to repair all the devices, but it is not required to configure everything again in Domoticz (every device will receive same ID after repairing?)
- is it an option to update to firmware 3 and should it fix this?

All options/suggestions are welcome, thanks in advance.

Koenkk, 2020-09-17:
What zigbee2mqtt version are you using?

hspjanssen, 2020-09-17:
1.14.4, see below...

info  2020-09-17 19:20:30: Starting Zigbee2MQTT version 1.14.4 (commit #166f1a8)
info  2020-09-17 19:20:30: Starting zigbee-herdsman...
info  2020-09-17 19:20:44: zigbee-herdsman started

Koenkk, 2020-09-18:
Probably the same issue as #4307, switching to the dev branch should fix the issue: https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

hspjanssen, 2020-09-18:
Update to dev version, no difference.....still a lot of 'No network route' (205) messages in the log.

Any other idea?

Thnx

Koenkk, 2020-09-19:
@hspjanssen could it be an interference issue, e.g.: https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html#reduce-wifi-interference-by-changing-the-zigbee-channel

hspjanssen, 2020-09-19:
Did not make any specific channel configuration regarding Zigbee in configuration.yaml, Zigbee should be on channel 11 then, correct? which corresponds with WiFi channel 1. I am measuring no wireless network on channel 1 till 4. 

I changed the 2530 for a different 2530 unit (reflashed with CC2530_CC2591_SOURCE_ROUTING_20190619 firmware), makes no difference. Hereby the begin part, and latest part of the log with some errors:

info  2020-09-19 14:23:15: Starting Zigbee2MQTT version 1.14.4-dev (commit #c0faae7)
info  2020-09-19 14:23:15: Starting zigbee-herdsman...
info  2020-09-19 14:23:29: zigbee-herdsman started
info  2020-09-19 14:23:29: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190619,"transportrev":2},"type":"zStack12"}'
---
info  2020-09-19 14:31:57: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c78dfe8', payload '{"battery":100,"illuminance":33114,"illuminance_lux":2047,"linkquality":39,"voltage":3000}'
error 2020-09-19 14:32:01: Publish 'set' 'state' to '0x000d6ffffe15206f' failed: 'Error: Command 0x000d6ffffe15206f/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"d$
info  2020-09-19 14:32:01: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x000d6ffffe15206f' failed: 'Error: Command 0x000d6ffffe15206f/1 genOnOff.off({}, $
info  2020-09-19 14:32:02: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c78dfe8', payload '{"battery":100,"illuminance":40216,"illuminance_lux":10509,"linkquality":39,"voltage":3000}'
info  2020-09-19 14:32:07: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c78dfe8', payload '{"battery":100,"illuminance":41699,"illuminance_lux":14787,"linkquality":39,"voltage":3000}'
info  2020-09-19 14:32:12: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c78dfe8', payload '{"battery":100,"illuminance":43275,"illuminance_lux":21256,"linkquality":39,"voltage":3000}'
error 2020-09-19 14:32:13: Publish 'set' 'state' to '0xec1bbdfffe023a95' failed: 'Error: Command 0xec1bbdfffe023a95/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"d$
info  2020-09-19 14:32:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0xec1bbdfffe023a95' failed: 'Error: Command 0xec1bbdfffe023a95/1 genOnOff.off({}, $
info  2020-09-19 14:32:15: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c78d565', payload '{"battery":100,"illuminance":35499,"illuminance_lux":3546,"linkquality":20,"voltage":3100}'
info  2020-09-19 14:32:17: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c78dfe8', payload '{"battery":100,"illuminance":37429,"illuminance_lux":5531,"linkquality":39,"voltage":3000}'
error 2020-09-19 14:32:22: Publish 'set' 'state' to '0x000d6ffffe15206f' failed: 'Error: Command 0x000d6ffffe15206f/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"d$
info  2020-09-19 14:32:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x000d6ffffe15206f' failed: 'Error: Command 0x000d6ffffe15206f/1 genOnOff.off({}, $
info  2020-09-19 14:32:22: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c78dfe8', payload '{"battery":100,"illuminance":36843,"illuminance_lux":4833,"linkquality":39,"voltage":3000}'
info  2020-09-19 14:32:27: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c78dfe8', payload '{"battery":100,"illuminance":42890,"illuminance_lux":19453,"linkquality":39,"voltage":3000}'
info  2020-09-19 14:32:32: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c78dfe8', payload '{"battery":100,"illuminance":37779,"illuminance_lux":5996,"linkquality":39,"voltage":3000}'
error 2020-09-19 14:32:34: Publish 'set' 'state' to '0xec1bbdfffe0234f8' failed: 'Error: Command 0xec1bbdfffe0234f8/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"d$
info  2020-09-19 14:32:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0xec1bbdfffe0234f8' failed: 'Error: Command 0xec1bbdfffe0234f8/1 genOnOff.off({}, $
info  2020-09-19 14:32:37: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c78dfe8', payload '{"battery":100,"illuminance":44284,"illuminance_lux":26815,"linkquality":39,"voltage":3000}'
error 2020-09-19 14:32:42: Publish 'set' 'state' to '0xec1bbdfffe9ec86a' failed: 'Error: Command 0xec1bbdfffe9ec86a/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"d$
info  2020-09-19 14:32:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0xec1bbdfffe9ec86a' failed: 'Error: Command 0xec1bbdfffe9ec86a/1 genOnOff.off({}, $
info  2020-09-19 14:32:42: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c78dfe8', payload '{"battery":100,"illuminance":41342,"illuminance_lux":13620,"linkquality":39,"voltage":3000}'
info  2020-09-19 14:32:47: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c78dfe8', payload '{"battery":100,"illuminance":42676,"illuminance_lux":18517,"linkquality":39,"voltage":3000}'
info  2020-09-19 14:32:52: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c78dfe8', payload '{"battery":100,"illuminance":40985,"illuminance_lux":12545,"linkquality":39,"voltage":3000}'
error 2020-09-19 14:32:54: Publish 'set' 'state' to '0xec1bbdfffe9ebb25' failed: 'Error: Command 0xec1bbdfffe9ebb25/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"d$
[... cut ...]
info  2020-09-19 14:33:17: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c78dfe8', payload '{"battery":100,"illuminance":44460,"illuminance_lux":27924,"linkquality":39,"voltage":3000}'
info  2020-09-19 14:33:22: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c78dfe8', payload '{"battery":100,"illuminance":44221,"illuminance_lux":26429,"linkquality":39,"voltage":3000}'
info  2020-09-19 14:33:27: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c78dfe8', payload '{"battery":100,"illuminance":44510,"illuminance_lux":28248,"linkquality":39,"voltage":3000}'

Koenkk, 2020-09-19:
Does the no network error happen for all devices? Also when a device is really close to the CC2530?

hspjanssen, 2020-09-19:
Will check that.

Meanwhile I tested with the 2530 on an Intel NUC instead on the Pi. Complete fresh install, copy the data directory from the Pi to the Intel NUC and attached the 2530, same results.

hspjanssen, 2020-09-20:
It does not happen with all devices, looks with about 6 of them.

Currently I do not see the message No network route anymore, but I receive two other messages, see below.

1.
---
info  2020-09-20 19:57:23: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c78dfe8', payload '{"battery":100,"illuminance":9543,"illuminance_lux":8,"linkquality":49,"voltage":3100}'
info  2020-09-20 19:57:23: Configuring '0x04cf8cdf3c78dfe8'
info  2020-09-20 19:57:28: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c78dfe8', payload '{"battery":100,"illuminance":10001,"illuminance_lux":9,"linkquality":46,"voltage":3100}'
error 2020-09-20 19:57:29: Failed to configure '0x04cf8cdf3c78d565', attempt 1 (Error: Bind 0x04cf8cdf3c78d565/1 genPowerCfg from '0x00124b001b7d96fc/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))
error 2020-09-20 19:57:33: Failed to configure '0x04cf8cdf3c78dfe8', attempt 1 (Error: Bind 0x04cf8cdf3c78dfe8/1 genPowerCfg from '0x00124b001b7d96fc/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))
info  2020-09-20 19:57:52: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c776dcd', payload '{"consumption":0.31,"current":0,"linkquality":46,"power":0,"state":"OFF","temperature":23,"voltage":226}'
info  2020-09-20 19:58:25: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c78d565', payload '{"battery":100,"illuminance":6021,"illuminance_lux":3,"linkquality":57,"voltage":3100}'
info  2020-09-20 19:58:25: Configuring '0x04cf8cdf3c78d565'
error 2020-09-20 19:58:35: Failed to configure '0x04cf8cdf3c78d565', attempt 2 (Error: Bind 0x04cf8cdf3c78d565/1 genPowerCfg from '0x00124b001b7d96fc/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)

2.
--
info  2020-09-20 19:56:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'brightness' to '0xec1bbdfffea6ed17' failed: 'Error: Command 0xec1bbdfffea6ed17/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":204,\"transtime\":0}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'MAC no ack' (233))'","meta":{"friendly_name":"0xec1bbdfffea6ed17"},"type":"zigbee_publish_error"}'
error 2020-09-20 19:56:33: Publish 'set' 'brightness' to '0xec1bbdfffe023a95' failed: 'Error: Command 0xec1bbdfffe023a95/1 genLevelCtrl.moveToLevelWithOnOff({"level":204,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC no ack' (233))'
info  2020-09-20 19:56:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'brightness' to '0xec1bbdfffe023a95' failed: 'Error: Command 0xec1bbdfffe023a95/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":204,\"transtime\":0}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'MAC no ack' (233))'","meta":{"friendly_name":"0xec1bbdfffe023a95"},"type":"zigbee_publish_error"}'

Koenkk, 2020-09-21:
Error 1: make sure to wakeup the device when configuring happens (on zigbee2mqtt startup) by pressing a button on it.
Error 2: looks like a connectivity issue (out of range or interference)

Jolly-Roger82, 2020-09-23:
I'm having the same issues, tried stable and edge version of the Homeassistant addon, deleted the database, re-paired the routers but to no avail. Ever since the HA Supervisor update my otherwise perfectly working mesh network is very unstable. 

If I restart the Z2M bridge it won't recognize my CC2531. ill have to re-plug the CC2531 to be able to start herdsman.

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
./run.sh: line 17: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
[Info] Socat is DISABLED and not started
2020-09-23T09:05:38: PM2 log: Launching in no daemon mode
2020-09-23T09:05:38: PM2 log: App [npm:0] starting in -fork mode-
2020-09-23T09:05:38: PM2 log: App [npm:0] online
> zigbee2mqtt@1.14.4 start /zigbee2mqtt-1.14.4
> node index.js
Zigbee2MQTT:info  2020-09-23 09:05:44: Logging to console and directory: '/share/zigbee2mqtt/log/2020-09-23.09-05-44' filename: log.txt
Zigbee2MQTT:info  2020-09-23 09:05:45: Starting Zigbee2MQTT version 1.14.4 (commit #1.14.4)
Zigbee2MQTT:info  2020-09-23 09:05:45: Starting zigbee-herdsman...
Zigbee2MQTT:error 2020-09-23 09:06:27: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-09-23 09:06:27: Failed to start zigbee
Zigbee2MQTT:error 2020-09-23 09:06:27: Exiting...
Zigbee2MQTT:error 2020-09-23 09:06:27: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.14.4/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:90:31)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.14.4/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)

hspjanssen, 2020-09-30:
My Zigbee network still did respond, but very very very slow. I disabled some events in Domoticz to make sure the Zigbee network was as "clear" as possible, now I can simply turn on a switch. Very strange, because I am using those events for months, without changing. I still see issues in the Zigbee log btw......

Koenkk, 2020-09-30:
There is a stability issue in 1.14.4, try 1.15.0 which will be released tomorrow this should fix it.

closed 2020-11-07 by github-actions[bot]