Koenkk/zigbee2mqtt #4390: Xiaomi WXKG01LM publishes only first action

AlexeyHimki, 2020-09-17:
When I press first time button WXKG01LM it publishes MQTT topic with payload "action":"single". But when I press next time after 1 second - it doesn't. All next clicks with delay 1s-5s will not publish anything. And only after more 10 seconds delay it publishes MQTT topic. Same things with Xiaomi MFKZQ01LM rotate action.

Here is my log https://pastebin.com/pz03fzVG
2020-09-17 22:59:01 First press - OK
2020-09-17 22:59:02 Second press - nothing
2020-09-17 22:59:03 Third - nothing
and so on
...
2020-09-17 22:59:40 Press after long delay - OK

Koenkk, 2020-09-18:
When you click the button, do you immediately see `Received Zigbee message from 'kidsroom.button', type 'attributeReport'....` in the log?

iarema7, 2020-09-18:
after last update having problems with xiaomi sensors too , they worked perfect before
Zigbee2MQTT:debug 2020-09-18 16:47:07: Received Zigbee message from 'Restroom Temp', type 'attributeReport', cluster 'msPressureMeasurement', data '{"16":10166,"20":-1,"measuredValue":1016}' from endpoint 1 with groupID 0
Zigbee2MQTT:error 2020-09-18 16:47:07: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at /zigbee2mqtt-1.14.4/lib/extension/receive.js:120:42
    at Array.filter (<anonymous>)
    at Receive.onZigbeeEvent (/zigbee2mqtt-1.14.4/lib/extension/receive.js:119:65)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.14.4/lib/controller.js:365:44)
    at Controller.onZigbeeEvent (/zigbee2mqtt-1.14.4/lib/controller.js:254:14)
    at Zigbee.emit (events.js:315:20)
    at Controller.<anonymous> (/zigbee2mqtt-1.14.4/lib/zigbee.js:68:52)
    at Controller.emit (events.js:315:20)
    at Controller.<anonymous> (/zigbee2mqtt-1.14.4/node_modules/zigbee-herdsman/dist/controller/controller.js:523:22)
    at Generator.next (<anonymous>))
Zigbee2MQTT:debug 2020-09-18 16:47:14: Received Zigbee message from 'Light stairs T', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":30596}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-09-18 16:47:14: MQTT publish: topic 'zigbee2mqtt/Light stairs T', payload '{"battery":100,"illuminance":30596,"illuminance_lux":1147,"linkquality":84,"voltage":3100}'

Zigbee2MQTT:debug 2020-09-18 16:44:48: Received Zigbee message from 'MQTT Button', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:error 2020-09-18 16:44:48: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at /zigbee2mqtt-1.14.4/lib/extension/receive.js:120:42
    at Array.filter (<anonymous>)
    at Receive.onZigbeeEvent (/zigbee2mqtt-1.14.4/lib/extension/receive.js:119:65)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.14.4/lib/controller.js:365:44)
    at Controller.onZigbeeEvent (/zigbee2mqtt-1.14.4/lib/controller.js:254:14)
    at Zigbee.emit (events.js:315:20)
    at Controller.<anonymous> (/zigbee2mqtt-1.14.4/lib/zigbee.js:68:52)
    at Controller.emit (events.js:315:20)
    at Controller.<anonymous> (/zigbee2mqtt-1.14.4/node_modules/zigbee-herdsman/dist/controller/controller.js:523:22)
    at Generator.next (<anonymous>))
Zigbee2MQTT:debug 2020-09-18 16:44:51: Received Zigbee message from 'MQTT Button', type 'attributeReport', cluster 'genOnOff', data '{"32768":2}' from endpoint 1 with groupID 0
Zigbee2MQTT:error 2020-09-18 16:44:51: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at /zigbee2mqtt-1.14.4/lib/extension/receive.js:120:42
    at Array.filter (<anonymous>)
    at Receive.onZigbeeEvent (/zigbee2mqtt-1.14.4/lib/extension/receive.js:119:65)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.14.4/lib/controller.js:365:44)
    at Controller.onZigbeeEvent (/zigbee2mqtt-1.14.4/lib/controller.js:254:14)
    at Zigbee.emit (events.js:315:20)
    at Controller.<anonymous> (/zigbee2mqtt-1.14.4/lib/zigbee.js:68:52)
    at Controller.emit (events.js:315:20)
    at Controller.<anonymous> (/zigbee2mqtt-1.14.4/node_modules/zigbee-herdsman/dist/controller/controller.js:523:22)
    at Generator.next (<anonymous>))

using usb-Silicon_Labs_slae.sh_cc2652rb_stick

AlexeyHimki, 2020-09-19:
No. I didn't. And that's the problem

Koenkk, 2020-09-19:
This would mean the device sends no Zigbee messages, would you be able to sniff the traffic of this? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

closed 2020-10-27 by github-actions[bot]