Koenkk/zigbee2mqtt #4392: Xiaomi QBKG11LM destroys the network if there are other routers in it

Rupunx, 2020-09-18:
**I have the same problem like #3944**

I have performed a series of tests and am confident that the case is related to the Xiaomi QBKG11LM switches

**My config:**

1. Latest zigbee2mqtt 1.14.4-dev 
2. сc2538 located on the USB extension cable
3. channel 25 (wifi on 1 channel)

**Devices:**

- 8 switches Xiaomi QBKG11LM - they are routers,
- more than 15 end devices,
- 3 Xiaomi ZNCZ02LM power plug routers,
- 3 Blitzwolf shp-13 power plug routers

### Testing
**Test 1. Initial work.**
Blitzwolf shp-13 was not available (not bought yet at that time)

From time to time, some Xiaomi QBKG11LM switches stopped responding to requests. Once a week or two, I re-paired them
Through analysis, I found out that these were devices that are connected according to the following **_scheme_**:
Coordinator сc2538 (or cc2531 on previous use) -> router switch Xiaomi QBKG11LM -> router power plug Xiaomi ZNCZ02LM -> end device (sensor / button)
**_Key point:_** the route from power plug Xiaomi ZNCZ02LM goes through the Xiaomi switch QBKG11LM
**Conclusion**: switch Xiaomi QBKG11LM breaks the signal from routers power plug Xiaomi ZNCZ02LM

**Test 2. Removing power plug Xiaomi ZNCZ02LM**
I removed all power plug Xiaomi ZNCZ02LM from the network
**Scheme**:
Coordinator -> router switch Xiaomi QBKG11LM -> end device (sensor / button)
**Conclusion**: The network worked steadily for more than a month and I continued the series of tests

**Test 3. Now (09/17/2020). Adding a Blitzwolf shp-13 power plug router**
I restored the network as in test 1, replacing the Xiaomi power plugs with Blitzwolf power plugs
**Scheme**:
Coordinator -> router switch Xiaomi QBKG11LM -> router power plug Blitzwolf shp-13 -> end device (sensor / button)
**Now**: everything returned as it was in test 1. The first signal from the end devices stopped reaching, the switches were no longer controlled (1 day after the start of the test)

**What errors are in the log now:**
1) There are such errors (this is the error of the Xiaomi QBKG11LM switch, as in the Scheme above)
`error 2020-09-17 12:00:15: Failed to execute LQI for 'child_swtich_with_n'`

2) An error that occurs when the device does not work (does not accept the command). Those. I turn on / off Xiaomi QBKG11LM via zigbee2mqtt (via Home Assistant). Xiaomi QBKG11LM does not react to the command in any way.

```
error 2020-09-17 15:55:58: Publish 'set' 'state' to 'bedr_swtich_with_n' failed: 'Error: Command 0x00158d0002335e7d / 1 genOnOff.on ({}, {"timeout": 10000, "disableResponse": false, "disableDefaultResponse": false, "direction": 0, "srcEndpoint": null, "reservedBits": 0, "manufacturerCode": null, "transactionSequenceNumber": null}) failed (Timeout - 31502 - 1 - 225 - 6 - 11 after 10000ms) '
info 2020-09-17 15:55:58: MQTT publish: topic 'zigbee2mqtt / bridge / log', payload '{"message": "Publish' set '' state 'to' bedr_swtich_with_n 'failed:' Error: Command 0x00158d0002335e7d / 1 genOnOff.on ({}, {\ "timeout \": 10000, \ "disableResponse \": false, \ "disableDefaultResponse \": false, \ "direction \": 0, \ "srcEndpoint \": null, \ "reservedBits \": 0, \ "manufacturerCode \": null, \ "transactionSequenceNumber \": null}) failed (Timeout - 31502 - 1 - 225 - 6 - 11 after 10000ms) '"," meta ": {" friendly_name ":" bedr_swtich_with_n "}," type ":" zigbee_publish_error "} '

```

## Debug info
Zigbee2MQTT version: 1.14.4-dev
Adapter hardware: CC2538
Adapter firmware version: 20200211

domnich, 2020-10-18:
Any update here ? Have similar issue with my Blitzwolf shp-13 power plug router

Rupunx, 2020-10-18:
I have a bug on the light switches. To reboot them - I need to turn it off in the electrical panel
I have now transferred all Xiaomi switches from zigbee2mqtt to Xioami gateway

domnich, 2020-10-18:
This is not Xiaomi related issue in my case as I don't have them...For me problem reproducible with Blitzwolf shp-13 power plug router and with with Tuya wall switch (model TS0012). So I think this is more common issue which doesn't depends only on Xiaomi routers

domnich, 2020-10-18:
> I have a bug on the light switches. To reboot them - I need to turn it off in the electrical panel
> I have now transferred all Xiaomi switches from zigbee2mqtt to Xioami gateway

Does your switches are routers or not ?

Rupunx, 2020-10-18:
> Does your switches are routers or not ?

Yes, they are routers

closed 2020-11-26 by github-actions[bot]