Koenkk/zigbee2mqtt #4393: Sunricher SR-ZG2868 WIP

VincentSC, 2020-09-18:
## Information about the device + link
https://www.sunricher.com/dim-cct-rgb-3-in-1-zigbee-remote-sr-zg2868.html#product_tabs_description_tabbed
https://www.sunricher.com/media/resources/manual/SR-ZG2868%20Instruction.pdf
Quote:
> SR-ZG2868 is a Dim CCT RGB 3-in-1 zigbee key remote controller that enables switch, dimming, color temperature control, RGB color control of zigbee lighting devices and works with a zigbee gateway/coordinator, it shall be added to a zigbee network and bind compatible zigbee lighting devices from zigbee controller or hub interface so that it can control the devices. This zigbee remote supports 4 groups for binding max. 30 lighting devices and enables to control single color, tunable white, RGBW and RGB+CCT zigbee lighting devices. It is a low cost zigbee remote powered by two 3VDC button batteries due to robust, reliable ZigBee PRO networking. With magnet bracket, the remote is easy to fasten.
>
> The zigbee remote can be compatible with universal zigbee gateway/coordinator and lighting devices thanks to interoperability of products with the same zigbee Wireless standard.
>
> The zigbee remote supports Touchlink commissioning to bind directly to a lighting device without a zigbee hub or Touchlink after the remote and device are both added to the zigbee hub network. 

Costs like €25.

By using `sudo journalctl -u zigbee2mqtt.service -f | grep "ZG2868\|0x000d6f00106a7cd7"` I could get a focused log (tip for the documentation).

Below is WIP!!! I need some help with 2 issues. I will send a PR/MR when done.

```
    {
        zigbeeModel: ['ZG2868A'],
        model: 'SR-ZG2868',
        vendor: 'Sunricher',
        description: 'RGB CCT DIM 3-in-1 Zigbee Remote',
        supports: 'action',
        fromZigbee: [
                fz.command_off, fz.command_on, fz.command_recall, fz.command_step, fz.command_move,
                fz...., fz....
        ],
        toZigbee: [],
    },
```
Till now I could use the standard convertors, but some are needed.

Got two one problem I cannot solve myself. I see is that two things don't generate any messages:
- "press&hold" for saving colors
- touching the color-wheel (which is the main reason I got it)

How can I see these messages? Or is it a compatibility issue?

Koenkk, 2020-09-18:
Try enabling herdsman debug logging and look for any: `Skipping command 'SOME_COMMAND_NAME' because it is missing from the lookup` when executing the action.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

VincentSC, 2020-09-18:
okidoki

VincentSC, 2020-11-30:
Note to self. 

Looks exactly the same as https://www.zigbee2mqtt.io/devices/4512706.html

```    {
        zigbeeModel: ['4512706'],
        model: '4512706',
        vendor: 'Namron',
        description: 'Remote control',
        supports: 'action',
        fromZigbee: [
            fz.command_on, fz.command_off, fz.command_step, fz.command_step_color_temperature,
            fz.command_recall, fz.command_move_to_color_temp, fz.battery,
        ],
        exposes: [e.battery(), e.action([
            'on', 'off', 'brightness_step_up', 'brightness_step_down', 'color_temperature_step_up', 'color_temperature_step_down',
            'recall_*', 'color_temperature_move',
        ])],
        toZigbee: [],
        meta: {multiEndpoint: true},
        endpoint: (device) => {
            return {l1: 1, l2: 2, l3: 3, l4: 4};
        },
    },
```

Eutychus69, 2021-02-07:
I created a definition like that:
`    {
        zigbeeModel: ['ZG2868A'],
        model: 'SR-ZG2868',
        vendor: 'Sunricher',
        description: 'Remote control',
        supports: 'action',
        fromZigbee: [
            fz.command_on, fz.command_off, fz.command_step, fz.command_step_color_temperature,
            fz.command_recall, fz.command_move_to_color_temp, fz.battery, fz.command_move_to_saturation, fz.command_move_to_hue,
        ],
        exposes: [e.battery(), e.action([
            'on_*', 'off_*', 'brightness_step_up_*', 'brightness_step_down_*', 'color_temperature_step_up_*', 'color_temperature_step_down_*',
            'recall_*', 'color_temperature_move_*', 'saturation_move', 
        ])],
        toZigbee: [],
        meta: {multiEndpoint: true},
        endpoint: (device) => {
            return {l1: 1, l2: 2, l3: 3, l4: 4};
        },
    },`

But the device sends a command: "moveToHue" which seems not to be defined in Herdsman.

cveld, 2021-04-10:
zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":89,"manufacturerCode":null,"commandIdentifier":0},"Payload":{"hue":76,"direction":0,"transtime":0},"Command":{"ID":0,"parameters":[{"name":"hue","type":32},{"name":"direction","type":32},{"name":"transtime","type":33}],"name":"moveToHue"}},"address":24909,"endpoint":4,"linkquality":204,"groupID":19716,"wasBroadcast":true}' +744ms
  zigbee-herdsman:controller:log Skipping command 'moveToHue' because it is missing from the lookup +0ms

cveld, 2021-04-10:
I commented out `fz.command_move_to_hue` and the first error went away.

`
{ 
        zigbeeModel: ['ZG2868A'], 
        model: 'SR-ZG2868', 
        vendor: 'Sunricher', 
        description: 'Remote control', 
        supports: 'action', 
        fromZigbee: [fz.command_on, fz.command_off, fz.command_step, fz.command_step_color_temperature, fz.command_recall, fz.command_move_to_color_temp, fz.battery, fz.command_move_to_saturation, 
        //    fz.command_move_to_hue
        ], 
        exposes: [e.battery(), e.action(['on_*', 'off_*', 'brightness_step_up_*', 'brightness_step_down_*', 'color_temperature_step_up_*', 'color_temperature_step_down_*', 'recall_*', 'color_temperature_move_*', 'saturation_move'])], 
        toZigbee: [], 
        meta: { multiEndpoint: true }, 
        endpoint: (device) => { return { l1: 1, l2: 2, l3: 3, l4: 4 }; }, 
    }
`

cveld, 2021-04-10:
The remote control provides 4 lights to be targeted. You as the remote control user can select one or more of lights to be targeted. 

An interesting bug happens when having targeting more than one light.

This is when only l1 is active and being switched on and off.
```
Zigbee2MQTT:info  2021-04-10 22:32:27: MQTT publish: topic 'zigbee2mqtt/0xccccccfffe259eaa', payload '{"action":"on_l1","action_group":19713,"linkquality":208}'
Zigbee2MQTT:info  2021-04-10 22:32:27: MQTT publish: topic 'zigbee2mqtt/0xccccccfffe259eaa', payload '{"action":"off_l1","action_group":19713,"linkquality":208}'
```

Now I activate both l1 and l2 and then hit the on/off button twice:
```
Zigbee2MQTT:info  2021-04-10 22:35:34: MQTT publish: topic 'zigbee2mqtt/0xccccccfffe259eaa', payload '{"action":"on_l1","action_group":19713,"linkquality":208}'
Zigbee2MQTT:info  2021-04-10 22:35:34: MQTT publish: topic 'zigbee2mqtt/0xccccccfffe259eaa', payload '{"action":"on_l1","action_group":19714,"linkquality":208}'
Zigbee2MQTT:info  2021-04-10 22:35:33: MQTT publish: topic 'zigbee2mqtt/0xccccccfffe259eaa', payload '{"action":"off_l1","action_group":19714,"linkquality":208}'
Zigbee2MQTT:info  2021-04-10 22:35:33: MQTT publish: topic 'zigbee2mqtt/0xccccccfffe259eaa', payload '{"action":"off_l1","action_group":19713,"linkquality":208}'
```

You see that the action group is different, but the computed action name is the same. It should be off_l1, off_l2, on_l1, on_l2. What can I change in the devices.js file to accomplish this?

cveld, 2021-04-13:
I was able to pull this debug log when long-pressing the S1 to S7 buttons:

`
 zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":46,"manufacturerCode":null,"commandIdentifier":4},"Payload":{"groupid":0,"sceneid":1},"Command":{"ID":4,"response":4,"parameters":[{"name":"groupid","type":33},{"name":"sceneid","type":32}],"name":"store"}},"address":24909,"endpoint":1,"linkquality":67,"groupID":19713,"wasBroadcast":true}' +9s
  zigbee-herdsman:controller:log Skipping command 'store' because it is missing from the lookup 
`

So this is a second command that is missing in the lookup. Where can I learn how to add a new command?

mdbraber, 2021-04-13:
Thanks for debugging this @cveld - I'm using the same remote (to use with my Sonos setup). Let me know if there's anything I can do to help. 

Btw, to make it work I needed to add: `fingerprint: [{modelID: 'ZG2868A', manufacturerName: 'Sunricher'}],` as that's how the device identifies with me.

cveld, 2021-04-14:
@mdbraber yes so I have two support questions:
1. How to add new commands? I.e. `moveToHue` and `store`. Can't find much documentation on this.
2. Assuming the remote control is just 'bugging' with sending the wrong id - how to pick the `groupid` instead of the `id` to get the action suffixed. I am trying to understand what this device structure can bring. It looks like I need to investigate `fromZigbee.js`:

```javascript
on_off: {
        cluster: 'genOnOff',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('onOff')) {
                const property = postfixWithEndpointName('state', msg, model);
                return {[property]: msg.data['onOff'] === 1 ? 'ON' : 'OFF'};
            }
        },
    },
```

I guess the logic that transforms the incoming zcl data to the actual mqtt json is found within `postfixWithEndpointName`.

You can find me on Discord if you would like to have a chat: Carl#0262.

mdbraber, 2021-04-14:
I think we first have to specificy what the (raw) inputs are when a scene button is pressed with multiple endpoints (groups) selected. I'll see if I can get a debug log of that first or did you already get that?

mdbraber, 2021-04-14:
```
Zigbee2MQTT:debug 2021-04-14 18:48:36: Received Zigbee message from 'sunricher_remote_1', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":1}' from endpoint 1 with groupID 48897
Zigbee2MQTT:info  2021-04-14 18:48:36: MQTT publish: topic 'zigbee2mqtt/sunricher_remote_1', payload '{"action":"recall_1_l1","action_group":48897,"battery":50,"linkquality":221}'
Zigbee2MQTT:info  2021-04-14 18:48:36: MQTT publish: topic 'zigbee2mqtt/sunricher_remote_1', payload '{"action":"","battery":50,"linkquality":221}'
Zigbee2MQTT:info  2021-04-14 18:48:36: MQTT publish: topic 'zigbee2mqtt/sunricher_remote_1/action', payload 'recall_1_l1'
Zigbee2MQTT:debug 2021-04-14 18:48:37: Received Zigbee message from 'sunricher_remote_1', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":1}' from endpoint 1 with groupID 48898
Zigbee2MQTT:info  2021-04-14 18:48:37: MQTT publish: topic 'zigbee2mqtt/sunricher_remote_1', payload '{"action":"recall_1_l1","action_group":48898,"battery":50,"linkquality":220}'
Zigbee2MQTT:info  2021-04-14 18:48:37: MQTT publish: topic 'zigbee2mqtt/sunricher_remote_1', payload '{"action":"","battery":50,"linkquality":220}'
Zigbee2MQTT:info  2021-04-14 18:48:37: MQTT publish: topic 'zigbee2mqtt/sunricher_remote_1/action', payload 'recall_1_l1'
```

I'm wondering if it's Zigbee2MQTT that's at fault here: this is the debug log from enabling endpoint 1 + 2 and it seems that both messages see to originate from endpoint 1 (but with different action groups). A very dirty hack would be to calculate the endpoint based on the action group, but I don't think that's the way. Wondering if there's any more debug info we could dig up here @Koenkk ?

Koenkk, 2021-04-14:
@mdbraber likely this information is correct, but to be sure you can sniff: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

mdbraber, 2021-04-14:
Thanks @koenkk. I'm thinking the Zigbee data is probably correct so it's just sending out this information with the lowest endpoint number for both messages  but different `action_groups`. I'm not quite sure how action_groups are supposed to work with HA. I read this: https://zigbee2mqtt.discourse.group/t/how-to-handle-multiple-endpoints-in-homeassistant/1461/7 and I'm starting to think that maybe adding custom handlers that translate action_group to a endpointID might be a practical hack.

```
2021-04-14T18:52:28.414Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 22
2021-04-14T18:52:28.417Z zigbee-herdsman:deconz:frameParser device state: 10100010
2021-04-14T18:52:28.417Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-04-14T18:52:38.417Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 23
2021-04-14T18:52:38.421Z zigbee-herdsman:deconz:frameParser device state: 10100010
2021-04-14T18:52:38.422Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-04-14T18:52:48.417Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 24
2021-04-14T18:52:48.420Z zigbee-herdsman:deconz:frameParser device state: 10100010
2021-04-14T18:52:48.420Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-04-14T18:52:55.342Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2021-04-14T18:52:55.342Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2021-04-14T18:52:55.344Z zigbee-herdsman:deconz:driver query aps data indication
2021-04-14T18:52:55.346Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2021-04-14T18:52:55.347Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2021-04-14T18:52:55.348Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 25
2021-04-14T18:52:55.349Z zigbee-herdsman:deconz:driver query aps data indication
2021-04-14T18:52:55.352Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 25 srcAddr: 0x7dbf destAddr: 0xbf01 profile id: 0x104 cluster id: 0x5
2021-04-14T18:52:55.353Z zigbee-herdsman:deconz:frameParser response payload: 1,82,5,0,0,1
2021-04-14T18:52:55.357Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":82,"manufacturerCode":null,"commandIdentifier":5},"Payload":{"groupid":0,"sceneid":1},"Command":{"ID":5,"parameters":[{"name":"groupid","type":33},{"name":"sceneid","type":32}],"name":"recall"}},"address":32191,"endpoint":1,"linkquality":223,"groupID":48897,"wasBroadcast":true}'
Zigbee2MQTT:debug 2021-04-14 20:52:55: Received Zigbee message from 'sunricher_remote_1', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":1}' from endpoint 1 with groupID 48897
2021-04-14T18:52:55.374Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
Zigbee2MQTT:info  2021-04-14 20:52:55: MQTT publish: topic 'zigbee2mqtt/sunricher_remote_1', payload '{"action":"recall_1_l1","action_group":48897,"battery":50,"linkquality":223}'
Zigbee2MQTT:info  2021-04-14 20:52:55: MQTT publish: topic 'zigbee2mqtt/sunricher_remote_1', payload '{"action":"","battery":50,"linkquality":223}'
Zigbee2MQTT:info  2021-04-14 20:52:55: MQTT publish: topic 'zigbee2mqtt/sunricher_remote_1/action', payload 'recall_1_l1'
2021-04-14T18:52:55.387Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 26
2021-04-14T18:52:55.387Z zigbee-herdsman:deconz:driver query aps data indication
2021-04-14T18:52:55.391Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 27
2021-04-14T18:52:55.392Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 26 srcAddr: 0x7dbf destAddr: 0xbf02 profile id: 0x104 cluster id: 0x5
2021-04-14T18:52:55.392Z zigbee-herdsman:deconz:frameParser response payload: 1,82,5,0,0,1
2021-04-14T18:52:55.392Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":82,"manufacturerCode":null,"commandIdentifier":5},"Payload":{"groupid":0,"sceneid":1},"Command":{"ID":5,"parameters":[{"name":"groupid","type":33},{"name":"sceneid","type":32}],"name":"recall"}},"address":32191,"endpoint":1,"linkquality":223,"groupID":48898,"wasBroadcast":true}'
Zigbee2MQTT:debug 2021-04-14 20:52:55: Received Zigbee message from 'sunricher_remote_1', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":1}' from endpoint 1 with groupID 48898
2021-04-14T18:52:55.393Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
Zigbee2MQTT:info  2021-04-14 20:52:55: MQTT publish: topic 'zigbee2mqtt/sunricher_remote_1', payload '{"action":"recall_1_l1","action_group":48898,"battery":50,"linkquality":223}'
[... cut ...] tion":"","battery":50,"linkquality":223}'
Zigbee2MQTT:info  2021-04-14 20:52:55: MQTT publish: topic 'zigbee2mqtt/sunricher_remote_1/action', payload 'recall_1_l1'
2021-04-14T18:52:58.418Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 28
2021-04-14T18:52:58.422Z zigbee-herdsman:deconz:frameParser device state: 10100010
2021-04-14T18:52:58.422Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
```

mdbraber, 2021-04-16:
@koenkk I'm trying to adjust the `endpoint.ID` (based on `groupID`) in each message at once, trying to prevent calling all slightly updated commands / functions. I  thought this might be possible with an `onEvent` handler, but it seems I can't manipulate the message that way (only read it). Is there a way to manipulate the `endpoint.ID` for each message or should I make different functions for these use cases (`fz.namron_on, fz.namron_off` etc.)

Koenkk, 2021-04-17:
Please don't go the hacky route :smile: 

I see this is indeed an unhandy situation for HA users. Try applying the changes from https://github.com/Koenkk/zigbee-herdsman-converters/pull/2486/files and add the following in your config:

```yaml
devices:
  '0x12345678':
    friendly_name: my_remote
    add_group_to_action: true # <-- add this
```

Now the group should be added to the `action`.

mdbraber, 2021-04-17:
I got it to work with extra functions, delivering the right endpoints, but this indeed saves some work and is definitely cleaner. Thanks @Koenkk. Got to remove the `multiEndpoint` meta setting. An additional idea could be to make the `add_group_to_action` a dict like `endpoint` in devices:

endpoint: (device) => {
            return {l1: 1, l2: 2, l3: 3, l4: 4};
        },

This way it could translate to the correct endpoint and no translation (groupID -> endpointID) would be needed in a next step - would that be possible?

cveld, 2021-04-18:
@Koenkk Thanks! 

And what about the other issue, i.e. how to add new commands? I.e. moveToHue and store. Can't find much documentation on this.

Koenkk, 2021-04-18:
@mdbraber I don't understand why this is needed or what the benefits over  `multiEndpoint` are. In this way you end up in a situation where every users configures it differently and thus different actions are generated.

@cveld this is possible, but then custom converters have to be added for this device (duplicate e.g. `command_on`)

For the moveToHue you can take `fz.command_ehanced_move_to_hue_and_saturation` as inspiration.

cveld, 2021-04-18:
I changed `postfixWithEndpointName`:

```javascript
function postfixWithEndpointName(value, msg, definition) {
    if (definition.meta && definition.meta.multiEndpoint) {
        const endpointName = definition.hasOwnProperty('endpoint') ?
            getKey(definition.endpoint(msg.device), msg.endpoint.ID) : (definition.hasOwnProperty('convertEndpoint') ?
            definition.convertEndpoint(msg) :
            msg.endpoint.ID);
        return `${value}_${endpointName}`;
    } else {
        return value;
    }
}
```

And my devices.js now contains a `convertEndpoint` implementation:

```javascript
        convertEndpoint: (msg) => {
            switch (msg.groupID) {
                case 19713: return "l1";
                case 19714: return "l2";
                case 19715: return "l3";
                case 19716: return "l4";
            }
            throw `${groupID} not expected`;
        }
```

Koenkk, 2021-04-18:
I see, but now we are deviating from the Zigbee ZCL spec (which the `command_*` adhere to). I propose to create new converters for this device instead.

cveld, 2021-04-18:
For the two unknown commands to get processed `moveToHue` and `store` I believe you also need to add them to the list of recognized commands in the file `events.js` in the package `zigbee-herdsman`? The const `CommandsLookup` contains all recognized commands so it seems. https://github.com/Koenkk/zigbee-herdsman/blob/master/src/controller/events.ts

cveld, 2021-04-18:
Can you point me to the relevant section in the Zigbee Cluster Library Specification? I found this file https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf but don't see any relationship with devices.js at a glance.

Koenkk, 2021-04-18:
- I've added the store and moveToHue. Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)
- Thats the ZCL indeed. The idea behind the `command_*` converters is that they provide received command as is, so no magic conversion with groups -> endpoint are done. If this makes sense for a specific device (which looks to be the case here) a custom set of converters can be created for it.

mdbraber, 2021-04-19:
I've created a set of specific functions - that works fine. It feels a bit superfluous to (re)create all those functions with just a single line difference (`msg.endpoint.ID = msg.groupID - 48896`).

It seems that @cveld has different groupID numbers than I have, which means we would have to configure them somewhere. That wouldn't be `devices.js` but rather in `configuration.yaml` under `devices:`. @koemkk That's why I suggested to make `add_group_to_action` configurable. Maybe I'm overlooking something here for this situation how this should be fixed when there are no hardcoded `groupID`s?

Koenkk, 2021-04-19:
@mdbraber if @cveld has different group IDs this makes no sense indeed. What is the advantage of allowing to specify the mapping? This would add the endpoint twice to the action right? Since the action is in the format of `COMMAND_ENDPOINT_GROUP`. (where the `GROUP` would be remapped to an endpoint again)

mdbraber, 2021-04-27:
@koenkk if the groupIDs are different there should be a way to remap, right? 

Goal is obviously not `COMMAND_ENDPOINT_GROUP` but `COMMAND_ENDPOINT` (or `COMMAND_GROUP` if that would be preferred). That's why I'm suggesting `multiEndpoint` should be switched if you're using the `COMMAND_GROUP` implementation) because otherwise you get `COMMAND_ENDPOINT_GROUP` indeed.

But if we can find a way to specify the groupIDs (which seem to be different) than we can make this implementation more universal using `_l1`, `_l2` etc. Is there a reason to not want to specify a remapping. We could also use a simple offset rather than a remapping as the groupIDs are consecutive (e.g. 48997 -> 1,  48998 -> 2 etc. so the offset would be -48996)

Koenkk, 2021-04-28:
I don't understand why remapping is needed. What is wrong with sending actions like `on_48997`? I understand it's not perfect cosmetic wise but having users to remap this isn't perfect either (manual configuration)

mdbraber, 2021-04-28:
No it isn't - I was supposing a 'universal' way would be preferred (`_l1` etc) so users know what to expect, but this would work fine. Guessing we can close this issue?

Koenkk, 2021-04-29:
> I was supposing a 'universal' way would be preferred (_l1 etc) so users know what to expect

This is indeed preferable, but if I understand correctly this cannot be accomplished since the use group numbers are random.

closed 2021-06-07 by github-actions[bot]