Koenkk/zigbee2mqtt #4394: Linkind ZBT-CCTLight-M3500107

iarema7, 2020-09-18:
## Information about the device + link
. Linkind .ZBT-CCTLight-M3500107 ,just added to devices,js  ;
  zigbeeModel: ['ZBT-CCTLight-C4700107', 'ZBT-CCTLight-M3500107'],
        model: 'ZL1000400-CCT-EU-2-V1A02',
        vendor: 'Linkind',
        description: 'Zigbee LED 5.4W C35 bulb E14, dimmable & tunable',
        extend: generic.light_onoff_brightness_colortemp,
    },
and it is working perfect , please add support in next update

## data/database.db entry of the device
..{"id":3,"type":"Router","ieeeAddr":"0xccccccfffe6a9444","nwkAddr":10062,"manufId":4456,"manufName":"lk","powerSource":"Mains (single phase)","modelId":"ZBT-CCTLight-M3500107","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,2821,4096,64642],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"ZBT-CCTLight-M3500107","manufacturerName":"lk","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20190509-346","swBuildId":"2.2"}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20190509-346","swBuildId":"2.2","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1597246052705}
{"id":4,"type":"Router","ieeeAddr":"0xccccccfffe6a9b40","nwkAddr":59324,"manufId":4456,"manufName":"lk","powerSource":"Mains (single phase)","modelId":"ZBT-CCTLight-M3500107","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,2821,4096,64642],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"ZBT-CCTLight-M3500107","manufacturerName":"lk","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20190509-346","swBuildId":"2.2"}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20190509-346","swBuildId":"2.2","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1597246052581}
{"id":5,"type":"Router","ieeeAddr":"0xccccccfffe6e826a","nwkAddr":61881,"manufId":4456,"manufName":"lk","powerSource":"Mains (single phase)","modelId":"ZBT-CCTLight-M3500107","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,2821,4096,64642],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"ZBT-CCTLight-M3500107","manufacturerName":"lk","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20190509-346","swBuildId":"2.2"}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20190509-346","swBuildId":"2.2","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1597246052677}
{"id":6,"type":"Router","ieeeAddr":"0xccccccfffe6e30a2","nwkAddr":32039,"manufId":4456,"manufName":"lk","powerSource":"Mains (single phase)","modelId":"ZBT-CCTLight-M3500107","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,2821,4096,64642],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"ZBT-CCTLight-M3500107","manufacturerName":"lk","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20190509-346","swBuildId":"2.2"}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20190509-346","swBuildId":"2.2","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1597246052331}
{"id":7,"type":"Router","ieeeAddr":"0xccccccfffe6a9d2f","nwkAddr":4592,"manufId":4456,"manufName":"lk","powerSource":"Mains (single phase)","modelId":"ZBT-CCTLight-M3500107","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,2821,4096,64642],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"ZBT-CCTLight-M3500107","manufacturerName":"lk","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20190509-346","swBuildId":"2.2"}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20190509-346","swBuildId":"2.2","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1597246052868}

Koenkk, 2020-09-18:
Added, will be supported in next release.

closed 2020-09-18 by Koenkk

iarema7, 2020-09-18:
thank you ,have good day