Koenkk/zigbee2mqtt #4397: Failed to call 'OTAUpdate' 'onZigbeeEvent' (Error:  code 434)

shajek, 2020-09-18:
## What happened
After including new Ikea treadfri remote control error "Failed to call 'OTAUpdate' 'onZigbeeEvent' (Error: Request failed with status code 434  at createError " apear (log in pastebin, error in the end)
## What did you expect to happen
Not throwing error
## How to reproduce it (minimal and precise)
remove device from network, and repair it again, and error show again

## Debug info
Zigbee2MQTT version:  1.14.4.1
Adapter hardware: CC2531
Adapter firmware version: latest zStack12 without source routing

https://pastebin.com/3nJGfs3U

sjorge, 2020-09-18:
Probably related to https://github.com/Koenkk/zigbee2mqtt/issues/2921#issuecomment-695014372

Andor, 2020-09-18:
@sjorge Let's rename this issue to something like `IKEA OTA broken http/https URL`?

Details:

The URL used to check updates for IKEA devices is broken:
```
$ curl -I https://fw.ota.homesmart.ikea.net/feed/version_info.json
HTTP/1.1 434 Unknown
Content-Length: 0
Connection: keep-alive
Server: CloudFront
Date: Fri, 18 Sep 2020 18:14:57 GMT
X-Cache: Miss from cloudfront
Via: 1.1 a529b95d300020af7b6819ecefd572f4.cloudfront.net (CloudFront)
X-Amz-Cf-Pop: HEL50-C1
X-Amz-Cf-Id: 7FKJeVJYYaPGX9EfefIKNndolGNtBQHSATlvkqqy0wgjMjbPoi6LbQ==
```

At the same time `http` works (thanks to @PedroLamas to notice that):
```
$ curl -I http://fw.ota.homesmart.ikea.net/feed/version_info.json
HTTP/1.1 200 OK
Content-Type: application/json
Content-Length: 7253
Connection: keep-alive
x-amz-id-2: WY1I5aYPV6X8nRt/hOuPP7EQFGz171EgXnNWn4nww7AePCMZ0SJZU1qr/bRZah7pDYuw7rtUbks=
x-amz-request-id: 86269B1E881EE67E
x-amz-replication-status: COMPLETED
Last-Modified: Tue, 15 Sep 2020 14:04:41 GMT
x-amz-version-id: QILyxJzCjfzQ27p5p54XC1fCpLZll5Ah
Accept-Ranges: bytes
Server: AmazonS3
Date: Fri, 18 Sep 2020 14:31:55 GMT
ETag: "873a4c9cd3ed0ecfa0ba5f473b50da4e"
X-Cache: Hit from cloudfront
Via: 1.1 daf591bf33395af21c13beff18f6613c.cloudfront.net (CloudFront)
X-Amz-Cf-Pop: HEL50-C1
X-Amz-Cf-Id: m7xlHR14ldwogJArLIdHK2CgFKG1mcx-8U4RgCDnrSYXzmuQLPGTyA==
```

PedroLamas, 2020-09-18:
Makes me wonder if this is yet another of those times where someone just forgot to renew an SSL Certificate! 😂

sjorge, 2020-09-18:
Yeah, that would better describe the actual cause for the exception I think.

~ sjorge

PedroLamas, 2020-09-18:
@Koenkk might want to revert this to fix this problem: https://github.com/Koenkk/zigbee-herdsman-converters/pull/978

Pauliakov, 2020-09-18:
Good evening,

sorry but I can't figure out where to change the URL in my files, could you tell me? 

Thank you

lwasy44, 2020-09-18:
Hello @Pauliakov ,
The file is here : ./node_modules/zigbee-herdsman-converters/ota/tradfri.js
I will try revert manually ; https://github.com/Koenkk/zigbee-herdsman-converters/pull/978/commits/ceffd41d208888fe9c30f07171adcf08e1a1bcb1 for the moment.
Good luck

Pauliakov, 2020-09-19:
Thanks for your reply but I cannot open it ... 
/node_modules/zigbee-herdsman-converters/ota/tradfri.js

I'm on Raspberry Pi 4

Koenkk, 2020-09-19:
Revert to http, so should be fixed.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-09-19 by Koenkk

Pauliakov, 2020-09-19:
ok thank you, I will wait so !

labaland, 2020-09-21:
i use this in docker: " image: koenkk/zigbee2mqtt:latest-dev "
Trying to update my SYMFONISK Sound Controller (Current Firmware version is 2.1.022)
Fails every time :P What am i doing wrong ?

dmak, 2020-09-22:
Would it be possible also to improve the error message, so instead of `Failed to call 'OTAUpdate' 'onZigbeeEvent' (Error: Request failed with status code 434` log `Failed to call 'OTAUpdate' 'onZigbeeEvent' (Error: HTTP GET request to https://fw.ota.homesmart.ikea.net/feed/version_info.json had failed with status code 434)` because otherwise it is not clear what exactly went wrong.

GaLaXiaN-Z, 2020-09-25:
hi, i'm using zigbee2mqtt in docker and the failure "Failed to call 'OTAUpdate' 'onZigbeeEvent' (Error: code 434)" continues to persist, I have seen that they have updated but still my failure is not solved, will this failure be solved in the short term?
sorry for my google translate english.

esbenr, 2020-09-27:
Hi,
I'm migrating from CC2531 to CC2652R1 (ZZH) and are experiencing this issue too.
It didn't happen when I was using the CC2531. But on CC2652R1 I am experiencing this issue.

I under stand that the issue is nonresponsive Trådfri update url, but wondering whey I didn't experience this on CC2531.
I'm running on 3.x.0 - CC26X2R1_20200805.

Any suggestions?

```
Zigbee2MQTT:error 2020-09-27 20:35:30: Failed to call 'OTAUpdate' 'onZigbeeEvent' (Error: Request failed with status code 434
    at createError (/zigbee2mqtt-1.14.4/node_modules/zigbee-herdsman-converters/node_modules/axios/lib/core/createError.js:16:15)
    at settle (/zigbee2mqtt-1.14.4/node_modules/zigbee-herdsman-converters/node_modules/axios/lib/core/settle.js:17:12)
    at IncomingMessage.handleStreamEnd (/zigbee2mqtt-1.14.4/node_modules/zigbee-herdsman-converters/node_modules/axios/lib/adapters/http.js:244:11)
    at IncomingMessage.emit (events.js:327:22)
    at endReadableNT (_stream_readable.js:1220:12)
    at processTicksAndRejections (internal/process/task_queues.js:84:21))
```