Koenkk/zigbee2mqtt #4398: UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms

Dustinhoefer, 2020-09-18:
Hello!

Since a few hours, my console is filled with these messages:

(node:2212) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout.waiter.timer.setTimeout [as _onTimeout] (C:\Users\Dustin\Desktop\zigbee2mqtt-master\node_modules\zigbee-herdsman\dist\utils\waitress.js:46:35)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
(node:2212) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:2212) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

Any Idea? I read in this issue (https://github.com/Koenkk/zigbee2mqtt/issues/2318), that it should be fixed and that restarting the PC without the Zigbee Stick should solve this problem. It solved it a bit, but not completely.

Any Ideas?

Koenkk, 2020-09-19:
There are some known stability issues in 1.14.4, should be fixed in latest dev. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Dustinhoefer, 2020-09-20:
@Koenkk 
Updated to the dev branch, copied the database and configuration over.

Getting this error now:

(node:9200) UnhandledPromiseRejectionWarning: Error: SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":2,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,3,2,11,254,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)')
    at Znp.<anonymous> (C:\Users\Dustin\Desktop\zigbee2mqtt-dev\node_modules\zigbee-herdsman\dist\adapter\z-stack\znp\znp.js:291:27)
    at Generator.next (<anonymous>)
    at fulfilled (C:\Users\Dustin\Desktop\zigbee2mqtt-dev\node_modules\zigbee-herdsman\dist\adapter\z-stack\znp\znp.js:24:58)
(node:9200) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:9200) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

Koenkk, 2020-09-20:
Can you try reflashing your CC2531?

Dustinhoefer, 2020-09-20:
I bought it with firmware Juli 2019 and have never flashed it before. Which version should I flash it with? Do you have a good guide for that? :)

Koenkk, 2020-09-20:
Firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing
Guide: https://www.zigbee2mqtt.io/getting_started/flashing_the_cc2531.html

Dustinhoefer, 2020-09-20:
Thanks a bunch! Will try that soonish

Right now it is actually working again. I restarted the pc without the stick, then inserted it in a different usb port.

Hope this fixes it, if not, I will flash the new firmware! Will close the issue for now and report back, if it persists with the new version :)

Dustinhoefer, 2020-09-20:
And it stopped working again ....
Will update the firmware soon and report back ...

Dustinhoefer, 2020-09-20:
@Koenkk 
But can you tell me something please. How can it be that, after 2 month of everything working, it suddenly stops working?
Without me updating anything?

I just unzipped my old version from 2 month ago, set it up again and even that is not working anymore. I have neither updated the stick, nor is zigbee2mqtt any different to the version that was wokring 2of 2 month...

Dustinhoefer, 2020-09-20:
Now I can not even get a single device to connect to the network

Koenkk, 2020-09-21:
The CC2531 is not very powerful and can crash if it receives a lot of commands at the same time. The firmware linked improves stability on the CC2531 (but it will still crash with larger networks, +- 30 devices).

Dustinhoefer, 2020-09-21:
I only have like 10 devices :/
Also, setting up a new network, I can not even get 1 device to connect, before I get the same error.

Koenkk, 2020-09-21:
Thats strange indeed. To start fresh clean the data directory to only contain https://github.com/Koenkk/zigbee2mqtt/tree/master/data and reflash the stick.

If the problem persist provide the herdsman debug logging.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

mrucki, 2020-09-29:
@Koenkk please find debug logs below:

OS: raspbian buster
Zigbee device: CC2531 flashed with CC2531_SOURCE_ROUTING_20190619.zip
Zigbee2Mqtt  checked with:
- docker latest
- github master branch
- github dev branch (pulled yesterday)

and always get `Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms`

The error occurs after about 2h 40 minutes from starting Zigbee2Mqtt.
I have been testing it for last few days: each time after 2h XX minutes !

Tried with dev branch as well - last git pull did yesterday.

The only workaround I figured out is cron job:
```
0 */2 * * * sudo systemctl restart zigbee2mqtt
```
I tested it over a night and in the morning worked fine (start stop commands visible in logs of course) but  I don't want to have such dirty hack in my Raspberry PI :(

The debug log from today, error after 2h 40 mins:

```
Sep 29 12:29:01 raspberrypi npm[788]: Zigbee2MQTT:info  2020-09-29 12:29:01: MQTT publish: topic 'zigbee2mqtt/ArtursRoomTempHum', payload '{"battery":37,"humidity":53.85,"linkquality":23,"pressure":1013
Sep 29 12:30:25 raspberrypi npm[788]: Zigbee2MQTT:debug 2020-09-29 12:30:25: Saving state to file /opt/zigbee2mqtt/data/state.json
Sep 29 12:35:25 raspberrypi npm[788]: Zigbee2MQTT:debug 2020-09-29 12:35:25: Saving state to file /opt/zigbee2mqtt/data/state.json
Sep 29 12:40:25 raspberrypi npm[788]: Zigbee2MQTT:debug 2020-09-29 12:40:25: Saving state to file /opt/zigbee2mqtt/data/state.json
Sep 29 12:45:25 raspberrypi npm[788]: Zigbee2MQTT:debug 2020-09-29 12:45:25: Saving state to file /opt/zigbee2mqtt/data/state.json
Sep 29 12:50:25 raspberrypi npm[788]: Zigbee2MQTT:debug 2020-09-29 12:50:25: Saving state to file /opt/zigbee2mqtt/data/state.json
Sep 29 12:52:13 raspberrypi npm[788]: Zigbee2MQTT:debug 2020-09-29 12:52:13: Received Zigbee message from 'ArtursRoomTempHum', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":2865,"10":0
Sep 29 12:52:13 raspberrypi npm[788]: Zigbee2MQTT:info  2020-09-29 12:52:13: MQTT publish: topic 'zigbee2mqtt/ArtursRoomTempHum', payload '{"battery":37,"humidity":55.62,"linkquality":34,"pressure":1012
Sep 29 12:55:25 raspberrypi npm[788]: Zigbee2MQTT:debug 2020-09-29 12:55:25: Saving state to file /opt/zigbee2mqtt/data/state.json
Sep 29 13:00:25 raspberrypi npm[788]: Zigbee2MQTT:debug 2020-09-29 13:00:25: Saving state to file /opt/zigbee2mqtt/data/state.json
Sep 29 13:04:23 raspberrypi npm[788]: Zigbee2MQTT:debug 2020-09-29 13:04:23: Received Zigbee message from 'KitchenWirelessSwitch', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3005,"1
Sep 29 13:04:23 raspberrypi npm[788]: Zigbee2MQTT:info  2020-09-29 13:04:23: MQTT publish: topic 'zigbee2mqtt/KitchenWirelessSwitch', payload '{"battery":100,"linkquality":55,"voltage":3005}'
Sep 29 13:05:25 raspberrypi npm[788]: Zigbee2MQTT:debug 2020-09-29 13:05:25: Saving state to file /opt/zigbee2mqtt/data/state.json
Sep 29 13:10:11 raspberrypi npm[788]: Zigbee2MQTT:debug 2020-09-29 13:10:11: Received Zigbee message from 'KitchenTempHum', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":2935,"10":0,"1
Sep 29 13:10:11 raspberrypi npm[788]: Zigbee2MQTT:info  2020-09-29 13:10:11: MQTT publish: topic 'zigbee2mqtt/KitchenTempHum', payload '{"battery":62,"humidity":53.03,"linkquality":78,"pressure":1013,"t
Sep 29 13:10:25 raspberrypi npm[788]: Zigbee2MQTT:debug 2020-09-29 13:10:25: Saving state to file /opt/zigbee2mqtt/data/state.json
Sep 29 13:10:43 raspberrypi npm[788]: Zigbee2MQTT:debug 2020-09-29 13:10:42: Received Zigbee message from 'ArtursRoomTempHum', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measure
Sep 29 13:10:43 raspberrypi npm[788]: Zigbee2MQTT:info  2020-09-29 13:10:43: MQTT publish: topic 'zigbee2mqtt/ArtursRoomTempHum', payload '{"battery":37,"humidity":55.62,"linkquality":42,"pressure":1012
Sep 29 13:10:43 raspberrypi npm[788]: Zigbee2MQTT:debug 2020-09-29 13:10:43: Received Zigbee message from 'ArtursRoomTempHum', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue
Sep 29 13:10:43 raspberrypi npm[788]: Zigbee2MQTT:info  2020-09-29 13:10:43: MQTT publish: topic 'zigbee2mqtt/ArtursRoomTempHum', payload '{"battery":37,"humidity":56.47,"linkquality":39,"pressure":1012
Sep 29 13:10:43 raspberrypi npm[788]: Zigbee2MQTT:debug 2020-09-29 13:10:43: Received Zigbee message from 'ArtursRoomTempHum', type 'attributeReport', cluster 'msPressureMeasurement', data '{"16":10129,
Sep 29 13:10:43 raspberrypi npm[788]: Zigbee2MQTT:info  2020-09-29 13:10:43: MQTT publish: topic 'zigbee2mqtt/ArtursRoomTempHum', payload '{"battery":37,"humidity":56.47,"linkquality":39,"pressure":1012
Sep 29 13:12:14 raspberrypi npm[788]: (node:801) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
Sep 29 13:12:14 raspberrypi npm[788]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
Sep 29 13:12:14 raspberrypi npm[788]:     at listOnTimeout (internal/timers.js:549:17)
Sep 29 13:12:14 raspberrypi npm[788]:     at processTimers (internal/timers.js:492:7)
Sep 29 13:12:14 raspberrypi npm[788]: (node:801) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block
Sep 29 13:12:14 raspberrypi npm[788]: (node:801) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node
[... cut ...]
Sep 29 13:15:34 raspberrypi npm[788]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
Sep 29 13:15:34 raspberrypi npm[788]:     at listOnTimeout (internal/timers.js:549:17)
Sep 29 13:15:34 raspberrypi npm[788]:     at processTimers (internal/timers.js:492:7)
Sep 29 13:15:34 raspberrypi npm[788]: (node:801) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block
lines 269-318/318 (END)

```

Koenkk, 2020-09-29:
Please provide the herdsman debug logging (normal debug logging isn't enough to debug this issue). See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging how to obtain this logging

mrucki, 2020-09-29:
@Koenkk this time with zigbee-herdsman debug log:

```
Sep 29 15:34:29 raspberrypi npm[31735]: Zigbee2MQTT:debug 2020-09-29 15:34:29: Saving state to file /opt/zigbee2mqtt/data/state.json
Sep 29 15:34:47 raspberrypi npm[31735]: 2020-09-29T13:34:47.349Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,55,68,129,0,0,0,0,145,210,1,1,0,49,0,52,42,172,0,0,35,28,95,17,90,10,1,255,66,26,1,33
Sep 29 15:34:47 raspberrypi npm[31735]: 2020-09-29T13:34:47.349Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,55,68,129,0,0,0,0,145,210,1,1,0,49,0,52,42,172,0,0,35,28,95,17,90,10,1,255,
Sep 29 15:34:47 raspberrypi npm[31735]: 2020-09-29T13:34:47.350Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 55 - 2 - 4 - 129 - [0,0,0,0,145,210,1,1,0,49,0,52,42,172,0,0,35,28,95,17,90,10,1,25
Sep 29 15:34:47 raspberrypi npm[31735]: 2020-09-29T13:34:47.351Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":53905,"srcendpoint":1,"dstendpoint":1
Sep 29 15:34:47 raspberrypi npm[31735]: 2020-09-29T13:34:47.354Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direct
Sep 29 15:34:47 raspberrypi npm[31735]: Zigbee2MQTT:debug 2020-09-29 15:34:47: Received Zigbee message from 'KitchenWirelessSwitch', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3005,
Sep 29 15:34:47 raspberrypi npm[31735]: Zigbee2MQTT:info  2020-09-29 15:34:47: MQTT publish: topic 'zigbee2mqtt/KitchenWirelessSwitch', payload '{"battery":100,"linkquality":49,"voltage":3005}'
Sep 29 15:34:47 raspberrypi npm[31735]: 2020-09-29T13:34:47.363Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Sep 29 15:36:12 raspberrypi npm[31735]: 2020-09-29T13:36:12.702Z zigbee-herdsman:controller:log Permit joining
Sep 29 15:36:12 raspberrypi npm[31735]: 2020-09-29T13:36:12.703Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0}
Sep 29 15:36:12 raspberrypi npm[31735]: 2020-09-29T13:36:12.704Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228]
Sep 29 15:36:12 raspberrypi npm[31735]: 2020-09-29T13:36:12.717Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,54,0,82]
Sep 29 15:36:12 raspberrypi npm[31735]: 2020-09-29T13:36:12.718Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82]
Sep 29 15:36:12 raspberrypi npm[31735]: 2020-09-29T13:36:12.718Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [0] - 82
Sep 29 15:36:12 raspberrypi npm[31735]: 2020-09-29T13:36:12.718Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0}
Sep 29 15:36:12 raspberrypi npm[31735]: 2020-09-29T13:36:12.719Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Sep 29 15:36:12 raspberrypi npm[31735]: 2020-09-29T13:36:12.721Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dst
Sep 29 15:36:12 raspberrypi npm[31735]: 2020-09-29T13:36:12.722Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,26,36,2,2,253,255,0,0,0,0,0,0,242,0,0,242,33,0,33,0,30,6,0,25,34,2,11,254,0,232
Sep 29 15:36:12 raspberrypi npm[31735]: 2020-09-29T13:36:12.727Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,182,0,0,0,240]
Sep 29 15:36:12 raspberrypi npm[31735]: 2020-09-29T13:36:12.728Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240]
Sep 29 15:36:12 raspberrypi npm[31735]: 2020-09-29T13:36:12.728Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [0,0,0] - 240
Sep 29 15:36:12 raspberrypi npm[31735]: 2020-09-29T13:36:12.728Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0}
Sep 29 15:36:12 raspberrypi npm[31735]: 2020-09-29T13:36:12.729Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Sep 29 15:36:12 raspberrypi npm[31735]: 2020-09-29T13:36:12.735Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103]
Sep 29 15:36:12 raspberrypi npm[31735]: 2020-09-29T13:36:12.735Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103]
Sep 29 15:36:12 raspberrypi npm[31735]: 2020-09-29T13:36:12.736Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103
Sep 29 15:36:12 raspberrypi npm[31735]: 2020-09-29T13:36:12.736Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0}
Sep 29 15:36:12 raspberrypi npm[31735]: 2020-09-29T13:36:12.736Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Sep 29 15:36:12 raspberrypi npm[31735]: 2020-09-29T13:36:12.741Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,242,33,20]
Sep 29 15:36:12 raspberrypi npm[31735]: 2020-09-29T13:36:12.742Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,242,33,20]
Sep 29 15:36:12 raspberrypi npm[31735]: 2020-09-29T13:36:12.742Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,242,33] - 20
Sep 29 15:36:12 raspberrypi npm[31735]: 2020-09-29T13:36:12.742Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":242,"transid":33}
Sep 29 15:36:12 raspberrypi npm[31735]: 2020-09-29T13:36:12.743Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
[... cut ...]
Sep 29 15:52:52 raspberrypi npm[31735]: 2020-09-29T13:52:52.732Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228]
Sep 29 15:52:58 raspberrypi npm[31735]: (node:31748) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
Sep 29 15:52:58 raspberrypi npm[31735]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
Sep 29 15:52:58 raspberrypi npm[31735]:     at listOnTimeout (internal/timers.js:549:17)
Sep 29 15:52:58 raspberrypi npm[31735]:     at processTimers (internal/timers.js:492:7)
```

Koenkk, 2020-09-29:
At the moment it crashes, can you provide the output of `dmesg` command?

mrucki, 2020-09-29:
This time I set `permit_join: false` in configuration file and it just stopped working.
No error in log file, only  sever these messages: `Saving state to file /opt/zigbee2mqtt/data/state.json`

Then `dmesg` 

```
pi@raspberrypi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.68-v7+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1343 SMP Mon Sep 28 12:38:29 BST 2020
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x37400000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   Normal zone: 2133 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 242688 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s49920 r8192 d23808 u81920
[    0.000000] pcpu-alloc: s49920 r8192 d23808 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=cc2b1ff6-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 880476K/970752K available (9216K kernel code, 697K rwdata, 2604K rodata, 1024K init, 830K bss, 24740K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 29216 entries in 58 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x32c/0x508 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000022] Switching to timer-based delay loop, resolution 52ns
[    0.000290] Console: colour dummy device 80x30
[    0.000876] printk: console [tty1] enabled
[    0.000944] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000992] pid_max: default: 32768 minimum: 301
[    0.001205] LSM: Security Framework initializing
[    0.001449] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001495] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002895] Disabling memory control group subsystem
[    0.003040] CPU: Testing write buffer coherency: ok
[    0.003631] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.004599] Setting up static identity map for 0x100000 - 0x10003c
[    0.004820] rcu: Hierarchical SRCU implementation.
[    0.005531] smp: Bringing up secondary CPUs ...
[    0.006726] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.008053] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.009271] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.009424] smp: Brought up 1 node, 4 CPUs
[    0.009513] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.009541] CPU: All CPU(s) started in HYP mode.
[    0.009566] CPU: Virtualization extensions available.
[    0.010671] devtmpfs: initialized
[    0.027163] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.027466] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.027523] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.030728] pinctrl core: initialized pinctrl subsystem
[    0.031864] NET: Registered protocol family 16
[    0.036164] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.041062] audit: initializing netlink subsys (disabled)
[    0.041366] audit: type=2000 audit(0.040:1): state=initialized audit_enabled=0 res=1
[    0.042835] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.042875] hw-breakpoint: maximum watchpoint size is 8 bytes.
[... 218 lines cut ...]
[    8.919082] random: crng init done
[    8.919097] random: 7 urandom warning(s) missed due to ratelimiting
[    9.835077] uart-pl011 3f201000.serial: no DMA platform data
[    9.855166] 8021q: 802.1Q VLAN Support v1.8
[   10.176517] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[... cut ...]
[   17.065282] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   17.065293] Bluetooth: BNEP filters: protocol multicast
[   17.065315] Bluetooth: BNEP socket layer initialized
[   17.228572] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.wlan0.retrans_time - use net.ipv6.neigh.wlan0.retrans_time_ms instead
[   36.254117] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[   36.265346] Bridge firewalling registered
```

Koenkk, 2020-09-29:
- How is the CC2531 connected? (usb extension cable/usb hub?
- Are there any other USB devices attached to your pi?
- Is below logged before or after Zigbee2MQTT started:
```
[    4.056650] usb 1-1.1.2: New USB device found, idVendor=0451, idProduct=16a8, bcdDevice= 0.09
[    4.062606] usb 1-1.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    4.068943] usb 1-1.1.2: Product: TI CC2531 USB CDC
[    4.072038] usb 1-1.1.2: Manufacturer: Texas Instruments
[    4.074950] usb 1-1.1.2: SerialNumber: __0X00124B001936A05F
[    4.386377] usb 1-1.1.1: new high-speed USB device number 5 using dwc_otg
[    4.516956] usb 1-1.1.1: New USB device found, idVendor=0424, idProduct=7800, bcdDevice= 3.00
[    4.522608] usb 1-1.1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
```

It looks you are having some USB instability causing the CC2531 to be disconnected. This can also be caused by a bad power supply on the Pi.

mrucki, 2020-09-30:
- CC2531 is connected with extension cable
- CC2531 is the only usb device connected
- I'm using zigbee2mqtt as a systemctl service - I rebooted Raspberry Pi, stopped zigbee2mqtt and the log above occured during dmesg
- I'm using power supply with output 5v and 3.1 A so should be enough

Ah, and one thing I missed in my comments is that my raspberry pi + zigbee2mqtt + openhab has worked fine for last year until few days ago I upgraded my Raspbian from `stretch` to `buster` version - perhaps this causes usb instability? Are there any known issues with `Raspbian Buster` ?

mrucki, 2020-09-30:
@Koenkk I found there is a possibility to set an option like `max_usb_current=1` in `/boot/config.txt` but it was missing in my device.
So I added it to my config file and after 4 hours (much longer then previously) issue occurred again:
```
pi@raspberrypi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.68-v7+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1343 SMP Mon Sep 28 12:38:29 BST 2020
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x37400000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   Normal zone: 2133 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 242688 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s49920 r8192 d23808 u81920
[    0.000000] pcpu-alloc: s49920 r8192 d23808 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=cc2b1ff6-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 880476K/970752K available (9216K kernel code, 697K rwdata, 2604K rodata, 1024K init, 830K bss, 24740K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 29216 entries in 58 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x32c/0x508 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000022] Switching to timer-based delay loop, resolution 52ns
[    0.000290] Console: colour dummy device 80x30
[    0.000877] printk: console [tty1] enabled
[    0.000944] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000993] pid_max: default: 32768 minimum: 301
[    0.001208] LSM: Security Framework initializing
[    0.001450] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001496] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002896] Disabling memory control group subsystem
[    0.003042] CPU: Testing write buffer coherency: ok
[    0.003628] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.004599] Setting up static identity map for 0x100000 - 0x10003c
[    0.004819] rcu: Hierarchical SRCU implementation.
[    0.005528] smp: Bringing up secondary CPUs ...
[    0.006717] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.008040] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.009254] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.009407] smp: Brought up 1 node, 4 CPUs
[    0.009495] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.009523] CPU: All CPU(s) started in HYP mode.
[    0.009548] CPU: Virtualization extensions available.
[    0.010650] devtmpfs: initialized
[    0.027154] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.027459] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.027516] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.030739] pinctrl core: initialized pinctrl subsystem
[    0.031875] NET: Registered protocol family 16
[    0.036180] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.041075] audit: initializing netlink subsys (disabled)
[    0.041379] audit: type=2000 audit(0.040:1): state=initialized audit_enabled=0 res=1
[    0.042847] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.042886] hw-breakpoint: maximum watchpoint size is 8 bytes.
[    0.043124] Serial: AMBA PL011 UART driver
[    0.045117] bcm2835-mbox 3f00b880.mailbox: mailbox enabled
[    0.070103] raspberrypi-firmware soc:firmware: Attached to firmware from 2020-09-28 12:54, variant start
[... 215 lines cut ...]
[    9.534759] uart-pl011 3f201000.serial: no DMA platform data
[... cut ...]
[   16.790004] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.wlan0.retrans_time - use net.ipv6.neigh.wlan0.retrans_time_ms instead
[   16.879275] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   16.879288] Bluetooth: BNEP filters: protocol multicast
[   16.879305] Bluetooth: BNEP socket layer initialized
[   36.205041] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[   36.216776] Bridge firewalling registered
```

Koenkk, 2020-09-30:
Try plugging the stick directly in the pi (without extension cable), does it survive longer?

mrucki, 2020-09-30:
Ok, will try tonight

One more thing, today I stopped zigbee2mqtt, 
then cleared dmesg with `sudo dmesg -C`
then `sudo iwconfig wlan0 power off`
then started zigbee2mqtt and when it  failed the dmesg showed only:
```
pi@raspberrypi:~ $ dmesg 
[26123.004286] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
```
no info about reconnecting usb  device

mrucki, 2020-10-01:
@Koenkk same issue when plugging the stick directly in the pi (without extension cable).

golddragon007, 2020-10-02:
Hello,

I've changed my cc2531 to slaesh’s CC2652RB stick. After I've forced the ZigBee stick to start with the old pan_id (even if I've changed the pan id it didn't want to start... see: https://github.com/Koenkk/zigbee2mqtt/issues/4513), now I've got both errors in the z2m log in debug log mode, in mixed order:
```
Zigbee2MQTT:debug 2020-10-02 10:09:48: Saving state to file /share/zigbee2mqtt/state.json
(node:235) UnhandledPromiseRejectionWarning: Error: SREQ '--> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')
    at Znp.<anonymous> (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:291:27)
    at Generator.next (<anonymous>)
    at fulfilled (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58)
(node:235) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 36)
Zigbee2MQTT:debug 2020-10-02 10:14:48: Saving state to file /share/zigbee2mqtt/state.json
(node:235) UnhandledPromiseRejectionWarning: Error: SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":188,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,189,2,11,254,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')
    at Znp.<anonymous> (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:291:27)
    at Generator.next (<anonymous>)
    at fulfilled (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58)
(node:235) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 37)
Zigbee2MQTT:debug 2020-10-02 10:19:48: Saving state to file /share/zigbee2mqtt/state.json
```

I don't see where do I need to put the `"zigbee_herdsman_debug": true` in the config, it's missing from here: https://www.zigbee2mqtt.io/information/configuration.html

Koenkk, 2020-10-02:
@mrucki is anything shown in dmesg at the time Zigbee2MQTT crashes?

@golddragon007 `zigbee_herdsman_debug: true` only applies to the Home Assistant addon. See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging . Might also be worth looking at `dmesg` at the time of the crash. Also what system do you use (Pi?), how is the adapter connected, and are any there any other USB devices attached?

mrucki, 2020-10-02:
@Koenkk as wrote above:
> sudo dmesg -C
> then sudo iwconfig wlan0 power off
> then started zigbee2mqtt and when it failed the dmesg showed only
> pi@raspberrypi:~ $ dmesg 
> [26123.004286] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled (which is fine as tuned off power management above)

so nothing in dmesg

Today I reflashed my RaspberryPi with brand new RaspberryPi OS Buster - started OpenHab, Grafana, Influxdb, Eclipse Mosquitto and Zigbee2Mqtt (latest) using Docker.

Unfortunatelly issue ocuured again - I wonder if someone else tested zigbee2mqtt with raspbian buster?

golddragon007, 2020-10-02:
@Koenkk 
I using Pi 4 with bash only (no GUI). My boot drive is an m.2 SSD connected via USB 3 and now the slaesh’s CC2652RB stick (directly).

"zigbee_herdsman_debug: true only applies to the Home Assistant addon"
Yes, I know, that's why I wanted to use that. I've put it into the root, it didn't cry so I hope it does the job.

Regarding `dmesg`, I have something but this time looks for me yesterday's midnight when I tried to figure out why it doesn't want to boot up and start broadcasting the Zigbee network:
```Shell
[...]
[ 6027.488915] scsi host0: uas_eh_device_reset_handler start
[ 6027.639647] usb 2-2: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[ 6027.673081] scsi host0: uas_eh_device_reset_handler success
[ 6030.532417] hassio: port 14(veth8c410e9) entered blocking state
[ 6030.532427] hassio: port 14(veth8c410e9) entered disabled state
[ 6030.532795] device veth8c410e9 entered promiscuous mode
[ 6030.573408] IPv6: ADDRCONF(NETDEV_CHANGE): vethcd6a52d: link becomes ready
[ 6030.573560] IPv6: ADDRCONF(NETDEV_CHANGE): veth8c410e9: link becomes ready
[ 6030.573660] hassio: port 14(veth8c410e9) entered blocking state
[ 6030.573666] hassio: port 14(veth8c410e9) entered forwarding state
[ 6032.193073] hassio: port 14(veth8c410e9) entered disabled state
[ 6032.203987] eth0: renamed from vethcd6a52d
[ 6032.270318] hassio: port 14(veth8c410e9) entered blocking state
[ 6032.270331] hassio: port 14(veth8c410e9) entered forwarding state
[ 6138.872649] hassio: port 14(veth8c410e9) entered disabled state
[ 6138.873340] vethcd6a52d: renamed from eth0
[ 6139.019898] hassio: port 14(veth8c410e9) entered disabled state
[ 6139.037527] device veth8c410e9 left promiscuous mode
[ 6139.037538] hassio: port 14(veth8c410e9) entered disabled state
[ 6176.650967] usb 1-1.3: USB disconnect, device number 6
[ 6176.651545] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[ 6176.651661] cp210x 1-1.3:1.0: device disconnected
[ 6187.850223] usb 1-1.3: new full-speed USB device number 7 using xhci_hcd
[ 6187.985569] usb 1-1.3: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[ 6187.985583] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6187.985593] usb 1-1.3: Product: slae.sh cc2652rb stick - slaesh's iot stuff
[ 6187.985603] usb 1-1.3: Manufacturer: Silicon Labs
[ 6187.985613] usb 1-1.3: SerialNumber: 00_12_4B_00_21_CC_09_E4
[ 6187.989631] cp210x 1-1.3:1.0: cp210x converter detected
[ 6187.997921] usb 1-1.3: cp210x converter now attached to ttyUSB0
[ 6218.140701] sd 0:0:0:0: [sda] tag#27 uas_eh_abort_handler 0 uas-tag 6 inflight: CMD
[ 6218.140710] sd 0:0:0:0: [sda] tag#27 CDB: opcode=0x2a 2a 00 00 cd 47 d8 00 00 08 00
[ 6218.140723] sd 0:0:0:0: [sda] tag#26 uas_eh_abort_handler 0 uas-tag 4 inflight: CMD
[ 6218.140729] sd 0:0:0:0: [sda] tag#26 CDB: opcode=0x2a 2a 00 01 cc 9f 70 00 00 08 00
[ 6218.140746] sd 0:0:0:0: [sda] tag#25 uas_eh_abort_handler 0 uas-tag 1 inflight: CMD IN
[ 6218.140751] sd 0:0:0:0: [sda] tag#25 CDB: opcode=0x28 28 00 00 b6 d6 c0 00 00 20 00
[ 6218.141072] sd 0:0:0:0: [sda] tag#11 uas_eh_abort_handler 0 uas-tag 5 inflight: CMD
[ 6218.141077] sd 0:0:0:0: [sda] tag#11 CDB: opcode=0x2a 2a 00 03 4c 70 48 00 00 08 00
[ 6218.141087] sd 0:0:0:0: [sda] tag#10 uas_eh_abort_handler 0 uas-tag 3 inflight: CMD
[ 6218.141092] sd 0:0:0:0: [sda] tag#10 CDB: opcode=0x2a 2a 00 03 4c 32 38 00 00 08 00
[ 6218.141108] sd 0:0:0:0: [sda] tag#7 uas_eh_abort_handler 0 uas-tag 2 inflight: CMD
[ 6218.141113] sd 0:0:0:0: [sda] tag#7 CDB: opcode=0x2a 2a 00 01 54 89 48 00 00 08 00
[ 6219.660740] sd 0:0:0:0: [sda] tag#24 uas_eh_abort_handler 0 uas-tag 7 inflight: CMD
[ 6219.660748] sd 0:0:0:0: [sda] tag#24 CDB: opcode=0x2a 2a 00 01 cc a3 b8 00 00 08 00
[ 6219.990754] sd 0:0:0:0: [sda] tag#4 uas_eh_abort_handler 0 uas-tag 8 inflight: CMD
[ 6219.990762] sd 0:0:0:0: [sda] tag#4 CDB: opcode=0x2a 2a 00 00 e1 32 c8 00 00 08 00
[ 6221.110754] sd 0:0:0:0: [sda] tag#8 uas_eh_abort_handler 0 uas-tag 9 inflight: CMD
[ 6221.110763] sd 0:0:0:0: [sda] tag#8 CDB: opcode=0x2a 2a 00 00 df 82 f0 00 00 08 00
[ 6221.190764] sd 0:0:0:0: [sda] tag#5 uas_eh_abort_handler 0 uas-tag 10 inflight: CMD
[ 6221.190773] sd 0:0:0:0: [sda] tag#5 CDB: opcode=0x35 35 00 00 00 00 00 00 00 00 00
[ 6223.260806] sd 0:0:0:0: [sda] tag#6 uas_eh_abort_handler 0 uas-tag 11 inflight: CMD OUT
[ 6223.260815] sd 0:0:0:0: [sda] tag#6 CDB: opcode=0x2a 2a 00 00 18 6e c8 00 02 00 00
[ 6227.580874] sd 0:0:0:0: [sda] tag#9 uas_eh_abort_handler 0 uas-tag 12 inflight: CMD OUT
[ 6227.580888] sd 0:0:0:0: [sda] tag#9 CDB: opcode=0x2a 2a 00 01 58 71 f8 00 00 08 00
[ 6229.740901] sd 0:0:0:0: [sda] tag#28 uas_eh_abort_handler 0 uas-tag 13 inflight: CMD OUT
[ 6229.740910] sd 0:0:0:0: [sda] tag#28 CDB: opcode=0x2a 2a 00 00 61 24 10 00 00 20 00
[ 6234.781009] sd 0:0:0:0: [sda] tag#23 uas_eh_abort_handler 0 uas-tag 25 inflight: CMD OUT
[ 6234.781017] sd 0:0:0:0: [sda] tag#23 CDB: opcode=0x2a 2a 00 01 82 9c d8 00 00 10 00
[ 6234.781319] sd 0:0:0:0: [sda] tag#22 uas_eh_abort_handler 0 uas-tag 24 inflight: CMD OUT
[ 6234.781326] sd 0:0:0:0: [sda] tag#22 CDB: opcode=0x2a 2a 00 01 82 93 c8 00 00 08 00
[ 6234.781630] sd 0:0:0:0: [sda] tag#21 uas_eh_abort_handler 0 uas-tag 23 inflight: CMD OUT
[... 342 lines cut ...]
[ 7764.833576] IPv6: ADDRCONF(NETDEV_CHANGE): veth6bcccb7: link becomes ready
[ 7764.833691] hassio: port 14(veth6bcccb7) entered blocking state
[ 7764.833700] hassio: port 14(veth6bcccb7) entered forwarding state
[... cut ...]
[ 8015.845207] hassio: port 14(veth0a4c4db) entered blocking state
[ 8015.845214] hassio: port 14(veth0a4c4db) entered forwarding state
[ 8015.880843] IPv6: ADDRCONF(NETDEV_CHANGE): veth0a4c4db: link becomes ready
[ 8017.070566] hassio: port 14(veth0a4c4db) entered disabled state
[ 8017.071975] eth0: renamed from veth17bfca5
[ 8017.142333] hassio: port 14(veth0a4c4db) entered blocking state
[ 8017.142342] hassio: port 14(veth0a4c4db) entered forwarding state
bash-5.0# cat /proc/uptime
79962.78 254697.07
```

Koenkk, 2020-10-02:
@golddragon007 looks that the stick is disconnected:

```
[ 7356.393791] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[ 7356.394086] cp210x 1-1.3:1.0: device disconnected
[ 7472.595120] usb 1-1.3: new full-speed USB device number 9 using xhci_hcd
[ 7472.730514] usb 1-1.3: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[ 7472.730522] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 7472.730529] usb 1-1.3: Product: slae.sh cc2652rb stick - slaesh's iot stuff
[ 7472.730535] usb 1-1.3: Manufacturer: Silicon Labs
[ 7472.730541] usb 1-1.3: SerialNumber: 00_12_4B_00_21_CC_09_E4
[ 7472.736186] cp210x 1-1.3:1.0: cp210x converter detected
[ 7472.740538] usb 1-1.3: cp210x converter now attached to ttyUSB0
```

I think this happens because the SSD draws to much current, leading to some instability on the USB bus. Probably not fixable from Zigbee2MQTT itself. @slaesh maybe you have a solution for this issue?

golddragon007, 2020-10-02:
> @golddragon007 looks that the stick is disconnected:
>
> ```
> [ 7356.393791] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
> [ 7356.394086] cp210x 1-1.3:1.0: device disconnected
> [ [...]

No, that happened when I physically removed the stick. It was 20 hours ago. After that I have no new disconnections logged, and I still get messages in the HA add-on log.

mrucki, 2020-10-03:
@Koenkk I run again zigbee2mqtt in docker but now with `DEBUG=*`

Then I save all logs to file and then removed logs related to mqtt:
```
less zigbee2mqtt-1.log | grep -v "mmqttjs:client" | grep -v "mqtt-packet" > zigbee2mqtt-no-mqtt.log
``` 

Please find `zigbee2mqtt-no-mqtt.log` attached - it has `zigbee-herdsman` and `serialport` lib debug logs
[zigbee2mqtt-no-mqtt.log](https://github.com/Koenkk/zigbee2mqtt/files/5321289/zigbee2mqtt-no-mqtt.log)

mrucki, 2020-10-03:
@Koenkk  please note that this command:
```
less zigbee2mqtt-1.log | grep -v "mmqttjs:client" | grep -v "mqtt-packet" | grep "serialport/bindings" | grep "read"
```
shows last log from 19:39 where first zigbee2mqtt timeout occured at 19:42

Looks like there is an issue with reading data from serial port...

golddragon007, 2020-10-03:
@Koenkk 
For me, after I've restarted the add-on with herdsman debug, I did not get any more error, but still, nothing wants to connect to the controller and I don't know why, I tried to repair it, but that's also doesn't help.
dmesg after restart:
```
[80512.212110] hassio: port 14(veth0a4c4db) entered disabled state
[80512.212471] veth17bfca5: renamed from eth0
[80512.336085] hassio: port 14(veth0a4c4db) entered disabled state
[80512.352997] device veth0a4c4db left promiscuous mode
[80512.353009] hassio: port 14(veth0a4c4db) entered disabled state
[80512.835365] hassio: port 14(veth274155e) entered blocking state
[80512.835373] hassio: port 14(veth274155e) entered disabled state
[80512.835648] device veth274155e entered promiscuous mode
[80512.836046] hassio: port 14(veth274155e) entered blocking state
[80512.836053] hassio: port 14(veth274155e) entered forwarding state
[80512.881920] IPv6: ADDRCONF(NETDEV_CHANGE): veth274155e: link becomes ready
[80514.105124] hassio: port 14(veth274155e) entered disabled state
[80514.105710] eth0: renamed from veth0d9a584
[80514.185625] hassio: port 14(veth274155e) entered blocking state
[80514.185634] hassio: port 14(veth274155e) entered forwarding state
```
Docker log:
```
Zigbee2MQTT:debug 2020-10-03 13:13:05: Saving state to file /share/zigbee2mqtt/state.json
2020-10-03T11:14:36.395Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,4,11,19,201,1,1,0,66,0,49,51,195,0,0,8,8,14,10,5,5,33,18,9,19,201,29,118]
2020-10-03T11:14:36.395Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,4,11,19,201,1,1,0,66,0,49,51,195,0,0,8,8,14,10,5,5,33,18,9,19,201,29,118]
2020-10-03T11:14:36.395Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,4,11,19,201,1,1,0,66,0,49,51,195,0,0,8,8,14,10,5,5,33,18,9,19,201,29] - 118
2020-10-03T11:14:36.395Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":2820,"srcaddr":51475,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":66,"securityuse":0,"timestamp":12792625,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[8,14,10,5,5,33,18,9]}}
2020-10-03T11:14:36.399Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":14,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":1285,"dataType":33,"attrData":2322}]},"address":51475,"endpoint":1,"linkquality":66,"groupID":0}'
2020-10-03T11:14:36.399Z zigbee-herdsman:controller:log 'zcl' data is from unknown device with address '51475', skipping...
2020-10-03T11:14:36.399Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-10-03T11:14:57.391Z zigbee-herdsman:controller:log Permit joining
2020-10-03T11:14:57.391Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0}
2020-10-03T11:14:57.391Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228]
2020-10-03T11:14:57.399Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,54,0,82]
2020-10-03T11:14:57.399Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82]
2020-10-03T11:14:57.399Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [0] - 82
2020-10-03T11:14:57.399Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0}
2020-10-03T11:14:57.399Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-10-03T11:14:57.400Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":78,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,79,2,11,254,0]}}
2020-10-03T11:14:57.401Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,26,36,2,2,253,255,0,0,0,0,0,0,242,0,0,242,33,0,78,0,30,6,0,25,79,2,11,254,0,234]
2020-10-03T11:14:57.401Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,182,0,0,0,240]
2020-10-03T11:14:57.401Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240]
2020-10-03T11:14:57.401Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [0,0,0] - 240
2020-10-03T11:14:57.402Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0}
2020-10-03T11:14:57.402Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-10-03T11:14:57.409Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103]
2020-10-03T11:14:57.409Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103]
2020-10-03T11:14:57.409Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103
2020-10-03T11:14:57.410Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0}
2020-10-03T11:14:57.410Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-10-03T11:14:57.433Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,225,242,78,154]
2020-10-03T11:14:57.433Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,225,242,78,154]
2020-10-03T11:14:57.433Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [225,242,78] - 154
[... cut ...]
2020-10-03T11:41:37.457Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,225,242,86,130]
2020-10-03T11:41:37.457Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,225,242,86,130]
2020-10-03T11:41:37.457Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [225,242,86] - 130
2020-10-03T11:41:37.458Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":225,"endpoint":242,"transid":86}
2020-10-03T11:41:37.458Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

mrucki, 2020-10-03:
@Koenkk please look what else I noticed:

```
2020-10-03T13:24:59.601685393Z   serialport/bindings/unixRead Starting read +5ms
2020-10-03T13:24:59.603077207Z   serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
2020-10-03T13:24:59.603741422Z   serialport/bindings/unixRead waiting for readable because of code: EAGAIN +1ms
**2020-10-03T13:24:59.604015118Z   serialport/bindings/poller Polling for "readable" +14ms**
2020-10-03T13:28:19.588261306Z   serialport/stream _write 10 bytes of data +3m
2020-10-03T13:28:19.588339899Z   serialport/binding-abstract write 10 bytes +3m
2020-10-03T13:28:19.589390882Z   serialport/bindings/unixWrite Starting write 10 bytes offset 0 bytesToWrite 10 +3m
2020-10-03T13:28:19.590786030Z   serialport/bindings/unixWrite write returned: wrote 10 bytes +1ms
2020-10-03T13:28:19.591875658Z   serialport/bindings/unixWrite Finished writing 10 bytes +1ms
2020-10-03T13:28:19.592046959Z   serialport/stream binding.write write finished +4ms
**2020-10-03T13:28:19.601888773Z   serialport/bindings/poller received "readable" +3m**
```

received readable after 3 minutes? Does it mean that if during that time zigbee-herdsman would try to send any data (permitJoin request or dataRequestExt, etc) it would crash due to timeout 6000ms ?

MORD0R, 2020-10-03:
> * CC2531 is connected with extension cable
>
>     * CC2531 is the only usb device connected
>
>     * I'm using zigbee2mqtt as a systemctl service - I rebooted Raspberry Pi, stopped zigbee2mqtt and the [...]

I have almost the same software installed and i do also have this error UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms. But the problem started afther i upgraded zigbee2mqtt. Maybe it has something to do with the zram. Openhabian added zram as default.  https://community.openhab.org/t/zram-status/80996

mrucki, 2020-10-03:
@MORD0R  but I checked previous versions of zigbee2mqtt (previous docker tags and same issue) and I don't have Openhabian but Raspbian Buster.
Currently OpenHab is running as a docker image and cannot see any zram installed in my OS

Koenkk, 2020-10-03:
@mrucki reading https://github.com/serialport/node-serialport/issues/1805, can you check if modemmanager is installed? https://www.zigbee2mqtt.io/information/FAQ.html#modemmanager-is-installed

mrucki, 2020-10-03:
@Koenkk nope - I checked and it wasn't installed.
Currently I reflashed again my RaspberryPi but this time with Ubuntu Server instead of Raspbian Buster.
```
ubuntu@ubuntu:~$ sudo apt-get purge modemmanager
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package 'modemmanager' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 49 not upgraded.
```
also here modemmanager is not installed - will keep you posted if there will be any issue in few hours 👍

mrucki, 2020-10-04:
As mentioned above tried with Ubuntu Server but still same error:
```
(node:17) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
(node:17) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:17) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

mrucki, 2020-10-04:
+ logs from Ubuntu Server when I set permit_join to false:

```
2020-10-04T13:41:50.068134966Z   serialport/stream _read reading { start: 2755, toRead: 62781 } +17ms
2020-10-04T13:41:50.068441266Z   serialport/binding-abstract read +3m
2020-10-04T13:41:50.068791524Z   serialport/bindings/unixRead Starting read +19ms
2020-10-04T13:41:50.074823101Z   serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
2020-10-04T13:41:50.074945652Z   errno: -11,
2020-10-04T13:41:50.074974089Z   code: 'EAGAIN',
2020-10-04T13:41:50.075000391Z   syscall: 'read'
2020-10-04T13:41:50.075026329Z } +1ms
2020-10-04T13:41:50.075341796Z   serialport/bindings/unixRead waiting for readable because of code: EAGAIN +1ms
2020-10-04T13:41:50.075416691Z   serialport/bindings/poller Polling for "readable" +22ms
2020-10-04T13:42:23.278849684Z Zigbee2MQTT:debug 2020-10-04 13:42:23: Saving state to file /app/data/state.json
2020-10-04T13:42:23.283361062Z winston:file: written true false
2020-10-04T13:42:23.286823280Z winston:file: logged 22757 debug 2020-10-04 13:42:23: Saving state to file /app/data/state.json
2020-10-04T13:42:23.286980935Z 
2020-10-04T13:47:23.290436292Z Zigbee2MQTT:debug 2020-10-04 13:47:23: Saving state to file /app/data/state.json
2020-10-04T13:47:23.290558166Z winston:file: written true false
2020-10-04T13:47:23.290592593Z winston:file: logged 22826 debug 2020-10-04 13:47:23: Saving state to file /app/data/state.json
2020-10-04T13:47:23.291050715Z 
2020-10-04T13:52:23.292702573Z Zigbee2MQTT:debug 2020-10-04 13:52:23: Saving state to file /app/data/state.json
2020-10-04T13:52:23.294951372Z winston:file: written true false
2020-10-04T13:52:23.297543504Z winston:file: logged 22895 debug 2020-10-04 13:52:23: Saving state to file /app/data/state.json
2020-10-04T13:52:23.297671681Z 
2020-10-04T13:57:23.302938675Z Zigbee2MQTT:debug 2020-10-04 13:57:23: Saving state to file /app/data/state.json
2020-10-04T13:57:23.304658721Z winston:file: written true false
2020-10-04T13:57:23.315475561Z winston:file: logged 22964 debug 2020-10-04 13:57:23: Saving state to file /app/data/state.json
2020-10-04T13:57:23.315606030Z 
2020-10-04T14:02:23.303500629Z Zigbee2MQTT:debug 2020-10-04 14:02:23: Saving state to file /app/data/state.json
2020-10-04T14:02:23.306220097Z winston:file: written true false
2020-10-04T14:02:23.309794300Z winston:file: logged 23033 debug 2020-10-04 14:02:23: Saving state to file /app/data/state.json
2020-10-04T14:02:23.309949977Z 
```

looks like serialport hangs, stops receiving

slaesh, 2020-10-05:
Did you tested the debug-fw which will make use of the LEDs? To see in which "state" the stick is.. Maybe any other resource is trying to access the serial-port and this may force the stick into reset-state.

Like this known bug: https://community.openhab.org/t/apparently-the-zwave-binding-blocks-the-dev-ttyusb0-port-in-combination-with-a-cc2652rb-zigbee2mqtt-dongle/103245

mrucki, 2020-10-05:
I didn't test it however once I installed fresh new Ubuntu Server on my Raspberry pi I added docker, docker-compose and run:

```
version: '3.3'
services:
  influxdb:
    image: influxdb:1.8.3
    restart: always
    ports:
      - '8086:8086'
    volumes:
      - /home/ubuntu/iot/influxdb-data:/var/lib/influxdb
      - /home/ubuntu/iot/influxdb-conf:/etc/influxdb
    environment:
      - INFLUXDB_ADMIN_USER=admin
      - INFLUXDB_ADMIN_PASSWORD=admin
  grafana:
    image: grafana/grafana:7.2.0
    restart: always
    depends_on:
        - influxdb
    ports:
      - '3000:3000'
    user: "1000"
    volumes:
      - /home/ubuntu/iot/grafana-data:/var/lib/grafana
      - /home/ubuntu/iot/grafana-conf:/etc/grafana
  mosquitto:
    image: eclipse-mosquitto:1.6.12
    restart: always
    ports:
      - '1883:1883'
      - '9001:9001'
  openhab:
    image: openhab/openhab:2.5.9
    restart: always
    depends_on:
        - mosquitto
        - influxdb
    ports:
      - '8080:8080'
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - /etc/timezone:/etc/timezone:ro
      - /home/ubuntu/iot/openhab-data/conf:/openhab/conf
      - /home/ubuntu/iot/openhab-data/userdata:/openhab/userdata
      - /home/ubuntu/iot/openhab-data/addons:/openhab/addons
      - /home/ubuntu/iot/openhab-data/backup:/openhab/backup
    environment:
      - USER_ID=1000
      - GROUP_ID=1000
  zigbee2mqtt:
    image: koenkk/zigbee2mqtt
    restart: always
    depends_on:
        - mosquitto
    devices:
      - "/dev/ttyACM0:/dev/ttyACM0"
    volumes:
      - /home/ubuntu/iot/zigbee2mqtt_data:/app/data
```

There is only one USB device and no EXTRA_JAVA_OPTS in  OpenHab (searched with docker exec -it openhab /bin/bash )

@slaesh what do you recommend for debug?

MORD0R, 2020-10-05:
> Did you tested the debug-fw which will make use of the LEDs? To see in which "state" the stick is.. Maybe any other resource is trying to access the serial-port and this may force the stick into [...]

Is disabled the z wave binding. And i think this error is gone. But im not sure. I have to wait a few days.

mrucki, 2020-10-07:
I setup such cron job:
```
0 */2 * * * docker-compose restart zigbee2mqtt >> /home/ubuntu/cron-jobs.log 2>&1
```
ugly hack but all 4 devices works

Dustinhoefer, 2020-10-08:
Hey everyone, I am the issue reporter.
Did @MORD0R find out if that did the trick?
@mrucki that is unfortunately not possible for me, since the devices aren't even working when I start the server ...

Any other ideas?

slaesh, 2020-10-08:
@mrucki ttyACM0 in your configuration looks weird? oO

you could use this python script to test the stick itself: https://github.com/slaesh/cc2652-stick/tree/master/coordinator-test

golddragon007, 2020-10-08:
@slaesh he has cc2531 as I understood well, that mounts to ttyACM0 by default. And I don't know if it will work the cc2652 test script for that chip. Especially because cc2531 doesn't have any serial chip like yours.

MORD0R, 2020-10-09:
> 
> 
> Hey everyone, I am the issue reporter.
> Did @MORD0R find out if that did the trick?
> @mrucki that is unfortunately not possible for me, since the devices aren't even working when I start the server ...
> 
> Any other ideas?

No it doesnt work error  is back.

@golddragon007 I have to use  /dev/ttyUSB0 and sometimes  /dev/ttyACM0 But i think thats because the pi 4 got an USB 3 port and USB2 port.

Use the command:ls /dev/*USB*

golddragon007, 2020-10-09:
@MORD0R it has nothing to do with the pi 4 USB ports. ttyUSBx means that the connected device communicates with "USB serial port adapter" the ttyACMx means "abstract control model" that is a sub-type of USB communication device (CDC). Also, there is the ttySx will be a serial port ttyAMAx is also serial interface, but ARM devices provide it as far as I know on pi 4 this is for Bluetooth if I'm right. The connection depends on the plugged-in device, and it's implementation, not on the port itself.

timdonovanuk, 2020-11-07:
My z2m just failed with this error all over the logs.

golddragon007, 2020-11-12:
@Koenkk I had this error now, with slaesh's stick. Here is my log: 
[network_fail.zip](https://github.com/Koenkk/zigbee2mqtt/files/5530456/network_fail.zip)

Koenkk, 2020-11-12:
@golddragon007 it seems the niko switches generate a lot of traffic which makes the coordinator crash. We should limit the traffic form these devices by configuring them correctly. 

You can start by changing the definition in devices.js to:

```js
{
    zigbeeModel: ['Connected socket outlet'],
    model: '170-33505',
    vendor: 'Niko',
    description: 'Connected socket outlet',
    supports: 'on/off, power measurement',
    fromZigbee: [fz.on_off, fz.electrical_measurement_power],
    toZigbee: [tz.on_off],
    meta: {configureKey: 5},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
        await readEletricalMeasurementPowerConverterAttributes(endpoint);
        await configureReporting.activePower(endpoint, {min: 5, max: 3600, change: 1});
        await configureReporting.rmsCurrent(endpoint, {min: 5, max: 3600, change: 1});
        await configureReporting.rmsVoltage(endpoint, {min: 5, max: 3600, change: 1});
    },
    exposes: [e.switch(), e.power(), e.current(), e.voltage()],
},
```

and trigger a reconfigure via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure

After this there will be a lot of reports, if e.g. the power is still reported every 5 seconds, multiple the change by 10, e.g. `await configureReporting.activePower(endpoint, {min: 5, max: 3600, change: 10});`. Repeat this until the device only reports it's power only when there is significant change.

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

golddragon007, 2020-11-12:
@Koenkk Well, I've just tried now, and it's still reported every 0-5 seconds/outlet. Even if I set to 1000 the change value... And a bulk reconfigure would nice now.

Koenkk, 2020-11-13:
@golddragon007 

- can you also try change 10000 and 100000?
- if that doesn't help, can you try changing the min to a very high value (like 60), that means the device should not report more frequent than 60 seconds
- I also suggest to switch to this firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin as it will increase the transmit power of the coordinator from 0dbm to 5dbm.

golddragon007, 2020-11-14:
@Koenkk I've changed both values and also reflashed with the new firmware and changed the values, send the message for configuration, but I don't see any change.

Koenkk, 2020-11-14:
We have to try something different then, can you check with the config below. This should greatly reduce the load on the coordinator

```
{
    zigbeeModel: ['Connected socket outlet'],
    model: '170-33505',
    vendor: 'Niko',
    description: 'Connected socket outlet',
    supports: 'on/off, power measurement',
    fromZigbee: [fz.on_off, fz.electrical_measurement_power],
    toZigbee: [tz.on_off],
    meta: {configureKey: 5},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
        await readEletricalMeasurementPowerConverterAttributes(endpoint);
        await configureReporting.activePower(endpoint, {min: 5, max: 3600, change: 1000});
        await configureReporting.rmsCurrent(endpoint, {min: 5, max: 3600, change: 1000});
        await configureReporting.rmsVoltage(endpoint, {min: 5, max: 3600, change: 1000});
    },
    exposes: [e.switch(), e.power(), e.current(), e.voltage()],
    onEvent: async (type, data, device) => {
        if (type === 'message') {
            data.meta.frameControl.disableDefaultResponse = true;
        }
    },
},
```

golddragon007, 2020-11-14:
I've added the onEvent part, however, the messages still arriving in the same was.
I've run the supervised HA and I've used the addon store z2m, which is in a docker container. I've changing there the file, but I just hope it takes into effect...

Koenkk, 2020-11-15:
@golddragon007 this will not prevent the messages from being send (there does not seem a way to do it, these Zigbee devices do not seem to care about the configure reporting), but it will prevent Zigbee2MQTT from sending an acknowledgement to every message which is were the heavy load came from (I expect)

golddragon007, 2020-11-15:
It still crashing:
```
2020-11-15T11:33:39.914Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228]
(node:245) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.16.1/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
(node:245) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:245) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
2020-11-15T11:36:59.911Z zigbee-herdsman:controller:log Permit joining
2020-11-15T11:36:59.912Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0}
2020-11-15T11:36:59.912Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228]
```

[network-crash.zip](https://github.com/Koenkk/zigbee2mqtt/files/5542344/network-crash.zip)

Koenkk, 2020-11-15:
Seems the default responses are still being send (which should be suppressed by https://github.com/Koenkk/zigbee2mqtt/issues/4398#issuecomment-727253598). Can you try with:

```js
{
    zigbeeModel: ['Connected socket outlet'],
    model: '170-33505',
    vendor: 'Niko',
    description: 'Connected socket outlet',
    supports: 'on/off, power measurement',
    fromZigbee: [fz.on_off, fz.electrical_measurement_power],
    toZigbee: [tz.on_off],
    meta: {configureKey: 5},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
        await readEletricalMeasurementPowerConverterAttributes(endpoint);
        await configureReporting.activePower(endpoint, {min: 5, max: 3600, change: 1000});
        await configureReporting.rmsCurrent(endpoint, {min: 5, max: 3600, change: 1000});`
        await configureReporting.rmsVoltage(endpoint, {min: 5, max: 3600, change: 1000});
    },
    exposes: [e.switch(), e.power(), e.current(), e.voltage()],
    onEvent: async (type, data, device) => {
        if (type === 'message') {
            console.log("HELLO, disabling default response");
            data.meta.frameControl.disableDefaultResponse = true;
        }
    },
},
```

and confirm that each time a message is received: `HELLO, disabling default response` is logged?

golddragon007, 2020-11-15:
No, there's no log message. It seems like the changes don't have any effect. As I running z2m from docker, how can I restart the service without restarting docker? Because if I restart docker all my changes will disappear...

[8 more replies]

closed 2020-12-31 by github-actions[bot]