Koenkk/zigbee2mqtt #4401: Stops working, only way to fix is unplug/plug cc2531 and restart service - on hassio/proxmox

stevetz, 2020-09-19:
## What happened/expect/reproduce
Zigbee devices stop responding in home assistant after a period of time. Z2M was running solidly for 3 months, with no disconnects or any issues as far as I could tell. I moved house a month ago, but otherwise no problems when reinstalling all the devices here. This issue first occured two weeks ago (3 weeks after moving - where everything was fine), so I'm not sure the issue is the fault of the changed physical network layout etc...but it could be?

In order to fix this, I must unplug and replug the CC2531 stick and restart the service (through home assistant frontend, restart the addon). As far as I can tell the device is showing as connected to the proxmox node, and the green LED remains lit.

The issue repeats every 24-36 hours, I haven't yet figured out if its a regular pattern

I've enabled debug logging, here is a section of the most recent log: [https://pastebin.com/UTU4pucK](https://pastebin.com/UTU4pucK). I'm not sure if it helps?

In troubleshooting, theres quite a few things that could be the cause: I hope I can eliminate z2m first, then the addon, then hasso, then proxmox... Maybe a reflash of the stick at some point - though I did it through a Pi which ive since repurposed, so the effort of flashing an OS sent me here in the first instance. Thank you!!

## Debug info
**Zigbee2MQTT version**: 1.14.4.1 via danielwelch's hassio addoin
**Adapter hardware**: CC2531
**Adapter firmware version**: Sorry I don't know how to find this? (I flashed latest version in early May 2020, the 1.2 coordiator version)
**Host**: Home Assistant OS, supervisor 245, HA 0.115, running on a Proxmox 6 node the USB device is passed through to the vm.

Koenkk, 2020-09-19:
Known issue, should be fixed by updating to the latest dev branch (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html), duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/4307

closed 2020-09-19 by Koenkk

stevetz, 2020-09-19:
Sorry I didn't find the other issue, must have failed my searching class!

This has also spurred me on to (finally) give a small paypal gift. Thank you for all you do @Koenkk !

Koenkk, 2020-09-19:
No problem, please let me know if after upgrading the issue is solved.

stevetz, 2020-09-19:
I'll report back in a couple of days if there have been no further crashes.

stevetz, 2020-09-22:
So far so good... I've not experienced any crashes or errors since switching to the dev branch - at 9am GMT on 19th - so 73 hours of uptime! (via the [home assistant z2m edge](https://github.com/danielwelch/hassio-zigbee2mqtt/tree/master/zigbee2mqtt-edge)).
Thanks!

lukasmedia, 2023-11-12:
Hi, I am also experiencing this problem now. Is there anything I can try?