Koenkk/zigbee2mqtt #4405: Bindings wont work

b-reich, 2020-09-19:
## What happened
I try to controll some Ikea light bulbs with a Ikea E1810 or E1524 Remote (both 5 buttons). 

## What did you expect to happen
Toggle the blub state, change color and brightness

## How to reproduce it (minimal and precise)
```
"zigbee2mqtt/bridge/unbind/Schlafzimmer Schalter" with payload "Schlafzimmer Lampen" (Schlafzimmer Lampen is a group in zigbee2mqtt, direkt binding also wont work)
```

## Debug info
Zigbee2MQTT version: 1.14.4 (docker latest and latest-dev)
Adapter hardware: CC2531
Adapter firmware version: `{'meta': {'maintrel': 3, 'majorrel': 2, 'minorrel': 6, 'product': 0, 'revision': 20190608, 'transportrev': 2}, 'type': 'zStack12'}`

b-reich, 2020-09-19:
```
zigbee2mqtt    | Zigbee2MQTT:info  2020-09-19 14:42:57: MQTT publish: topic 'zigbee2mqtt/Schlafzimmer Schalter/action', payload 'toggle'
zigbee2mqtt    | Zigbee2MQTT:info  2020-09-19 14:43:00: Successfully bound cluster 'genLevelCtrl' from 'Schlafzimmer Schalter' to 'Schlafzimmer Lampen'
zigbee2mqtt    | Zigbee2MQTT:info  2020-09-19 14:43:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"cluster":"genLevelCtrl","from":"Schlafzimmer Schalter","to":"Schlafzimmer Lampen"},"type":"device_bind"}'
```

hoggerz, 2020-09-19:
b-reich try unbinding the remote from default_bind_group and then bind it to your lights/group again.

I have a possibly related issue, Ikea GU10 LED1537R6/LED1739R5 bulbs (white spectrum), previously allowed an Ikea E1810 dimmer to control on/off, dimming & changing colour temperature, now I've re paired them all and bound the remote to the group, it only supports on/off and dimming, the colour temperature control seems to have been lost.

Hardware is ZZH with latest coordinator firmware, and latest dev branch of zigbee2mqtt as of today.
Version of Zigbee2Mqtt: 1.14.4-dev
Coordinator version: 20200830

MattWestb, 2020-09-19:
IKEA Zigbee 3 controlling devices (not the old motion sensor LL) need first being unbinded before accepting one binding. Read the wiki for the device for how to do.
For the moment is the color changing not working with the remote (the default scenes config is broken in Z2M).

hoggerz, 2020-09-19:
Ah, that explains my issue, thanks.

b-reich, 2020-09-19:
@hoggerz thanks
@MattWestb do you know when the color change is on the roadmap?

closed 2020-09-19 by b-reich