Koenkk/zigbee2mqtt #4407: RTCGQ01LM can only pair to the coordinator

mr2c12, 2020-09-19:
I own a bunch of Xiaomi aqara motion sensor with illuminance sensor (RTCGQ11LM) but I suffer from the known issue with osram/philips/inner bulbs (sometimes the motion message is not forwarded to the coordinator).

So I decided to buy a couple of RTCGQ01LM, which is the previous version, without a illumance sensor.

The problem with this device is that it *cannot* pair to something different from the coordinator. With RTCGQ11LM, once it joined the network, I can move it to its final position, single press its pairing button et voilà it connects to the nearest router.
When it comes to RTCGQ01LM, I can pair it only where there is good connection to the coordinator. But once moved, it never reconnect to a different router.
The problem is when its position is on the edge of the coordinator coverage and, from time to time, it stops working.

I suppose that there is nothing that z2m can do. Is it true?
Does anybody else use the RTCGQ01LM version? Are you able to use it far away from the coordinator.

For what it matters, I am unable to pair the RTCGQ01LM even being 2cm close to a cc2530+cc2591 router or a Xiaomi Smart Plug router.

Koenkk, 2020-09-19:
can you sniff close to the xiaomi plug when trying to pair the sensor close to it?

mr2c12, 2020-09-20:
I am a bit lost. Today I was able to pair both RTCGQ01LM and RTCGQ11LM in the same position (really close to the Xiaomi smart plug).
But it seems that the both of them use the coordinator as direct father, am I right?
Attached the sniffed log from wireshark. The zip is password protected. I will send the passkey with telegram.

[20200920_report.zip](https://github.com/Koenkk/zigbee2mqtt/files/5252183/20200920_report.zip)

Koenkk, 2020-09-20:
yes, network map is reliable.

mr2c12, 2020-09-24:
I successfully captured two different scenario:

- RTCGQ01LM pairing to the coordinator (200924_success.pcapng)
- RTCGQ01LM failed to pair far from the coordinator, but close to both a Philips Hue and a IKEA tradfiri signal repeater (200924_failed.pcapng). I tried 3 times to pair the device.
Here is the report:
[20200924_report.zip](https://github.com/Koenkk/zigbee2mqtt/files/5279110/20200924_report.zip)

I'm not very good in traffic sniffing and I am not able to identify what really change between the two scenario.
At package number 85 in the failed log, I see:

> ----
> # Packet 84 from 200924_failed.pcapng
> - 85
> - 15.992867
> - 00:17:88:01:06:01:8f:9f
> - 00:15:8d:00:02:c5:a1:00
> - IEEE 802.15.4
> - 27
> - Association Response, PAN: 0x1a64 Addr: 0x9fcd

Than a lot of "Data request" from RTCGQ01LM to the Philips bulb..
..but actually the RTCGQ01LM in a few seconds made a long blink that represent the failure.

I suspect there is some "incompatibility issue" with RTCGQ01LM and non-xiaomi or non-standard (cc2530 and so on) routers.

Archive password has been sent via telegram on September the 20th.
Please let me know if you need something more info.

Koenkk, 2020-09-25:
The reason I doesn't join is because the hue bulb does not send the key ("transport key") message.

mr2c12, 2020-09-25:
Ok so I made some test, interesting results

Router: 8718699673147 | Philips Hue white A60 bulb E27 bluetooth (on/off, brightness, power-on behavior)
End devices: RTCGQ01LM, RTCGQ11LM, E1525/E1745 (IKEA TRADFRI motion sensor)
No one of the devices above is able to pair

Router: Osram light bulb (color temperature)
End devices: RTCGQ01LM, RTCGQ11LM
Both of the devices above is able to pair

So it seems to me that the problem come with philips hue light bulb. I will probably need to swap the osram and philips light bulbs.

mr2c12, 2020-09-26:
I swapped the philips bulbs with the osram ones.
Seems that philips bulbs act strange as router:
- end devices can't use them to pair
- osram bulbs can't use them as routers

Now I have changed the topology of my network (at least, the part interesting the garden) and see if now I can get a stable connection with the RTCGQ01LM.

closed 2020-11-04 by github-actions[bot]