Koenkk/zigbee2mqtt #4408: Retry logic for MQTT availability on startup

andrasg, 2020-09-19:
## Is your feature request related to a problem? Please describe
I have zigbee2mqtt running in docker on a pi4. The same docker host is hosting mosquitto, the MQTT broker. By the time zigbee2mqtt is booted up, the MQTT server is not yet available.

## Describe the solution you'd like
Some retry logic in the initial MQTT connection would solve this issue. As I have experienced, there is a retry/reconnection logic if MQTT broker is lost after bootup.
Alternatively some option to delay the boot would also work.

## Describe alternatives you've considered
Moving the MQTT broker to some other host is a workaround I could utilize.

Koenkk, 2020-09-20:
Zigbee2MQTT already tries to reconnect, does this not work in your case? Can you provide the debug logging?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

andrasg, 2020-09-20:
Sorry my bad, forgot the `--restart=unless-stopped` switch from docker, that's why it appeared to be stopping on reboot.

closed 2020-09-20 by andrasg