Koenkk/zigbee2mqtt #4410: Error on retrieve devices in new deCONZ firmware

czm97176, 2020-09-20:
# Bug Report
## What happened
Unable to retrieve device information from homebridge plugin or zigbee2mqtt assistant.

[image]

## What did you expect to happen

Able to retrieve device .

## How to reproduce it (minimal and precise)

Run zigbee2mqtt assistant or homebridge - zigbee2mqtt plugin

## Debug Info
Zigbee2MQTT version:
Adapter hardware: deCONZ
Adapter firmware version: 26650700

http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26650700.bin.GCF

Koenkk, 2020-09-20:
Seems `await this.zigbee.getCoordinatorVersion();` returns null, @chrishae could you look at this?

ChrisHae, 2020-09-22:
I think it is because of the new firmware where the command to read the fw version has changed. I adapted the code of the zigbee-herdsman deconz adapter last friday. It should be in the latest dev-release of zigbee-herdsman. So I don't know if you have already the latest code.

czm97176, 2020-09-22:
Thanks @ChrisHae @Koenkk 

i am running on a docker which version is 1.14.4 . so i think it doesn't not contain the latest dev-release.
i will wait. Thank you

closed 2020-10-31 by github-actions[bot]