Koenkk/zigbee2mqtt #4412: Cover position issues with >=1.14.3 version

vytautasgimbutas, 2020-09-20:
# Bug Report
## What happened
I'm using HomeAssistant (0.115.0) with zigbee2mqtt.

Issuing open/close cover in HomeAssistant works fine, but when I try to set currently opened curtains position to 90% via HomeAssistant zigbee2mqtt actually sends 10% to curtain motor. Final position report coming from zigbee2mqtt is 10%.

Setting cover position started working incorrectly after upgrading from 1.14.2 to to 1.14.3 on `Zemismart ZM79E-DT` Curtain Motor. Latest version is also affected but by upgrading one version at a time I've managed to find a release that has introduced this bug.

## What did you expect to happen
- Send 95% position to curtain motor

## How to reproduce it (minimal and precise)
- Open curtains
- Set cover position in home hub to 90%
- Zigbee2MQTT 1.14.3+ it'll try to set curtains position to 10%
```
Zigbee2MQTT:info  2020-09-20 12:51:50: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffed2d261', payload '{"running":true,"linkquality":57,"position":100}'
Zigbee2MQTT:info  2020-09-20 12:51:50: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffed2d261', payload '{"running":true,"linkquality":57,"position":100}'

Zigbee2MQTT:info  2020-09-20 12:52:00: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffed2d261', payload '{"running":false,"linkquality":63,"position":10}'
```

Zigbee2MQTT <=1.14.2 works fine:
```
zigbee2mqtt:info  2020-09-20 12:57:21: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffed2d261', payload '{"running":true,"linkquality":84,"position":100}'
zigbee2mqtt:info  2020-09-20 12:57:26: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffed2d261', payload '{"running":false,"linkquality":75,"position":90}'
```

## Debug Info
Zigbee2MQTT version: 0.114.3+
Adapter hardware: both CC2531 and zzh! 
Adapter firmware version:

Koenkk, 2020-09-20:
Fixed!

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-09-20 by Koenkk

vytautasgimbutas, 2020-10-03:
@Koenkk for some reason again it started working incorrectly. Using just Open/Close works fine. Once curtains are closed or opened they report position correctly (0 - closed, 100 - open). However, if I try to set position to let's say 80 it inteprets it as 20.
```
Zigbee2MQTT:debug 2020-10-03 09:54:09: Received MQTT message on 'zigbee2mqtt/0xec1bbdfffed2d261/set' with data '{ "position": 86 }' <--- i try to set it to "86"
Zigbee2MQTT:debug 2020-10-03 09:54:09: Publishing 'set' 'position' to '0xec1bbdfffed2d261'
Zigbee2MQTT:debug 2020-10-03 09:54:09: Received MQTT message on 'zigbee2mqtt/0xec1bbdfffed2d261/set' with data '{ "position": 86 }'
Zigbee2MQTT:debug 2020-10-03 09:54:09: Publishing 'set' 'position' to '0xec1bbdfffed2d261'
Zigbee2MQTT:debug 2020-10-03 09:54:09: Received Zigbee message from '0xec1bbdfffed2d261', type 'commandGetData', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[0,0,0,86],"type":"Buffer"},"dp":514,"fn":0,"status":0,"transid":121}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-10-03 09:54:09: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffed2d261', payload '{"linkquality":147,"position":100,"running":true}'
Zigbee2MQTT:debug 2020-10-03 09:54:09: Received Zigbee message from '0xec1bbdfffed2d261', type 'commandGetData', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[0,0,0,86],"type":"Buffer"},"dp":514,"fn":0,"status":0,"transid":220}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-10-03 09:54:09: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffed2d261', payload '{"linkquality":147,"position":100,"running":true}'
Zigbee2MQTT:debug 2020-10-03 09:54:09: Received Zigbee message from '0xec1bbdfffed2d261', type 'commandGetData', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[0,0,0,86],"type":"Buffer"},"dp":514,"fn":0,"status":0,"transid":220}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-10-03 09:54:09: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffed2d261', payload '{"linkquality":150,"position":100,"running":true}'
Zigbee2MQTT:debug 2020-10-03 09:54:17: Received Zigbee message from '0xec1bbdfffed2d261', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[0,0,0,90],"type":"Buffer"},"dp":515,"fn":0,"status":0,"transid":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-10-03 09:54:17: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffed2d261', payload '{"linkquality":147,"position":10,"running":false}' <------ final position
```

I've just completely recloned zigbee2mqtt (most recent master commit), reflashed zzh us stick, repaired curtains motor, recalibrated but that didn't help. Could this be my curtain motor malfunctioning?

Koenkk, 2020-10-03:
By default for Zigbee2MQTT 100 = open, 0 = closed so the behaviour you see is correct. You can invert this with the `invert_cover` option: https://www.zigbee2mqtt.io/devices/TS0601_curtain.html#device-type-specific-configuration

DomiStyle, 2020-10-08:
@Koenkk Something seems wrong with this update, setting to 100% closes the curtain, setting to 100% again opens it again and so on. The reported state switches between 100% and 0% after the curtain arrives.

Koenkk, 2020-10-08:
@DomiStyle can you provide the herdsman debug logging of this?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

DomiStyle, 2020-10-08:
@Koenkk Sure: https://gist.github.com/DomiStyle/7f3c19839f418bbbbdc2c50e654083a0

Probably some other stuff in there but should be mostly the curtain. Commands came from Home Assistant (100%), keeps flip-flopping between open/closed.
Same thing happens from the new Zigbee2MQTT UI.

Koenkk, 2020-10-09:
@DomiStyle when sending position 100, does the curtain open or close. 

Likely caused by https://github.com/Koenkk/zigbee-herdsman-converters/pull/1603, @dinhchinh82  looking at the PR now, I would also expect that an inversion needs to be done in the `fromZigbee` converter. I think https://github.com/Koenkk/zigbee-herdsman-converters/blob/144f32abcb3a1f6083afb3262dc468f94d6cb965/converters/fromZigbee.js#L5720 should be `position = options.invert_cover ? 100 - position : position;`

DomiStyle, 2020-10-09:
@Koenkk Both. Clicking it once opens it, clicking it again closes it and so on.