Koenkk/zigbee2mqtt #4413: Please help adding Lerlink/TuYa dimmer

Vinndimon, 2020-09-20:
Lerlink rotary dimmer for TuYa. [AliExpress](https://aliexpress.ru/item/4000426724333.html?spm=a2g0s.9042311.0.0.335e33edzScrl8&_ga=2.242117889.240655884.1600519534-230377271.1591693236)

```
{
    "id": 7,
    "type": "Router",
    "ieeeAddr": "0xec1bbdfffe66c4d7",
    "nwkAddr": 11298,
    "manufId": 4098,
    "manufName": "_TZE200_579lguh2",
    "powerSource": "Mains (single phase)",
    "modelId": "TS0601",
    "epList": [1],
    "endpoints": {
        "1": {
            "profId": 260,
            "epId": 1,
            "devId": 81,
            "inClusterList": [0, 10, 4, 5, 61184],
            "outClusterList": [25],
            "clusters": {
                "genBasic": {
                    "attributes": {
                        "modelId": "TS0601",
                        "manufacturerName": "_TZE200_579lguh2",
                        "powerSource": 1,
                        "zclVersion": 3,
                        "appVersion": 82,
                        "stackVersion": 0,
                        "hwVersion": 1,
                        "dateCode": ""
                    }
                }
            },
            "binds": [{
                "cluster": 6,
                "type": "endpoint",
                "deviceIeeeAddress": "0x00124b0014d90850",
                "endpointID": 1
            }, {
                "cluster": 8,
                "type": "endpoint",
                "deviceIeeeAddress": "0x00124b0014d90850",
                "endpointID": 1
            }]
        }
    },
    "appVersion": 82,
    "stackVersion": 0,
    "hwVersion": 1,
    "dateCode": "",
    "zclVersion": 3,
    "interviewCompleted": true,
    "meta": {
        "configured": 1
    },
    "lastSeen": 1600589776173
}
```
Tried to use standard TuYa dimmer converter this way:
```
 {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_579lguh2'}],
        model: 'TS0601',
        vendor: 'Lerlink',
        description: 'Smart dimmer',
        fromZigbee: [fz.tuya_dimmer, fz.ignore_basic_report],
        toZigbee: [tz.tuya_dimmer_state, tz.tuya_dimmer_level],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        },
    },
```

It can switch on/off, but it works quite unstable: sometimes the device does not react on z2m payloads. It also can correctly read brightness level from the device, but setting brightness does not work. Please help.

Koenkk, 2020-09-20:
Can you try: 
```js
{
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_579lguh2'}],
    model: 'TS0601',
    vendor: 'Lerlink',
    description: 'Smart dimmer',
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
    },
    extend: generic.light_onoff_brightness,
},
```

Vinndimon, 2020-09-20:
> ```js
>     extend: generic.light_onoff_brightness,
> ```

Unfortunately, this hasn't changed the dimmer behaviour: still doesn't toggle sometimes and I cannot write brightness, only read it from the device.

Koenkk, 2020-09-20:
Unfortunately the only way to figure out how to send the brightness is by sniffing the traffic with the original bridge, seems that this device use yet another slightly different protocol (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

Vinndimon, 2020-09-20:
> Unfortunately the only way to figure out how to send the brightness is by sniffing the traffic with the original bridge, seems that this device use yet another slightly different protocol (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

So, I need an original bridge for that? No way to emulate it? The problem is that I don't have one : (

Koenkk, 2020-09-20:
Yes you need that, we need to see what the original bridge sends so we can copy that.

Vinndimon, 2020-09-20:
> Yes you need that, we need to see what the original bridge sends so we can copy that.
Unfortunately I can't have it now. Let's hope someone else with the original bridge will buy this dimmer too.

MUzair030, 2020-10-01:
### **[Solved: TS0601]**
Add the following in `toZigbee.js` located in `zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters`:
```
uzi_dimmer_control: {
        key: ['state', 'brightness'],
        convertSet: async (entity, key, value, meta) => {
            if (key === 'brightness') {
                if (value >= -254 && value <= 0) {
                    const invert = !(meta.mapped.meta && meta.mapped.meta.coverInverted ?
                        !meta.options.invert_cover : meta.options.invert_cover);
                    value = invert ? 0 - value : value;
		    let Val;
		    newVal = Math.round(Number(value) * 1000 / 255);
		    const val1 = Val >> 8;
		    const val2 = Val & 0xFF;
		    sendTuyaCommand(entity, 514, 0, [4, 0, 0, val1, val2]);

                } else {
                    meta.logger.debug('Dimmer brightness is out of range');
                }
            }
        },
    },
``` 

Add `tz.uzi_dimmer_control` in devices.js under zigbee model TS0601.
**Note:** **_Use negative range to control brightness (0 to -254)_**

Vinndimon, 2020-10-01:
> tz.uzi_dimmer_control

Unfortunately, no effect. Still cannot control brightness via Home Assistant and on/off functionality is still quite unstable.

Vinndimon, 2020-10-01:
Also z2m shows this error:
```
Zigbee2MQTT:error 2020-10-01 15:35:50: Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'key' of undefined
    at /zigbee2mqtt-1.14.4/lib/extension/publish.js:188:56
    at Array.find (<anonymous>)
    at EntityPublish.onMQTTMessage (/zigbee2mqtt-1.14.4/lib/extension/publish.js:188:42)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.14.4/lib/controller.js:365:44)
    at Controller.onMQTTMessage (/zigbee2mqtt-1.14.4/lib/controller.js:262:14)
    at MQTT.emit (events.js:315:20)
    at MQTT.onMessage (/zigbee2mqtt-1.14.4/lib/mqtt.js:95:14)
    at MqttClient.emit (events.js:315:20)
    at MqttClient._handlePublish (/zigbee2mqtt-1.14.4/node_modules/mqtt/lib/client.js:1271:12)
    at MqttClient._handlePacket (/zigbee2mqtt-1.14.4/node_modules/mqtt/lib/client.js:410:12))
```

viperet, 2020-10-31:
@Vinndimon I have similar dimmer (but not exactly the same as yours), and code by @MUzair030 helped me to make it work. You can try this https://github.com/Koenkk/zigbee-herdsman-converters/compare/master...viperet:master and check if it will work for your dimmer. You will probably have to add your manufacturerName to devices.js, 'EDM_1ZAA' dimmer configuration. My dimmer https://www.aliexpress.com/item/1005001505069156.html

closed 2020-12-08 by github-actions[bot]