Koenkk/zigbee2mqtt #4415: device_options: temperature_precision not working

MadDoct, 2020-09-20:
## What happened
device_options:
  temperature_precision: 1

isn't working - I still get 2 decimals
## What did you expect to happen
only one decimal
## How to reproduce it (minimal and precise)
include the above in the config
## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2538
Adapter firmware version:

Koenkk, 2020-09-20:
Make sure the indentation is correct (2 spaces before `temperature_precision`).

Also for what device is this?

MadDoct, 2020-09-20:
Yes, it is... Anyway, it wouldn't start if it was wrong... There's no error in the log.
The devices are mijia and aqara th and thp sensors

Koenkk, 2020-09-20:
- Can you provide your `configuration.yaml` (leaving out passwords)
- The debug logging when the sensor reports it's temperature.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

MadDoct, 2020-09-21:
It is now solved... For some reason, I also had humidity_precision and that seemed to be the problem. I removed it and it is now as it should... Isn't humidity_precision, an option?

MadDoct, 2020-09-21:
Strangely, after changing it to 0 or 2, It stays with only 1 decimal... Very very strange...

Koenkk, 2020-09-21:
You have to wait until a new report comes in from the device until is taken into effect.

closed 2020-10-29 by github-actions[bot]

delicon, 2021-08-04:
I have the same problem.
I want z2m to round the temperature unit to 1 decimal, but it always shows 2 decimals.

Here is my device config:
`'0x00158d00031b0611':
  friendly_name: livingroom_temperature_sensor
  debounce: 1
  qos: 1
  retain: true
  device_options:
    temperature_precision: 1
    humidity_precision: 1
    pressure_precision: 1
  optimistic: true
  retention: 900`

Here is the debug logs:
`Zigbee2MQTT:debug 2021-08-04 14:32:39: Received Zigbee message from 'livingroom_temperature_sensor', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2622}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-08-04 14:32:39: Received Zigbee message from 'livingroom_temperature_sensor', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":4777}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-08-04 14:32:39: Received Zigbee message from 'livingroom_temperature_sensor', type 'attributeReport', cluster 'msPressureMeasurement', data '{"measuredValue":976,"scale":-1,"scaledValue":9766}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-08-04 14:32:40: MQTT publish: topic 'zigbee2mqtt/livingroom_temperature_sensor', payload '{"battery":100,"humidity":47.77,"linkquality":75,"pressure":976,"temperature":26.22,"voltage":3015}'`