Koenkk/zigbee2mqtt #4418: Unable to start up (after update)

Davincible, 2020-09-20:
I have auto update turned on, and worked perfectly untill this week. Now when I try to start, I get ```"Missing option availability_blocklist in advanced."```

And if I try to save the settings, I get:
```
Failed to save addon configuration, not a valid value for dictionary value @ data['options']. Got {'data_path': '/share/zigbee2mqtt', 'external_converters': [], 'devices': 'devices.yaml', 'groups': 'groups.yaml', 'homeassistant': True, 'permit_join': True, 'mqtt': {'base_topic': 'zigbee2mqtt', 'server': 'mqtt://core-mosquitto', 'user': '.....', 'password': '.......'}, 'serial': {'port': '/dev/ttyACM0', 'disable_led': True}, 'advanced': {'pan_id': 6754, 'channel': 11, 'network_key': [114, .... 173]}, 'device_options': {}, 'block...
```
My Config:
```
data_path: /share/zigbee2mqtt
external_converters: []
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://core-mosquitto'
  user: .....
  password: ....
serial:
  port: /dev/ttyACM0
  disable_led: true
advanced:
  pan_id: 6754
  channel: 11
  network_key:
    - 114
    - .....
    - 173
  availability_blacklist: []
device_options: {}
blocklist: []
passlist: []
queue: {}
experimental: {}
socat:
  enabled: false
  master: 'pty,raw,echo=0,link=/dev/ttyZ2M,mode=777'
  slave: 'tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5'
  restartdelay: 1
  initialdelay: 1
  options: '-d -d'
  log: false
ban: []
whitelist: []

```

cjd, 2020-09-21:
Change availability_blacklist to availability_blocklist

Koenkk, 2020-09-21:
Dupe of #4391

closed 2020-09-21 by Koenkk