Koenkk/zigbee2mqtt #4419: error setting auto mode on essentials GS361A-H04

frlp542, 2020-09-20:
hello Guys,

I receive the following error in my log files when setting the mode on my TRV.

can anybody help with this how I troubleshoot /resolve this ?

Thanks

Filip

debug 2020-09-20 23:31:13: Received MQTT message on 'zigbee2mqtt/0x14b457fffe3ce248/set/system_mode' with data 'auto'
debug 2020-09-20 23:31:13: Publishing 'set' 'system_mode' to '0x14b457fffe3ce248'
debug 2020-09-20 23:31:15: Received Zigbee message from '0x14b457fffe3ce248', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[1],"type":"Buffer"},"dp":1028,"fn":0,"status":0,"transid":222}' from endpoint 1 with groupID 0
error 2020-09-20 23:31:15: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'hasOwnProperty' of undefined
    at Object.tuyaThermostat [as convert] (/app/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js:366:89)
    at /app/lib/extension/receive.js:167:41
    at Array.forEach (<anonymous>)
    at Receive.onZigbeeEvent (/app/lib/extension/receive.js:166:20)
    at Controller.callExtensionMethod (/app/lib/controller.js:371:44)
    at Controller.onZigbeeEvent (/app/lib/controller.js:260:14)
    at Zigbee.emit (events.js:315:20)
    at Controller.<anonymous> (/app/lib/zigbee.js:68:52)
    at Controller.emit (events.js:315:20)
    at Controller.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/controller.js:544:22))

Koenkk, 2020-09-21:
Can you try with the latest dev branch and provide the logging if this still fails? 

https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

frlp542, 2020-09-21:
Hi Koen,

I was already running the dev branch but tested it again to be sure.
for your info - the TRV sends it's feedback 2 times but that appears to be normal.

Greetings

info  2020-09-21 18:32:01: Starting Zigbee2MQTT version 1.14.4-dev (commit #c0faae7)
...

debug 2020-09-21 18:33:17: Received MQTT message on 'zigbee2mqtt/0x14b457fffe3ce248/set/system_mode' with data 'auto'
debug 2020-09-21 18:33:17: Publishing 'set' 'system_mode' to '0x14b457fffe3ce248'
debug 2020-09-21 18:33:19: Received Zigbee message from '0x14b457fffe3ce248', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[1],"type":"Buffer"},"dp":1028,"fn":0,"status":0,"transid":54}' from endpoint 1 with groupID 0
error 2020-09-21 18:33:19: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'hasOwnProperty' of undefined
    at Object.tuyaThermostat [as convert] (/app/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js:351:89)
    at /app/lib/extension/receive.js:167:41
    at Array.forEach (<anonymous>)
    at Receive.onZigbeeEvent (/app/lib/extension/receive.js:166:20)
    at Controller.callExtensionMethod (/app/lib/controller.js:371:44)
    at Controller.onZigbeeEvent (/app/lib/controller.js:260:14)
    at Zigbee.emit (events.js:315:20)
    at Controller.<anonymous> (/app/lib/zigbee.js:68:52)
    at Controller.emit (events.js:315:20)
    at Controller.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/controller.js:544:22))
debug 2020-09-21 18:33:20: Received Zigbee message from '0x14b457fffe3ce248', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[1],"type":"Buffer"},"dp":1028,"fn":0,"status":0,"transid":54}' from endpoint 1 with groupID 0
error 2020-09-21 18:33:20: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'hasOwnProperty' of undefined
    at Object.tuyaThermostat [as convert] (/app/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js:351:89)
    at /app/lib/extension/receive.js:167:41
    at Array.forEach (<anonymous>)
    at Receive.onZigbeeEvent (/app/lib/extension/receive.js:166:20)
    at Controller.callExtensionMethod (/app/lib/controller.js:371:44)
    at Controller.onZigbeeEvent (/app/lib/controller.js:260:14)
    at Zigbee.emit (events.js:315:20)
    at Controller.<anonymous> (/app/lib/zigbee.js:68:52)
    at Controller.emit (events.js:315:20)
    at Controller.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/controller.js:544:22))
debug 2020-09-21 18:33:20: Received Zigbee message from '0x14b457fffe3ce248', type 'commandGetData', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[0,0,0,215],"type":"Buffer"},"dp":514,"fn":0,"status":0,"transid":108}' from endpoint 1 with groupID 0
info  2020-09-21 18:33:20: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe3ce248', payload '{"battery":97,"child_lock":"UNLOCKED","current_heating_setpoint":"21.5","linkquality":66,"local_temperature":"23.7","valve_detection":"ON","window_detection":"ON"}'
debug 2020-09-21 18:33:20: Received Zigbee message from '0x14b457fffe3ce248', type 'commandGetData', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[0,0,0,215],"type":"Buffer"},"dp":514,"fn":0,"status":0,"transid":108}' from endpoint 1 with groupID 0
info  2020-09-21 18:33:20: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe3ce248', payload '{"battery":97,"child_lock":"UNLOCKED","current_heating_setpoint":"21.5","linkquality":63,"local_temperature":"23.7","valve_detection":"ON","window_detection":"ON"}'

Koenkk, 2020-09-21:
Fixed.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

frlp542, 2020-09-21:
works !

Thanks

closed 2020-09-21 by frlp542