Koenkk/zigbee2mqtt #4420: Sonoff SNZB-04 Door/Window contact battery state not working

daha76, 2020-09-21:
## What happened
Sonoff SNZB-04 isn't providing battery state in percentage. It looks li it provide only information whether the battery is low. This is from a log file:
info  2020-09-21 10:14:21: MQTT publish: topic 'zigbee2mqtt/Sonoff W1', payload '{"battery_low":false,"contact":false,"linkquality":21,"tamper":false}'

info  2020-09-21 10:14:21: MQTT publish: topic 'homeassistant/binary_sensor/0x00124b001f90552e/contact/config', payload '{"availability_topic":"zigbee2mqtt/bridge/state","device":{"identifiers":["zigbee2mqtt_0x00124b001f90552e"],"manufacturer":"SONOFF","model":"Contact sensor (SNZB-04)","name":"Sonoff W1","sw_version":"Zigbee2MQTT 1.14.4"},"device_class":"door","json_attributes_topic":"zigbee2mqtt/Sonoff W1","name":"Sonoff W1 contact","payload_off":true,"payload_on":false,"state_topic":"zigbee2mqtt/Sonoff W1","unique_id":"0x00124b001f90552e_contact_zigbee2mqtt","value_template":"{{ value_json.contact }}"}'

info  2020-09-21 10:14:21: MQTT publish: topic 'homeassistant/sensor/0x00124b001f90552e/battery/config', payload '{"availability_topic":"zigbee2mqtt/bridge/state","device":{"identifiers":["zigbee2mqtt_0x00124b001f90552e"],"manufacturer":"SONOFF","model":"Contact sensor (SNZB-04)","name":"Sonoff W1","sw_version":"Zigbee2MQTT 1.14.4"},"device_class":"battery","json_attributes_topic":"zigbee2mqtt/Sonoff W1","name":"Sonoff W1 battery","state_topic":"zigbee2mqtt/Sonoff W1","unique_id":"0x00124b001f90552e_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'

info  2020-09-21 10:14:21: MQTT publish: topic 'homeassistant/sensor/0x00124b001f90552e/linkquality/config', payload '{"availability_topic":"zigbee2mqtt/bridge/state","device":{"identifiers":["zigbee2mqtt_0x00124b001f90552e"],"manufacturer":"SONOFF","model":"Contact sensor (SNZB-04)","name":"Sonoff W1","sw_version":"Zigbee2MQTT 1.14.4"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/Sonoff W1","name":"Sonoff W1 linkquality","state_topic":"zigbee2mqtt/Sonoff W1","unique_id":"0x00124b001f90552e_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'

## What did you expect to happen
I expect to have battery state.

## How to reproduce it (minimal and precise)
Add Sonoff SNZB-04 to the network and check battery state.

## Debug info
Zigbee2MQTT version: 1.14.4
Adapter hardware: CC2530
Adapter firmware version: zStack12 - 20190608

sjorge, 2020-09-21:
How long have they been in the network? All four of mine report battery, 3 still at 100 and one at 60% now.

They only update once a day I think.

daha76, 2020-09-21:
Actually few hours, I made several restarts of HA and zigbee2mqtt and nothing helped so far. I keep it some days to see if the battery % state will pop up.

sjorge, 2020-09-21:
Yeah give it +- 24hours, restarting z2m or ha is not going to do much as the device will report the battery when it thinks it needs to.

daha76, 2020-09-23:
So I gave it 24+ hours but I still didn't get battery state, sensor itself works and I get open/close state reliably. Other Sonoff sensors have no problem to report battery state, like action button or temperature and humidity sensor.

sjorge, 2020-09-23:
I wonder if the configuration of the reporting failed due to a timeout or the device already sleeping
https://github.com/Koenkk/zigbee-herdsman-converters/blob/bcba5901f807ff5d06ea9b8fa7929a68d796d607/devices.js#L11564

You could try to configure it again using https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure,

Make sure the device is awake though (open/closing the contact while z2m does it's thing usually works)

daha76, 2020-09-25:
Reconfigure didn't helped. So I forced remove the sensor as remove didn't work for some reason, restart HA, respectively whole RPi, pair the sensor again and so far so good. Obviously something went wrong but I'm not sure what. Will see if it will work reliable.

daha76, 2020-09-28:
After several removals/pairings and restarts it started to work again as expected, hopefully it will be fine from now.

closed 2020-09-28 by daha76