Koenkk/zigbee2mqtt #4423: Same value for "illuminance" and "illuminance_lux": what

davekr8, 2020-09-21:
```
Sep 21 19:39:47 raspberrypi3 npm[28573]: Zigbee2MQTT:info  2020-09-21 19:39:47: MQTT publish: topic 'home/zigbee/pir/living', payload '{"battery":100,"illuminance":30,"illuminance_lux":30,"linkquality":75,"occupancy":false,"voltage":3015}'
Sep 21 19:39:51 raspberrypi3 npm[28573]: Zigbee2MQTT:info  2020-09-21 19:39:51: MQTT publish: topic 'home/zigbee/pir/hall', payload '{"battery":100,"illuminance":22,"illuminance_lux":22,"linkquality":70,"occupancy":false,"voltage":3025}'
Sep 21 19:40:21 raspberrypi3 npm[28573]: Zigbee2MQTT:info  2020-09-21 19:40:21: MQTT publish: topic 'home/zigbee/pir/sofa', payload '{"battery":100,"illuminance":1,"illuminance_lux":1,"linkquality":72,"occupancy":false,"voltage":3005}'
Sep 21 19:40:28 raspberrypi3 npm[28573]: Zigbee2MQTT:info  2020-09-21 19:40:28: MQTT publish: topic 'home/zigbee/pir/ben', payload '{"battery":91,"illuminance":0,"illuminance_lux":0,"linkquality":70,"occupancy":false,"voltage":2985}'
```

## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

Koenkk, 2020-09-21:
If this is for an Xiaomi Aqara occupancy sensor this is expected behaviour. 

Background: according to the Zigbee spec illuminance values are not directly reported in lux. After finding this out we've create a new property: `illuminance_lux` which is guaranteed to be the illuminance value in lux. However the Xiaomi already reports it's illuminance in lux, therefore `illuminance` and `illuminance_lux` have the same value for this device.

davekr8, 2020-09-21:
Thanks for the clarification.

closed 2020-09-21 by davekr8