Koenkk/zigbee2mqtt #4424: GLEDOPTO light RGBW ( GL-S-005Z ) - White color not working ( with fix proposal ) / Transition time issue

Jeff94, 2020-09-21:
Hi Everone,

First thx for your work ! ... A quick feedback with fix and a problem I'm trying to solve

I'm evaluating an GLEDOPTO light MR16 GU5.3 RGBW ( GL-S-005Z ) - Device supported - 2 issues ( first one fixed see below )

1. RGB control works fine ( but only 2 LED in use vs 10 ) ... impossible to use the 8 others LED ( Warm / Cold White )
Problem fixed by adding colortemp property in devices.js - zigbeeModel GL-S-005Z

Original > "extend: gledopto.light_onoff_brightness_colorxy," 
Replaced by > to "extend: gledopto.light_onoff_brightness_colortemp_colorxy,"

Without colortemp property the light is stuck into RGB mode only.

2. I have the feeling this light have a transition time built in ??? ( ~ 2s+ ). I have tried to play around all properties ( state, brightness, color, color_temp with transition to 0 ) ... still 2s+ transition all the time. 

It does seems GLEDOPTO do not provide / allow any firmware / OTA update solution ?

Any input welcome to solve this very strange "feature" incompatible with my needs.

Thank you,

Koenkk, 2020-09-21:
1. Added
2. Seems it always uses a transition indeed, unfortunately Gledopto does not provide OTA updates so not fixable.

Jeff94, 2020-09-21:
Noted thx

closed 2020-10-29 by github-actions[bot]