Koenkk/zigbee2mqtt #4426: Router _TYZB01_zsl6z0pw TS0003 with lqi zero in only one direction

McGiverGim, 2020-09-22:
## What happened
Looking at the map network, I see all my nodes are ok, but one relay, that is a router, identified as `_TYZB01_zsl6z0pw TS0003` (the official name is `Lonsonho QS-Zigbee-S04-2C-LN` https://www.zigbee2mqtt.io/devices/QS-Zigbee-S04-2C-LN.html) shows the lqi with target the coordinator as zero always.

This router pairs with other three routers (Xiaomi relays), some of them are next to the coordinator. With all of them the lqi is correct, minimum over 70 lqi, so it does not seem to have a weak signal. In the opposite direction it has a 73 for example.

The relay works ok, it turns on and off the light without problem.

The map:
[image]

```json
{"links":[{"depth":1,"linkquality":93,"lqi":93,"relationship":1,"routes":[],"source":{"ieeeAddr":"0x00158d0003f453e3","networkAddress":54002},"sourceIeeeAddr":"0x00158d0003f453e3","sourceNwkAddr":54002,"target":{"ieeeAddr":"0x00124b0018e86d21","networkAddress":0},"targetIeeeAddr":"0x00124b0018e86d21"},{"depth":1,"linkquality":55,"lqi":55,"relationship":1,"routes":[],"source":{"ieeeAddr":"0x00158d00044bda5b","networkAddress":35708},"sourceIeeeAddr":"0x00158d00044bda5b","sourceNwkAddr":35708,"target":{"ieeeAddr":"0x00124b0018e86d21","networkAddress":0},"targetIeeeAddr":"0x00124b0018e86d21"},{"depth":1,"linkquality":84,"lqi":84,"relationship":1,"routes":[],"source":{"ieeeAddr":"0x00158d0003e7cf82","networkAddress":29730},"sourceIeeeAddr":"0x00158d0003e7cf82","sourceNwkAddr":29730,"target":{"ieeeAddr":"0x00124b0018e86d21","networkAddress":0},"targetIeeeAddr":"0x00124b0018e86d21"},{"depth":1,"linkquality":58,"lqi":58,"relationship":1,"routes":[],"source":{"ieeeAddr":"0x00158d00042dd87c","networkAddress":24690},"sourceIeeeAddr":"0x00158d00042dd87c","sourceNwkAddr":24690,"target":{"ieeeAddr":"0x00124b0018e86d21","networkAddress":0},"targetIeeeAddr":"0x00124b0018e86d21"},{"depth":1,"linkquality":101,"lqi":101,"relationship":1,"routes":[],"source":{"ieeeAddr":"0x00158d00036b8ca4","networkAddress":10359},"sourceIeeeAddr":"0x00158d00036b8ca4","sourceNwkAddr":10359,"target":{"ieeeAddr":"0x00124b0018e86d21","networkAddress":0},"targetIeeeAddr":"0x00124b0018e86d21"},{"depth":1,"linkquality":46,"lqi":46,"relationship":1,"routes":[],"source":{"ieeeAddr":"0x00158d0003cc2d19","networkAddress":26099},"sourceIeeeAddr":"0x00158d0003cc2d19","sourceNwkAddr":26099,"target":{"ieeeAddr":"0x00124b0018e86d21","networkAddress":0},"targetIeeeAddr":"0x00124b0018e86d21"},{"depth":1,"linkquality":73,"lqi":73,"relationship":1,"routes":[],"source":{"ieeeAddr":"0x00158d00045e8f67","networkAddress":25234},"sourceIeeeAddr":"0x00158d00045e8f67","sourceNwkAddr":25234,"target":{"ieeeAddr":"0x00124b0018e86d21","networkAddress":0},"targetIeeeAddr":"0x00124b0018e86d21"},{"depth":1,"linkquality":43,"lqi":43,"relationship":1,"routes":[],"source":{"ieeeAddr":"0x00158d0004445813","networkAddress":39},"sourceIeeeAddr":"0x00158d0004445813","sourceNwkAddr":39,"target":{"ieeeAddr":"0x00124b0018e86d21","networkAddress":0},"targetIeeeAddr":"0x00124b0018e86d21"},{"depth":1,"linkquality":93,"lqi":93,"relationship":1,"routes":[],"source":{"ieeeAddr":"0x00158d00044adbcc","networkAddress":48768},"sourceIeeeAddr":"0x00158d00044adbcc","sourceNwkAddr":48768,"target":{"ieeeAddr":"0x00124b0018e86d21","networkAddress":0},"targetIeeeAddr":"0x00124b0018e86d21"},{"depth":255,"linkquality":0,"lqi":0,"relationship":3,"routes":[],"source":{"ieeeAddr":"0x680ae2fffe096ce3","networkAddress":39089},"sourceIeeeAddr":"0x680ae2fffe096ce3","sourceNwkAddr":39089,"target":{"ieeeAddr":"0x00124b0018e86d21","networkAddress":0},"targetIeeeAddr":"0x00124b0018e86d21"},{"depth":0,"linkquality":101,"lqi":101,"relationship":0,"routes":[],"source":{"ieeeAddr":"0x00124b0018e86d21","networkAddress":0},"sourceIeeeAddr":"0x00124b0018e86d21","sourceNwkAddr":0,"target":{"ieeeAddr":"0x00158d00044adbcc","networkAddress":48768},"targetIeeeAddr":"0x00158d00044adbcc"},{"depth":2,"linkquality":201,"lqi":201,"relationship":1,"routes":[],"source":{"ieeeAddr":"0x04cf8cdf3c78c831","networkAddress":39777},"sourceIeeeAddr":"0x04cf8cdf3c78c831","sourceNwkAddr":39777,"target":{"ieeeAddr":"0x00158d00044adbcc","networkAddress":48768},"targetIeeeAddr":"0x00158d00044adbcc"},{"depth":2,"linkquality":147,"lqi":147,"relationship":1,"routes":[],"source":{"ieeeAddr":"0x00158d0003f431e6","networkAddress":43747},"sourceIeeeAddr":"0x00158d0003f431e6","sourceNwkAddr":43747,"target":{"ieeeAddr":"0x00158d00044adbcc","networkAddress":48768},"targetIeeeAddr":"0x00158d00044adbcc"},{"depth":2,"linkquality":118,"lqi":118,"relationship":1,"routes":[],"source":{"ieeeAddr":"0x00158d0002c8d4ab","networkAddress":55279},"sourceIeeeAddr":"0x00158d0002c8d4ab","sourceNwkAddr":55279,"target":{"ieeeAddr":"0x00158d00044adbcc","networkAddress":48768},"targetIeeeAddr":"0x00158d00044adbcc"},{"depth":2,"linkquality":111,"lqi":111,"relationship":1,"routes":[],"source":{"ieeeAddr":"0x00158d000323146e","networkAddress":28726},"sourceIeeeAddr":"0x00158d000323146e","sourceNwkAddr":28726,"target":{"ieeeAddr":"0x00158d00044adbcc","networkAddress":48768},"targetIeeeAddr":"0x00158d00044adbcc"},{"depth":1,"linkquality":132,"lqi":132,"relationship":2,"routes":[],"source":{"ieeeAddr":"0x00158d00044bda5b","networkAddress":35708},"sourceIeeeAddr":"0x00158d00044bda5b","sourceNwkAddr":35708,"target":{"ieeeAddr":"0x00158d00044adbcc","networkAddress":48768},"targetIeeeAddr":"0x00158d00044adbcc"},{"depth":1,"linkquality":76,"lqi":76,"relationship":2,"routes":[],"source":{"ieeeAddr":"0x680ae2fffe096ce3","networkAddress":39089},"sourceIeeeAddr":"0x680ae2fffe096ce3","sourceNwkAddr":39089,"target":{"ieeeAddr":"0x00158d00044adbcc","networkAddress":48768},"targetIeeeAddr":"0x00158d00044adbcc"},{"depth":1,"linkquality":196,"lqi":196,"relationship":2,"routes":[],"source":{"ieeeAddr":"0x00158d0003f453e3","networkAddress":54002},"sourceIeeeAddr":"0x00158d0003f453e3","sourceNwkAddr":54002,"target":{"ieeeAddr":"0x00158d00044adbcc","networkAddress":48768},"targetIeeeAddr":"0x00158d00044adbcc"},{"depth":0,"linkquality":104,"lqi":104,"relationship":0,"routes":[],"source":{"ieeeAddr":"0x00124b0018e86d21","networkAddress":0},"sourceIeeeAddr":"0x00124b0018e86d21","sourceNwkAddr":0,"target":{"ieeeAddr":"0x00158d0003f453e3","networkAddress":54002},"targetIeeeAddr":"0x00158d0003f453e3"},{"depth":2,"linkquality":192,"lqi":192,"relationship":1,"routes":[],"source":{"ieeeAddr":"0x00158d0004147c40","networkAddress":30780},"sourceIeeeAddr":"0x00158d0004147c40","sourceNwkAddr":30780,"target":{"ieeeAddr":"0x00158d0003f453e3","networkAddress":54002},"targetIeeeAddr":"0x00158d0003f453e3"},{"depth":2,"linkquality":66,"lqi":66,"relationship":1,"routes":[],"source":{"ieeeAddr":"0x00158d00045ee734","networkAddress":388},"sourceIeeeAddr":"0x00158d00045ee734","sourceNwkAddr":388,"target":{"ieeeAddr":"0x00158d0003f453e3","networkAddress":54002},"targetIeeeAddr":"0x00158d0003f453e3"},{"depth":2,"linkquality":49,"lqi":49,"relationship":1,"routes":[],"source":{"ieeeAddr":"0x00158d000323328a","networkAddress":26665},"sourceIeeeAddr":"0x00158d000323328a","sourceNwkAddr":26665,"target":{"ieeeAddr":"0x00158d0003f453e3","networkAddress":54002},"targetIeeeAddr":"0x00158d0003f453e3"},{"depth":1,"linkquality":66,"lqi":66,"relationship":2,"routes":[],"source":{"ieeeAddr":"0x680ae2fffe096ce3","networkAddress":39089},"sourceIeeeAddr":"0x680ae2fffe096ce3","sourceNwkAddr":39089,"target":{"ieeeAddr":"0x00158d0003f453e3","networkAddress":54002},"targetIeeeAddr":"0x00158d0003f453e3"},{"depth":1,"linkquality":199,"lqi":199,"relationship":2,"routes":[],"source":{"ieeeAddr":"0x00158d00044adbcc","networkAddress":48768},"sourceIeeeAddr":"0x00158d00044adbcc","sourceNwkAddr":48768,"target":{"ieeeAddr":"0x00158d0003f453e3","networkAddress":54002},"targetIeeeAddr":"0x00158d0003f453e3"},{"depth":1,"linkquality":135,"lqi":135,"relationship":2,"routes":[],"source":{"ieeeAddr":"0x00158d00044bda5b","networkAddress":35708},"sourceIeeeAddr":"0x00158d00044bda5b","sourceNwkAddr":35708,"target":{"ieeeAddr":"0x00158d0003f453e3","networkAddress":54002},"targetIeeeAddr":"0x00158d0003f453e3"},{"depth":0,"linkquality":65,"lqi":65,"relationship":0,"routes":[],"source":{"ieeeAddr":"0x00124b0018e86d21","networkAddress":0},"sourceIeeeAddr":"0x00124b0018e86d21","sourceNwkAddr":0,"target":{"ieeeAddr":"0x00158d00044bda5b","networkAddress":35708},"targetIeeeAddr":"0x00158d00044bda5b"},{"depth":2,"linkquality":90,"lqi":90,"relationship":1,"routes":[],"source":{"ieeeAddr":"0x00158d000444aa44","networkAddress":35288},"sourceIeeeAddr":"0 [... cut ...]
I suppose all this lqi must be over zero to show that it has a correct signal. I don't know if it is a problem in the relay firmware or something strange in the coordinator software. As I say the relay works ok (as device, it has not any other device attached so it is not working as a router).

## How to reproduce it (minimal and precise)
I don't have other relay with the same model to test if it happens to the same device always.

## Debug info
Zigbee2MQTT version: 1.14.4.2 (but I have observed the issue in the past releases too)
Adapter hardware: CC2531
Adapter firmware version: zStack12.20190608

McGiverGim, 2020-10-05:
If it helps, using the new MAP feature of the new UI, the relay appears as "disconnected" in the default filter:

[image]

It connects if I mark the `IsASibling` with LQI 70:

[image]

Or `NoneOfTheAbove` options with LQI 0:

[image]

McGiverGim, 2020-10-05:
More differences... my other routers show this in the new bind tab:
[image]

But this router shows this:
[image]

I didn't do anything different in this case, maybe is normal, but I don't understand exactly what this means.

McGiverGim, 2020-10-06:
Thanks for your hints @Koenkk but it seems the thing has going worst...

I have removed the device and joined again several times. Each time it does something different:
- Sometimes it joins, but as unknown device
- Sometimes it start join but never ends
- Sometimes it starts join and ends, but it does not appear in the logs ??????

When I joined this the first time I didn't have this problems. I will keep trying. I have tried to restart all without luck.

> The bind part is normal, this is to receive state updates when the switch is toggled manually.

Strange. My other routers are Xiaomi with manually/physical switch too.

McGiverGim, 2020-10-06:
If it helps, here is the log when it joins the network but it appears as unknown device:
```
info  2020-10-06 09:23:40: Starting interview of '0x680ae2fffe096ce3'
info  2020-10-06 09:23:40: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x680ae2fffe096ce3","ieee_address":"0x680ae2fffe096ce3"},"type":"device_joined"}'
info  2020-10-06 09:23:40: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x680ae2fffe096ce3","ieee_address":"0x680ae2fffe096ce3","status":"started"},"type":"device_interview"}'
info  2020-10-06 09:23:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x680ae2fffe096ce3"},"type":"device_connected"}'
info  2020-10-06 09:23:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x680ae2fffe096ce3"},"type":"pairing"}'
info  2020-10-06 09:23:45: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x680ae2fffe096ce3","ieee_address":"0x680ae2fffe096ce3"},"type":"device_announce"}'
info  2020-10-06 09:23:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x680ae2fffe096ce3"},"type":"device_announced"}'
error 2020-10-06 09:24:58: Failed to interview '0x680ae2fffe096ce3', device has not successfully been paired
```
And the result:
[image]

This is the same device that the first time appeared as a Zemismart being a Lonsonho: https://github.com/Koenkk/zigbee2mqtt/issues/4363

You said that you fixed it, maybe the fix is giving now this problems?

McGiverGim, 2020-10-06:
After a lot of tries, moving the router, enabling debug logs, etc. now it has joined and it appears as correctly connected.

[image]

In the map it continues as zero. So the initial problem is not fixed. As I said it works, so I will ignore it.

Only as curiosity: it detects the power correctly (plug) but the Xiaomi LLKZMK11LM appear as unknown as you can see in the screen capture.

Koenkk, 2020-10-06:
Can you provide the `data/database.db` entry of that device?

McGiverGim, 2020-10-06:
I suppose you are talking about the `share/zigbee2mqtt/database.db` file (I'm using Home Assistant addon).

This is the entry "formatted":
```json
{
	"id": 20,
	"type": "Router",
	"ieeeAddr": "0x680ae2fffe096ce3",
	"nwkAddr": 50831,
	"manufId": 4098,
	"manufName": "_TYZB01_zsl6z0pw",
	"powerSource": "Mains (single phase)",
	"modelId": "TS0003",
	"epList": [
		1,
		2
	],
	"endpoints": {
		"1": {
			"profId": 260,
			"epId": 1,
			"devId": 256,
			"inClusterList": [
				0,
				4,
				5,
				6,
				6
			],
			"outClusterList": [
				25,
				10
			],
			"clusters": {
				"genBasic": {
					"attributes": {
						"modelId": "TS0003",
						"manufacturerName": "_TYZB01_zsl6z0pw",
						"powerSource": 1,
						"zclVersion": 3,
						"appVersion": 65,
						"stackVersion": 0,
						"hwVersion": 1,
						"dateCode": ""
					}
				},
				"genOnOff": {
					"attributes": {
						"onOff": 1,
						"offWaitTime": 0,
						"onTime": 0
					}
				}
			},
			"binds": [
				{
					"cluster": 6,
					"type": "endpoint",
					"deviceIeeeAddress": "0x00124b0018e86d21",
					"endpointID": 1
				}
			],
			"meta": {}
		},
		"2": {
			"profId": 260,
			"epId": 2,
			"devId": 256,
			"inClusterList": [
				4,
				5,
				6,
				6
			],
			"outClusterList": [],
			"clusters": {
				"genOnOff": {
					"attributes": {
						"onOff": 0,
						"offWaitTime": 0,
						"onTime": 0
					}
				}
			},
			"binds": [
				{
					"cluster": 6,
					"type": "endpoint",
					"deviceIeeeAddress": "0x00124b0018e86d21",
					"endpointID": 1
				}
			],
			"meta": {}
		}
	},
	"appVersion": 65,
	"stackVersion": 0,
	"hwVersion": 1,
	"dateCode": "",
	"zclVersion": 3,
	"interviewCompleted": true,
	"meta": {
		"configured": 1
	},
	"lastSeen": 1601989515099
}
```

Another curiosity, the `bind` has disappeared in this new "join", now is the same than the Xiaomi routers, empty...

Koenkk, 2020-10-06:
Sorry, I mean the `LLKZMK11LM` entry

McGiverGim, 2020-10-06:
I have several of them, I suppose any of them will work:
```json
{
	"id": 17,
	"type": "Router",
	"ieeeAddr": "0x00158d00044bda5b",
	"nwkAddr": 21484,
	"manufId": 4151,
	"manufName": "LUMI",
	"powerSource": "DC Source",
	"modelId": "lumi.relay.c2acn01",
	"epList": [
		1,
		2
	],
	"endpoints": {
		"1": {
			"profId": 260,
			"epId": 1,
			"devId": 257,
			"inClusterList": [
				0,
				3,
				4,
				5,
				1,
				2,
				10,
				6,
				16,
				2820,
				12
			],
			"outClusterList": [
				25,
				10
			],
			"clusters": {
				"genBasic": {
					"attributes": {
						"65281": {
							"3": 38,
							"5": 31,
							"7": "0x0000000000000000",
							"8": 4643,
							"9": 1796,
							"100": 0,
							"101": 0,
							"110": 0,
							"111": 0,
							"148": 7,
							"149": 0.10343749821186066,
							"150": 2227.466064453125,
							"151": 0.005009882152080536,
							"152": 0.08244037628173828,
							"155": 0
						},
						"manufacturerName": "LUMI",
						"powerSource": 4,
						"zclVersion": 0,
						"stackVersion": 2,
						"hwVersion": 18,
						"dateCode": "09-20-2018"
					}
				},
				"genAnalogInput": {
					"attributes": {
						"presentValue": 0.04606962203979492
					}
				},
				"genOnOff": {
					"attributes": {
						"61440": 117440644,
						"onOff": 0
					}
				}
			},
			"binds": [],
			"meta": {}
		},
		"2": {
			"profId": 260,
			"epId": 2,
			"devId": 257,
			"inClusterList": [
				6,
				16,
				4,
				5
			],
			"outClusterList": [],
			"clusters": {
				"genOnOff": {
					"attributes": {
						"61440": 117440643,
						"onOff": 0
					}
				}
			},
			"binds": [],
			"meta": {}
		}
	},
	"appVersion": 1,
	"stackVersion": 2,
	"hwVersion": 18,
	"dateCode": "09-20-2018",
	"zclVersion": 0,
	"interviewCompleted": true,
	"meta": {},
	"lastSeen": 1601993871959
}
```

Koenkk, 2020-10-06:
@nurikk can you add the "plug" icon also for `"powerSource": "DC Source",`?

McGiverGim, 2020-10-06:
I opened an issue some minutes ago too: https://github.com/nurikk/z2m-frontend/issues/32

closed 2020-11-13 by github-actions[bot]