Koenkk/zigbee2mqtt #4428: Docker container crash due to "fatal error"

PedroLamas, 2020-09-22:
## What happened

Zigbee2Mqtt docker container crashed and docker restarted the container, I got this from the logs:

```
#
# Fatal error in , line 0
# Check failed: args[1].IsName().
#
#
#
#FailureMessage Object: 0xffffc4820768
```

I'm sorry I don't have more information on what could have caused this issue but thought it best to report it any way.

## What did you expect to happen

Not to crash!

## How to reproduce it (minimal and precise)

Currently unknown...

## Debug info
Zigbee2MQTT version: 1.14.4
Adapter hardware: CC26X2R1
Adapter firmware version: 20200805

megavolts, 2020-09-25:
@PedroLamas do you have any advanced configuration in your z2m config? I found out that the docker keep crashing unless I remove any advanced configuration like channel change or availability_timeout, or even debug.

PedroLamas, 2020-09-25:
Quite a few, yes, but then again that was never a problem before... Here's my full configuration: https://github.com/PedroLamas/nanopineo2-docker-config/blob/master/zigbee2mqtt/data/configuration.yaml

This has been working fine for a looooong time, I just noticed this on the latest release - and I've not seen it since I opened this issue!

closed 2020-11-03 by github-actions[bot]

TheQue42, 2022-05-12:
I am now having exactly this, after updating the container with portainer, that is, the container has been running fine, I have not made (knowingly...) any config changes (adding devices though) for a while, and now, this happends.
Trying to figure out how to increase debuglevel...?

Koenkk, 2022-05-12:
@TheQue42 see https://www.zigbee2mqtt.io/guide/installation/02_docker.html#notes-for-raspberry-pi-users