Koenkk/zigbee2mqtt #4430: Zigbee2mqtt can't control devices

demiiiiian, 2020-09-22:
Ive been using a pi4 with a cc2531 stick and its been working fine for months now. 
A few days ago I started experiencing problems and updated the Mosquito broker, this seemed to have fixed the problem. However a few days later I started having problems with my IKEA Tradfri lamp. I tried pairing it again and as far as I could tell that did work. However I still can't control any of my Zigbee devices.

Right now im running ZIgbee2mqtt version 1.14.4.2

demiiiiian, 2020-09-22:
The log is filled with entries like this:

2020-09-22 18:09:38: Publish 'set' 'state' to '0x00158d00040e51ff' failed: 'Error: Command 0x00158d00040e51ff/2 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC transaction expired' (240))'

Jolly-Roger82, 2020-09-23:
I'm having the same issue. Ever since the last Home Assistant Supervisor update my Z2M logs are full of errors. Very much the same as the errors shown above.

Koenkk, 2020-09-23:
There are some stability issues for CC2531 users in 1.14.4. Downgrade or switching to the dev branch solves it: https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

demiiiiian, 2020-09-23:
Im very new to all of this… how can I downgrade if I don't want to switch to the development branch?

demiiiiian, 2020-09-24:
I've now decided to switch over to the dev branch. This did not resolve any of my problems however

Koenkk, 2020-09-25:
Is your CC2531 connected through a USB extension cable? (positioned 30cm+ from the pi 4)

demiiiiian, 2020-09-26:
No, its plugged in directly to the pi

Koenkk, 2020-09-26:
Try via a usb extension cable. The pi 4 generates a lot of interference!

closed 2020-11-04 by github-actions[bot]