Koenkk/zigbee2mqtt #4434: Strange behavior, end device no longer works...

NicolasBoulanger, 2020-09-22:
Zigbee2MQTT version: latest
Adapter hardware: zzh!
Adapter firmware version: CC26X2R1_20200805

hello 

i have an issue with a philips hue color bulb

Zigbee2MQTT:error 2020-09-22 20:21:11: Publish 'set' 'brightness' to 'salon_lampadaire' failed: 'Error: Command 0x00178801039c2fc7/11 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":10}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 2458 - 11 - 33 - 8 - 11 after 10000ms)'

i'm running zigbee2mqtt on debian 10 with docker :

root@zigbee2mqtt:/home/zigbee2mqtt# docker --version
Docker version 19.03.12, build 48a66213fe
root@zigbee2mqtt:/home/zigbee2mqtt#

other thing, sometimes, end devices like hue motion sensor or aqara motion sensor not working,  i have some strange behavior, and no log from docker logs :/ and i need to restart my container to make it works, and after some minutes device work again... 

Zigbee2MQTT:info  2020-09-22 20:38:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"capteur_hall"},"type":"device_announced"}'
Zigbee2MQTT:info  2020-09-22 20:38:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"capteur_hall"},"type":"device_announced"}'

i have almost 50 devices with 80% of router 

other people has same issue ? thanks

Koenkk, 2020-09-23:
Can you share your `configuration.yaml`?

NicolasBoulanger, 2020-09-23:
> Can you share your `configuration.yaml`?

yes sure 👍 http://pastebin.fr/66286

Dr4kn, 2020-09-23:
Yeah I have the same issue. I believe since I updated to 1.14.4. Running in docker on ubuntu. I only have 15 devices most of them tradfri. Those close to the device are as random unresponsive as the others.

Koenkk, 2020-09-25:
- Can you update to the latest dev branch and check if the issue is still present there? (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)
- Does it only happen with this bulb or also others?
- Could it be an interference/poor network range issue? (is this bulb located at a more remote location than others?)

NicolasBoulanger, 2020-09-25:
> * Can you update to the latest dev branch and check if the issue is still present there? (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)
>
>     * Does it only happen with this [...]

it only affect end device like hue remote switch or hue motion sensor
my network is not really poor i have a lot of router; the lowest link quality is 57 (only 1-2 devices)
https://imgur.com/dWGjKDS
i can try with dev branch sure

NicolasBoulanger, 2020-09-27:
I lost 2 hue remote switch this weekend with latest-dev docker image :/
no log, just hue remote blinking red when i trigger a button :(

closed 2020-11-05 by github-actions[bot]