Koenkk/zigbee2mqtt #4435: Power attribute not showing anymore in different devices

jd1900, 2020-09-22:
## What happened
After some upgrade (I don't know which) power in Watts for two different devices with different brands don't appear anymore.
Devices are:
- Legrand Power socket with power consumption monitoring (067775)
- Ubisys Power switch S1 (S1)

None of them report the W anymore.

The rest of the attributes in those devices and every other devices are working fine.

Has any of the recent versions made any change with regard to Watts or power consumption? 

## What did you expect to happen
Report the power in Watts.

## How to reproduce it (minimal and precise)
Try to see the power in W.

## Debug info
Zigbee2MQTT version: 1.14.4.2
Adapter hardware: CC26X2R1
Adapter firmware version: CC26X2R1_20191106.zip 20191106 Z-Stack_3.x.0 firmware.

Koenkk, 2020-09-25:
Try reconfiguring the device via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure

jd1900, 2020-09-25:
It did the trick. 

Thank you!

closed 2020-09-25 by jd1900