Koenkk/zigbee2mqtt #4436: Sinope TH1124ZB Can't set temperature

MRobi1, 2020-09-22:
## What happened 
Trying to set temperature of multiple TH1124ZB thermostats. Temperature not set

## What did you expect to happen
Using home assistant, trying to set temperature of TH1124ZB and getting errors in logs. Happens on all 15 thermostats.

## How to reproduce it (minimal and precise)
Set temperature of device through home assistant

## Debug info
Zigbee2MQTT version: 1.14.4.2
Adapter hardware: Conbee II
Adapter firmware version:

Zigbee2MQTT:info  2020-09-22 19:02:19: MQTT publish: topic 'zigbee2mqtt/thermostat_living_room_hallway', payload '{"brightness":0,"linkquality":139,"local_temperature":19.5,"occupied_heating_setpoint":16,"operation":"idle","pi_heating_demand":0,"state":"ON","system_mode":"heat"}'
Zigbee2MQTT:error 2020-09-22 19:02:19: Publish 'set' 'occupied_heating_setpoint' to 'thermostat_living_room_hallway' failed: 'Error: Write 0x500b914000011021/1 hvacThermostat({"occupiedHeatingSetpoint":1900}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Status 'undefined')'

Koenkk, 2020-09-25:
Can you check if this has already been fixed in the edge addon?

If not, provide the herdsman debug logging so @chrishae can further investigate this.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

MRobi1, 2020-09-25:
Just moved to the edge addon. Will do some testing when the AC kicks out and let you know.

MRobi1, 2020-09-26:
Same error on edge
`Zigbee2MQTT:error 2020-09-26 08:45:23: Publish 'set' 'occupied_heating_setpoint' to 'thermostat_downstairs_bathroom' failed: 'Error: Write 0x500b914000011052/1 hvacThermostat({"occupiedHeatingSetpoint":1500}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Status 'undefined')'
Zigbee2MQTT:error 2020-09-26 08:46:15: Publish 'set' 'occupied_heating_setpoint' to 'thermostat_theatre' failed: 'TypeError: Write 0x500b914000017931/1 hvacThermostat({"occupiedHeatingSetpoint":1500}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Cannot read property 'frame' of null)'`

I've also noticed calling the "turn_on" climate service does not turn the thermostats back on. I also tried HA's set_hvac_mode to heat thinking that may turn them back on but it doesn't. turn_off works as expected and sets them to off.

MRobi1, 2020-09-26:
Debug log
`2020-09-26T11:53:47.548Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-09-26T11:53:56.545Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2020-09-26T11:53:56.553Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2020-09-26T11:53:56.571Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 76
2020-09-26T11:53:56.574Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 76 srcAddr: 0xbd11 destAddr: 0x0 profile id: 0x104 cluster id: 0x402
2020-09-26T11:53:56.575Z zigbee-herdsman:deconz:frameParser response payload: 24,254,10,0,0,41,86,7
2020-09-26T11:53:56.579Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":254,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":41,"attrData":1878}]},"address":48401,"endpoint":1,"linkquality":52,"groupID":null}'
2020-09-26T11:53:56.581Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-09-26T11:53:57.516Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 77
2020-09-26T11:53:57.521Z zigbee-herdsman:deconz:frameParser device state: 10100010
2020-09-26T11:53:57.521Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-09-26T11:54:03.762Z zigbee-herdsman:deconz:adapter no response received
2020-09-26T11:54:03.763Z zigbee-herdsman:deconz:adapter no response received
2020-09-26T11:54:03.763Z zigbee-herdsman:deconz:adapter no response received
2020-09-26T11:54:03.764Z zigbee-herdsman:deconz:adapter no response received
2020-09-26T11:54:03.764Z zigbee-herdsman:deconz:adapter no response received
2020-09-26T11:54:03.764Z zigbee-herdsman:deconz:adapter no response received
2020-09-26T11:54:03.765Z zigbee-herdsman:deconz:adapter no response received
2020-09-26T11:54:03.765Z zigbee-herdsman:deconz:adapter no response received
2020-09-26T11:54:03.767Z zigbee-herdsman:controller:endpoint Write 0x500b914000011498/1 hvacThermostat({"occupiedHeatingSetpoint":2000}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Cannot read property 'frame' of null)
2020-09-26T11:54:03.769Z zigbee-herdsman:controller:endpoint Write 0x500b914000010f63/1 hvacThermostat({"occupiedHeatingSetpoint":1800}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Cannot read property 'frame' of null)
2020-09-26T11:54:03.769Z zigbee-herdsman:controller:endpoint Write 0x500b914000017647/1 hvacThermostat({"occupiedHeatingSetpoint":1500}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Cannot read property 'frame' of null)
2020-09-26T11:54:03.770Z zigbee-herdsman:controller:endpoint Write 0x500b914000017931/1 hvacThermostat({"occupiedHeatingSetpoint":1500}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Cannot read property 'frame' of null)
2020-09-26T11:54:03.770Z zigbee-herdsman:controller:endpoint Write 0x500b914000011052/1 hvacThermostat({"occupiedHeatingSetpoint":1500}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Cannot read property 'frame' of null)
2020-09-26T11:54:03.770Z zigbee-herdsman:controller:endpoint Write 0x500b914000017b69/1 hvacThermostat({"occupiedHeatingSetpoint":1500}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Cannot read property 'frame' of null)
2020-09-26T11:54:03.771Z zigbee-herdsman:controller:endpoint Write 0x500b91400001152f/1 hvacThermostat({"occupiedHeatingSetpoint":1800}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Cannot read property 'frame' of null)
2020-09-26T11:54:03.771Z zigbee-herdsman:controller:endpoint Write 0x500b914000011828/1 hvacThermostat({"occupiedHeatingSetpoint":1800}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Cannot read property 'frame' of null)
Zigbee2MQTT:error 2020-09-26 08:54:03: Publish 'set' 'occupied_heating_setpoint' to 'thermostat_play_room' failed: 'TypeError: Write 0x500b914000011498/1 hvacThermostat({"occupiedHeatingSetpoint":2000}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":nu [... cut ...]
2020-09-26T11:54:06.318Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 79 srcAddr: 0x89d7 destAddr: 0x0 profile id: 0x104 cluster id: 0x201
2020-09-26T11:54:06.318Z zigbee-herdsman:deconz:frameParser response payload: 24,63,10,8,0,32,0
2020-09-26T11:54:06.322Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":63,"manufacturerCode":null,"commandIdentifier":10},"Payload":`

ChrisHae, 2020-09-29:
I would say this does not look like the latest code. @Koenkk are you sure the hassio Addon has the latest commits of zigbee-herdsman?
Besides @MRobi1 Do the ohter devices in your network work normal? Was the interview of the thermostats successfull?

@Koenkk what would I have to implement to forward deconz adapter debug log to the hassio addon? I think just add -e DEBUG=zigbee-herdsman:* is not possible there?

Koenkk, 2020-09-29:
@MRobi1 can you uninstall and install to make sure you are on the latest version? To enable the herdsman debug logging for the hassio addon set the following in the config via the web interface: `"zigbee_herdsman_debug": true` (only works on edge!)

MRobi1, 2020-09-29:
My edge version was indeed outdated. uninstalled/re-installed.
I have zigbee_herdsman_debug: true in my config but the log still looks the same to me.

For the climate turn_on service, I've noticed if I send the turn_on to a few thermostats it seems to work. If I send it to all 15, it's as if the command goes nowhere and none turn on. turn_off works when sending to all 15.

`2020-09-29T20:15:52.949Z zigbee-herdsman:controller:endpoint DefaultResponse 0x286d97000110db9f/1 1026(10, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
2020-09-29T20:15:52.951Z zigbee-herdsman:deconz:adapter zclFrame.payload:
2020-09-29T20:15:52.952Z zigbee-herdsman:deconz:adapter { cmdId: 10, statusCode: 0 }
2020-09-29T20:15:52.952Z zigbee-herdsman:deconz:adapter no response expected
2020-09-29T20:15:52.952Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-09-29T20:15:52.976Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x794f EP:1 SeqNr. 190 request id: 31
2020-09-29T20:15:52.980Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 31 status: 0
2020-09-29T20:15:52.983Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-09-29T20:15:52.991Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2020-09-29T20:15:52.991Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-09-29T20:15:53.026Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 191
2020-09-29T20:15:53.030Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x794f request id: 31 confirm status: 0
2020-09-29T20:15:53.030Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-09-29T20:15:53.030Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 108
2020-09-29T20:15:53.859Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2020-09-29T20:15:53.860Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2020-09-29T20:15:53.880Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 192
2020-09-29T20:15:53.885Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 192 srcAddr: 0xdab3 destAddr: 0x0 profile id: 0x104 cluster id: 0x402
2020-09-29T20:15:53.885Z zigbee-herdsman:deconz:frameParser response payload: 24,5,10,0,0,41,76,8
2020-09-29T20:15:53.889Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":5,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":41,"attrData":2124}]},"address":55987,"endpoint":1,"linkquality":89,"groupID":null}'
2020-09-29T20:15:53.894Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-09-29T20:15:58.174Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 193
2020-09-29T20:15:58.182Z zigbee-herdsman:deconz:frameParser device state: 10100010
2020-09-29T20:15:58.183Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-09-29T20:15:59.794Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2020-09-29T20:15:59.795Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2020-09-29T20:15:59.796Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 194
2020-09-29T20:15:59.804Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 194 srcAddr: 0x8278 destAddr: 0x0 profile id: 0x104 cluster id: 0x201
2020-09-29T20:15:59.805Z zigbee-herdsman:deconz:frameParser response payload: 24,96,10,0,0,41,4,8
2020-09-29T20:15:59.808Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":96,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":41,"attrData":2052}]},"address":33400,"endpoint":1,"linkquality":89,"groupID":null}'
2020-09-29T20:15:59.808Z zigbee-herdsman:controller:log 'zcl' data is from unknown device with address '33400', skipping...
2020-09-29T20:15:59.809Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-09-29T20:16:06.020Z zigbee-herdsman:controller:endpoint Write 0x500b914000017b69/1 hvacThermostat({"occupiedHeatingSetpoint":1200}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (no response received)
[... cut ...]
Zigbee2MQTT:info  2020-09-29 17:16:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'occupied_heating_setpoint' to 'thermostat_spare_bedroom' failed: 'Error: Write 0x500b914000017647/1 hvacThermostat({\"occupiedHeatingSetpoint\":1200}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (no response `

ChrisHae, 2020-10-01:
@MRobi1 
Could you deliver additional debug log? I can see where it fails but not where it sends the request. A log where you send the request until it fails and the case when it works would be fine. You could also copy the log to pastebin or something else if it doesn't fit here.

ChrisHae, 2020-10-02:
I think it could be because fast sending multiple requests is not handled very well by the deCONZ adapter. I will improve that in the next commits.

MRobi1, 2020-10-10:
@ChrisHae sorry, I've been out on the road for work this week. 
I just updated again to the latest edge version and I see big improvements with this device.

First, I'm now getting local temp and energy/power. Awesome!

Second, the "turn_on" command in home assistant to my thermostat group turned them all on the first time. This is great! However, the "state" of the thermostats still say "off". They're currently on and heating but the state didn't update. 

Third, I'm still getting errors in the logs
ERROR: Publish 'set' 'occupied_heating_setpoint' to 'thermostat_play_room' failed: 'Error: Write 0x500b914000011498/1 hvacThermostat({"occupiedHeatingSetpoint":2000}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (no response received)'
ERROR: Publish 'set' 'thermostat_outdoor_temperature' to 'all_thermostats_group' failed: 'TypeError: Cannot create property 'message' on string 'APS TIMEOUT''

Even with the errors, the outdoor temp is still being updated and the setpoint is being changed.

Here's a pastbin from the log file
https://pastebin.com/2CeH6XFZ

The log file appears to be the standard info log so here's a paste from the log inside supervisor which seems to have more details
`2020-10-10T14:26:32.800Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 199 srcAddr: 0xb5df destAddr: 0x0 profile id: 0x104 cluster id: 0x201
2020-10-10T14:26:32.801Z zigbee-herdsman:deconz:frameParser response payload: 24,189,10,18,0,41,208,7
2020-10-10T14:26:32.808Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":189,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":18,"dataType":41,"attrData":2000}]},"address":46559,"endpoint":1,"linkquality":202,"groupID":null}'
Zigbee2MQTT:info  2020-10-10 11:26:32: MQTT publish: topic 'zigbee2mqtt/thermostat_play_room', payload '{"brightness":0,"energy":24.14,"linkquality":202,"local_temperature":20.4,"occupied_heating_setpoint":20,"operation":"heating","pi_heating_demand":16,"state":"ON","system_mode":"off"}'
2020-10-10T14:26:32.813Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-10-10T14:26:39.621Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 200
2020-10-10T14:26:39.630Z zigbee-herdsman:deconz:frameParser device state: 10100010
2020-10-10T14:26:39.630Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-10-10T14:26:41.703Z zigbee-herdsman:deconz:driver Timeout for aps request CMD: 0x17 seq: 149
2020-10-10T14:26:41.826Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2020-10-10T14:26:41.826Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
query aps data indication
2020-10-10T14:26:42.830Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 201
2020-10-10T14:26:42.835Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 201 srcAddr: 0xb5df destAddr: 0x0 profile id: 0x104 cluster id: 0x702
2020-10-10T14:26:42.835Z zigbee-herdsman:deconz:frameParser response payload: 24,190,10,0,0,37,120,94,0,0,0,0
2020-10-10T14:26:42.845Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":190,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":37,"attrData":[0,24184]}]},"address":46559,"endpoint":1,"linkquality":202,"groupID":null}'
Zigbee2MQTT:info  2020-10-10 11:26:42: MQTT publish: topic 'zigbee2mqtt/thermostat_play_room', payload '{"brightness":0,"energy":24.18,"linkquality":202,"local_temperature":20.4,"occupied_heating_setpoint":20,"operation":"heating","pi_heating_demand":16,"state":"ON","system_mode":"off"}'
2020-10-10T14:26:42.851Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-10-10T14:26:42.851Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 239
2020-10-10T14:26:44.705Z zigbee-herdsman:deconz:driver Timeout for aps request CMD: 0x17 seq: 151
2020-10-10T14:26:49.650Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 202
2020-10-10T14:26:49.655Z zigbee-herdsman:deconz:frameParser device state: 10100010
2020-10-10T14:26:49.656Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-10-10T14:26:50.130Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2020-10-10T14:26:50.134Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
query aps data indication
2020-10-10T14:26:50.855Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 203
[... cut ...] ress":35287,"endpoint":1,"linkquality":202,"groupID":null}'
Zigbee2MQTT:info  2020-10-10 11:27:08: MQTT publish: topic 'zigbee2mqtt/thermostat_haileys_room', payload '{"brightness":0,"energy":523.79,"linkquality":202,"local_temperature":18.76,"occupied_heating_setpoint":18,"operation":"idle","pi_heating_demand":0,"state":"ON","system_mode":"off"}'
2020-10-10T14:27:08.771Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1`

closed 2020-11-18 by github-actions[bot]

MRobi1, 2021-01-13:
Just an update, still getting flooded with errors in the logs. For the most part the thermostat changes set points when it receives the command, but I'd estimate they don't respond about 5% of the time. Which is a lot when you've got 18 of these now in the house...

`ERROR: Publish 'set' 'system_mode' to 'thermostat_theatre' failed: 'Error: Write 0x500b914000017931/1 hvacThermostat({"systemMode":4}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (no response received)'`

ChrisHae, 2021-01-14:
Do you use the latest zigbee-mqtt version? There is a new parameter that you can add in the configuration.yaml
in the advanced section add:
adapter_delay: 400

Please try this and check if this helps.

MRobi1, 2021-01-14:
@ChrisHae Yes I'm on the edge version updated a few days ago. Been playing with adapter_delay trying to get some sort of stability. Mainly with my row of Fyrtur blinds in the living room that don't respond almost 50% of the time. Currently set at 1100. At least with these thermostats and the way I have my automation set, it re-evaluates every few minutes and if any are not set where they should it sends another set temp command. So they all eventually get to where they need to be.