Koenkk/zigbee2mqtt #4438: can't start zigbee

cchalong, 2020-09-23:
## What happened
after changing my router, I can't start zigbee-herdsman...

## What did you expect to happen
I unplug and replug my coordinator(CC1352p-2) but I failed to fix it.

my zigbee2mqtt log is as follows.

Zigbee2MQTT:info 2020-09-23 13:06:10: Logging to console and directory: '/share/zigbee2mqtt/log/2020-09-23.13-06-10' filename: log.txt 
Zigbee2MQTT:info 2020-09-23 13:06:11: Starting Zigbee2MQTT version 1.14.4 (commit #1.14.4) 
Zigbee2MQTT:info 2020-09-23 13:06:11: Starting zigbee-herdsman... 
Zigbee2MQTT:error 2020-09-23 13:06:13: Error while starting zigbee-herdsman 
Zigbee2MQTT:error 2020-09-23 13:06:13: Failed to start zigbee 
Zigbee2MQTT:error 2020-09-23 13:06:13: Exiting... 
Zigbee2MQTT:error 2020-09-23 13:06:13: Error: SREQ '--> SYS - osalNvWrite - {"id":1,"offset":0,"osal":true,"product":-1,"value":[10,178,161,28,0,75,18,0],"len":8}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)')
   at Znp.<anonymous> (/zigbee2mqtt-1.14.4/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:291:27)
   at Generator.next (<anonymous>) 
   at fulfilled (/zigbee2mqtt-1.14.4/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58) 
npm 
 ERR! code ELIFECYCLE 
npm ERR! errno 1 
npm 
 ERR! zigbee2mqtt@1.14.4 start: `node index.js` 
npm ERR! Exit status 1 
npm ERR! npm ERR! Failed at the zigbee2mqtt@1.14.4 start script. 
npm ERR! This is probably not a problem with npm. There is likely additional logging output above. 
npm ERR! A complete log of this run can be found in: 
npm ERR! /root/.npm/_logs/2020-09-23T04_06_13_367Z-debug.log 
2020-09-23T13:06:14: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT] 
2020-09-23T13:06:14: PM2 log: App [npm:0] starting in -fork mode- 
2020-09-23T13:06:14: PM2 log: App [npm:0] online 
> zigbee2mqtt@1.14.4 start /zigbee2mqtt-1.14.4 > node index.js

## How to reproduce it (minimal and precise)
is there anything to do after changing the router in order to re-connect z2m?

## Debug info
Zigbee2MQTT version: latest
Adapter hardware: CC1352P-2, 
Adapter firmware version:

Jolly-Roger82, 2020-09-23:
I'm having the same issue. To restart my Z2M I have to unplug my CC2531 sniffer, re-plug it, and then it will start. 
The host logs show me an error which relates to serial adapters not working correctly.

But after a while, Z2M shows other errors, not able to set LQI for my (router) lightbulbs. 

All occurred after the home assistant Supervisor update, since then i've not been able to get my otherwise stable and good working Z2M mesh up and running again.

Koenkk, 2020-09-23:
@cchalong can you try reflashing your CC1352?

Jolly-Roger82, 2020-09-23:
am in the process of re-flashing, any suggestion on what firmware to use?

JHurk, 2020-09-23:
I do not want to cause any confusion but I had the same issue (and multiple issues that are described in other recent issue threads). I flashed the CC2531 with default firmware and with the sourcing firmware. Nothing helped.

I downgraded to 1.14.3 (Docker Synology NAS) and instantly everything worked again.
So it seems to me that it is a bug in 1.14.4 which could be solved with reflashing the stick firmware, but in my perspective should be fixed in Zigbee2MQTT.

Maybe I am wrong on this, not to cause any confusion, but if you have problems and are able to downgrade to 1.14.3 (or other previous version which worked for you); just do it.

cima37, 2020-09-23:
Hallo! I have the same problem with the cc2652 usb stick!

cchalong, 2020-09-24:
> @cchalong can you try reflashing your CC1352?

I'm afraid that all connection to my zigbee devices is gone and I should do paring again for all devices. 
if there is no ways except for reflashing, I have no choice.

Koenkk, 2020-09-24:
@cchalong reflashing does not require repairing of all devices. On first star the `coordinator_backup.json` will be restored.

closed 2020-11-02 by github-actions[bot]