Koenkk/zigbee2mqtt #4439: Ikea tradfri Outlets falling off the network

tabnul, 2020-09-23:
## What happened
I have a zigbee network with 10 ikea tradfri outlets, 2 osram lights and 3 ikea lights and 20-ish xiaomi sensors.
Some of the Ikea Tradfri Outlets falling off the network every now and then, just a single device every time. The rest keeps working.
All devices are updated to the latest firmware.
Dont have logging currently (just put it on debug), but the devices just become unavailable (not responding to availability pings) and stop reporting by themself.

> power cycling the device fixes the issue.

I am not sure whether this is related to zigbee2mqtt or to the actual hardware. it happens to both old outlets (2018/2019) and brand new ones. I cannot find the issue online for people that use the tradfri gateway, so maybe a combination issue?
Then again i would expect this to also happen to the lightbulbs of ikea (assuming same chips/firmware)

## What did you expect to happen
Devices to stay connected.

## How to reproduce it (minimal and precise)
Not able to reproduce. Just have to wait a few days/weeks for one to fall off the network.

## Debug info
Zigbee2MQTT version: 1.14.4
Adapter hardware: cc1352p-2
Adapter firmware version: 20200805

Below my configuration.

`
{
  "external_converters": [],
  "devices": "devices.yaml",
  "groups": "groups.yaml",
  "homeassistant": true,
  "permit_join": false,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "mqtt://core-mosquitto",
    "user": "ZZZZZZZZZZ",
    "password": "ZZZZZZZZZZZ"
  },
  "serial": {
    "port": "/dev/ttyACM0"
  },
  "advanced": {
    "availability_timeout": 900,
    "report": true,
    "log_level": "debug",
    "pan_id": 6759,
    "channel": 21,
    "last_seen": "ISO_8601",
    "network_key": [
ZZZZZZZZZZZZZ
    ],
    "availability_blocklist": [],
    "availability_passlist": []
  },
  "device_options": {},
  "blocklist": [],
  "passlist": [],
  "queue": {},
  "experimental": {}
}
`

tabnul, 2020-09-23:
i want to add the remark that i did not have this issue a year ago when i was using less devices with cc2531 and obviously older zigbee2mqtt version. 
Maybe it is related to amount of devices, amount of traffic, routing configuration, or something else?

tabnul, 2020-09-23:
After reading the linked post above i started looking at my link quality and it looks like that some routers have very poor range **directly** to the coordinator. Especially some of the routers that i saw dropping off the network.

However those routers are in the same room as other routers. They have 200+ link quality with those neighbour routers.
Those neighbour routers on their turn have respectable link quality with the coordinator.

So basically, the mesh itself is healthy, but router/outlet > coordinator is not always healthy.
Could this cause them to drop off sometimes? Doesnt make sense to me..

Koenkk, 2020-09-25:
To debug this I would recommend sniffing the network and identify the timeframe they drop off as narrow as possible. (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html).

tabnul, 2020-09-25:
Thanks Koen, i will wait for it to happen again.
I relocated the coordinator slightly and changed the channel to 25 and this at least increases link quality. Lets see if this fixes the issue. Wouldnt make sense for me but lets see.

tabnul, 2020-10-04:
Ok, it has been a day or 10 and i havent seen the issue come back, whereas it would occur at least after a few days before.
So changing to channel 25 and changing the location of the adapter.

Still i dont understand why only some individual devices after a while drop off the network, while the mesh itself should keep them available.
Is this something in Z2M or in the ikea devices?

tabnul, 2020-10-16:
Ok, it seems to be an issue with the outlet placement in my case.
The outlets that gave issues were sitting next to eachother in the power strip, without any room in between.

Since i replaced them to have a few centimeter space in between i havent seen them falling off the network. Also no LQI failed messages anymore in the logs! Also , my xiaomi sensors seem to be more stable, probably because their routers are having less distortion.

Could it be that placing routers near each other causes signal interference?

Anyway doesnt look like to be a zigbee2mqtt issue ;)

closed 2020-10-16 by tabnul