Koenkk/zigbee2mqtt #4440: Failed to connect to zig-a-zig-ah!

tpaulus, 2020-09-23:
## What happened
I cannot get Z2M to come up, I have tried flashing the firmware multiple times, and running Z2M both via Docker on an Intel NUC running Ubuntu and Bare Metal on a RasPi, with no success. I tried switching to the Dev Branch as well, since I saw some issues indicating a potential fix for a similar issue there. I have never had this adapter come up, but I want to make sure it isn't me.

herdsman debug logs: https://pastebin.com/mSvwtffk
debug log: https://pastebin.com/yfR7LUe2

## What did you expect to happen
Z2M to come up.

## How to reproduce it (minimal and precise)
¯\\_(ツ)_/¯

## Debug info
Zigbee2MQTT version:
Adapter hardware:  CC26X2R1 (zig-a-zig-ah!)
Adapter firmware version: CC26X2R1_20200805.zip

PedroLamas, 2020-09-23:
Were you using a different adapter before? If so, which one?

tpaulus, 2020-09-23:
I was not. This is my first time using Z2M.

Sandriekus, 2020-09-23:
i think your zig-a-zig-ah! stick is not on /dev/ttyUSB0

what do you see on:
`ls -l /dev/serial/by-id`

tpaulus, 2020-09-23:
```
pi@raspberrypi:~ $ ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 Sep 23 17:47 usb-1a86_USB_Serial-if00-port0 -> ../../ttyUSB0
```

Sandriekus, 2020-09-23:
oke, strange..
and if you use the following in zigbee2mqtt config:

serial:
  port: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0

tpaulus, 2020-09-23:
No changes, Z2M still does not come up.

syphernl, 2020-09-24:
This particular problem is covered in the [Zigbee2MQTT FAQ](https://www.zigbee2mqtt.io/information/FAQ.html#cc1352p-2cc26x2r1-coordinators-only-press-the-reset-button-on-the-device):

> If Zigbee2MQTT fails to start with a CC1352P-2 with Error: SRSP - SYS - version after 6000ms, you most probably have connected your device to a system that requires pressing the reset button (the one next to the USB connector) momentarily/shortly after connecting the USB cable. This issue has primarily been observed on x86 architectures only (e.g., Intel NUC, HPE Microserver, i7 laptop), see also #2162. The procedure has to be repeated every time the CC1352P-2 is re-connected and it’s not clear yet, whether this can be fixed at all. It does not seem to occur on ARM based boards (Raspberry Pi, ODROID XU4).
> Something that can also solve the issue is to replug the USB cable.

The [official documentation](https://electrolama.com/projects/zig-a-zig-ah/#troubleshooting) also has something that covers this particular error, although it also mainly "is your configuration right" (which looks fine to me, port + rtscts off) and replugging.

Hope this helps!

Koenkk, 2020-09-24:
Can you try reflashing the firmware? If it even doesn't work on a bare metal system (multiple tested), and you are sure you are using the correct port the only thing I can imagine is that the stick itself is broken.

krencl, 2020-09-27:
Hi, I have same problem. I tried everything - flashing in Ubuntu (using python script), flashing in Windows (using TI Flash Programmer 2), different versions of firmware.
But I z2m is unable connect to zig-a-zig-ah! Again in Ubuntu (I am using /dev/ttyUSB0, because nothing like /dev/ttyAMC0 is not showing up) and in Windows tried connect to \.\\COM3.

So it is really probably broken hardware?

omerk, 2020-09-27:
We are dealing with two cases here.

In the case of @tpaulus, flashing the stick works but the MCU does not run any programs (although the factory logs show that the unit has passed the "blink test" which is burning blink.bin and observing the LED). I've already sent him a replacement stick to try and see if this is indeed a hardware issue and his stick is on his way back to me for diagnostics.

What @krencl is reporting is different, from what I can gather, in that he can flash blink.bin and it runs fine but the coordinator firmware is not working?

@krencl Please download and run this Python script and post the output here: https://gist.github.com/omerk/0ee0e447a9e36786b4ff71d8f8126a23 Please also let us know the filename of the coordinator firmware you are using.

krencl, 2020-09-28:
@omerk I tried flash it now again, and now it can't connect to serial:
`ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'`

Tried it multiple times, same result. So I can't flash it now.

Before I used this FW: [CC26X2R1_20200805.hex](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/CC26X2R1_20200805.zip) and older version CC26X2R1_20200229.hex

omerk, 2020-09-28:
@krencl Please provide more information so we can help you:

1) What do you see when you plug your zzh in and issue `dmesg`?

2) What command line are you using to flash?

3) Are you sure you are pushing the BSL button in as you're plugging the stick in? The process is, with the stick unplugged to start with: Press firmly and hold the BSL button, plug the stick in, wait a few seconds and then release the BSL button. At this point, you should be able to run cc2538-bsl.py.

krencl, 2020-09-28:
1. dmesg:
```
[8666695.014091] usb 3-1: new full-speed USB device number 36 using ohci-pci
[8666695.207323] usb 3-1: New USB device found, idVendor=1a86, idProduct=7523
[8666695.207328] usb 3-1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[8666695.207331] usb 3-1: Product: USB Serial
[8666695.209298] ch341 3-1:1.0: ch341-uart converter detected
[8666695.219438] usb 3-1: ch341-uart converter now attached to ttyUSB0
```

2. Now trying erase only: `./cc2538-bsl.py -p /dev/ttyUSB0 -evw blink.bin` 
```
Opening port /dev/ttyUSB0, baud 500000
Reading data from blink.bin
Cannot auto-detect firmware filetype: Assuming .bin
Connecting to target...
ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
```

3. Of course I ame pushing. I had successfull flashing yesterday. Everytime holding button and after plug still holding about 5 sec.

omerk, 2020-09-28:
@krencl Did you flash, accidentally or otherwise, any firmware other than blink.bin or CC26X2R1_20200805.hex on your stick?

Given that you were able to flash this yesterday and now you can't, just curious if perhaps you disabled BSL somehow.

krencl, 2020-09-28:
@omerk Yes, I also tried really early version of firmware [CC2652R_20190426.zip](https://github.com/Koenkk/Z-Stack-firmware/blob/d70c4f14ad3f70b04757c04cd4ceee8d418bdc22/coordinator/Z-Stack_3.x.0/bin/CC2652R_20190426.zip)

I tried now script what you sent yesterday `./znp-uart-test.py` result is 
```
Got 7 bytes in response to PING command: b'\xfe\x02a\x01Y\x06='
PASS|OK
```

omerk, 2020-09-28:
@krencl As long as it was firmware from @Koenkk's repo specifically for CC2652R then rogue firmware disabling BSL is not a concern, so that's good.

The result of that script means that you were able to flash the ZNP coordinator firmware on your stick and it is indeed running there.

One more test, now that we've established that you have the ZNP coordinator firmware running: 

  - Unplug the stick
  - Press and hold BSL button, plug it back in
  - Wait a few seconds and release the BSL button
  - Run the `znp-uart-test.py` script again

If the result is the same as above (`PASS|OK`) then that means the stick is not entering the BSL mode. If the stick is entering the BSL mode as expected, that means the ZNP coordinator firmware is not running and the script should fail.

krencl, 2020-09-28:
@omerk I tried again (multiple times), same result - `znp-uart-test.py` ourput is `PASS|OK`. Can't enter BSL mode.

omerk, 2020-09-28:
@krencl OK, so you were able to use BSL before (as you were able to flash both blink and ZNP firmware on your board) and it somehow stopped working even though you are following the same steps as when it actually worked. 

Are you using the same computer? Can you replicate this on another machine?

If you can not get this working on another computer, then the only thing I can think about is that somehow the BSL button got damaged in between the "working" and "not working" points. If you want to verify this and have a multimeter handy, you can put your meter in [continuity testing mode](https://www.fluke.com/en-gb/learn/best-practices/test-tools-basics/digital-multimeters/how-to-test-for-continuity-with-a-digital-multimeter) and probe points A and B below:

[image]

Normally (when the button is not pressed down), it should read open (not connected) on your meter. When the button is pressed, you should see continuity (and likely hear a beep).

If you want to take this one step further, in "normal" (not BSL) mode, you should see 3.3V on point A when the stick is plugged in (you can use the usb connector shield as GND). When the button is pressed, point A should read 0V.

If electrical checks pass, then unfortunately we are back at looking at this as a software issue.

stigfaerch, 2020-09-28:
> 1. Now trying erase only and default firmware: `./cc2538-bsl.py -p /dev/ttyUSB0 -evw blink.bin`
> 
> ```
> Opening port /dev/ttyUSB0, baud 500000
> Reading data from blink.bin
> Cannot auto-detect firmware filetype: Assuming .bin
> Connecting to target...
> ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
> ```
> 
> 1. Of course I ame pushing. I had successfull flashing yesterday. Everytime holding button and after plug still holding about 5 sec.

I also had a problem with the same output as yours. For me the problem was that I apparently did not push the button hard enough. I was not aware that when in bootloader mode, the firmware would be active. So when I pressed it harder, the LED didn't blink anymore.

omerk, 2020-09-28:
> I was not aware that when in bootloader mode, the firmware would be active.

In bootloader mode, firmware is **not** active, hence the blinking stopping.

omerk, 2020-09-28:
> My next problem though, is that I cannot get z2m working in HA:
> Zigbee2MQTT:error 2020-09-28 18:28:14: Exiting...
> Zigbee2MQTT:error 2020-09-28 18:28:14: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyUSB0'

@stigfaerch If you are seeing /dev/ttyUSB0 as a valid serial port, this might be a permissions issue. Does the user account running z2m have access to the serial ports? One way to check for this would be to switch to that account and see if you can open the serial port using minicom or any other terminal emulator.

stigfaerch, 2020-09-28:
> > I was not aware that when in bootloader mode, the firmware would be active.
> 
> In bootloader mode, firmware is **not** active, hence the blinking stopping.

Sorry. That was exactly what i meant. Wrote the opposite :-D

stigfaerch, 2020-09-28:
> @stigfaerch If you are seeing /dev/ttyUSB0 as a valid serial port, this might be a permissions issue. Does the user account running z2m have access to the serial ports? One way to check for this [...]

Hi Omer

I think I might got it working now. At least there is a connection.
Before I was using a cc2531.
My zigbee devices do not show up in HA as devices. In the mqtt log, I first got at lot of messages like:
Zigbee2MQTT:info  2020-09-28 21:34:24: MQTT publish: topic 'homeassistant/light/0xf0d1b8000010a036/light/config', payload 'null'
I reset one of the Osram bulbs, and succeeded with a pairing.
Do I really need to repair all of my zigbee devices?

omerk, 2020-09-28:
@stigfaerch I am not sure what the recommended path for migration is going from your CC2531 stick to zzh (CC2652). FWIW, I recommend starting fresh (this is what I've done when I moved from CC2531 to the first zzh prototype in January) although if you have many devices I can see this being a bit of a pain. I'll let someone who might have done this before or @Koenkk comment.

Koenkk, 2020-09-29:
@stigfaerch repairing is necessary indeed (https://www.zigbee2mqtt.io/information/FAQ.html#requires-repairing)

stigfaerch, 2020-09-29:
@Koenkk 
Ok, I can live with that. I was only half ways setting things up anyway.
When will there be a new zigbee2mqtt release? I'm eagerly waiting as the next release will include a couple of devices I have.

Koenkk, 2020-09-29:
thursday

Sandriekus, 2020-09-29:
> thursday

Offtopic: is the IKEA OTA also fixed in the new release?

Koenkk, 2020-09-29:
@Sandriekus should also be fixed (if you want to be sure it's fixed in your case switch to the latest dev branch, that's 99% equal to what will be released on thursday)

krencl, 2020-09-29:
@omerk Yes, it is as you wrote. I flashed only Koenkk's firmware and blink.

I am using two computers. One with Ubuntu, second with Windows. Flashing was ok on both machines on different USB inputs. Now can't flash on both compusters.

HW button is OK, all tests with multimeter passed.

tpaulus, 2020-10-05:
A new stick arrived in the mail today thanks to @omerk, and Z2M is happy.
Thanks for the help everyone!

closed 2020-10-05 by tpaulus

karis79, 2020-11-12:
Can i still comment on this even it is closed? Let me know if i need to open new item or if im missing something?

I think I have similar issue than tpaulus had. I received zzh few weeks ago and now I have tried to get it working but I cannot get it running. 

Im trying it with rpi4 + homeassistant. I have edge version of addon (stable says no support for armv7?). I have attached zzh directly to pi4 or usb3 hub which have own power supply. 
- I tried znp-uart-test.py and i got PASS|OK
- I have tried to press button just after rebooting system
- I can flash (with ubuntu) it normally and when flashed blink, it started blinking.
- I have now flashed CC26X2R1_20201026.hex
- i see this device /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 and i used this or /dev/ttyUSB0 in config
- My system says "Unknown options zigbee_herdsman_debug" so i cannot show you debug log but i can see this:
"Zigbee2MQTT:error 2020-11-12 15:34:06: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)"

Tinus050, 2021-05-16:
I had the same problem, the solution in my case was to check the configuration.yaml for another reference to another port.
Before I installed the Zig-zig I had my DSMR (energy meter) configured in the configuration.yaml like this:

sensor:
  - platform: dsmr
    port: /dev/ttyUSB0

But after connecting the zig-a-zig, the zig-a-zig used port: /dev/ttyUSB0 and the DSRM used port: /dev/ttyUSB1
So please also check your configuration.yaml and also the port setting in: supervisor --> zigbee2mqtt --> configuration

miguelarios, 2021-07-08:
I believe I have the same issue that @tpaulus has experienced. I got my zzh and flashed it with my Mac per the instruction by entering bsl (i.e. holding the button down when inserting the usb). The script for pass fail shows `PASS|OK`. I am not sure what else to do #7923 here is the issue I have open.

sygys, 2021-08-07:
Hi there i was directed here because i have a light Hue fugato spot that is not recognized by z2mqtt. I search the supported list and saw all fugato spots are added except the single light version. Is there a possibility this could be added? https://www.coolshop.nl/product/philips-hue-fugato-single-spot-black-white-color-ambiance/AE35PC/

iEEE address is 0x00178801068dedbb (0x0FF0)
Model no. 5063131P7

If more info is needed please let me know