Koenkk/zigbee2mqtt #4441: Error Starting ZigBee2MQTT

joemadd3n, 2020-09-23:
Error from Log:
https://pastebin.com/vsLwwqyA

Running on:
Ubuntu 20.04.1 LTS
Docker version 19.03.11

## Debug info
Zigbee2MQTT version: **1.14.4.2** 
Adapter hardware: **CC2531**
Adapter firmware version: **Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}'**

PedroLamas, 2020-09-23:
How did you set the MQTT server address in the configuration?

Please ensure you prefix the IP address with "mqtt://" as indicated in the configuration options here: https://www.zigbee2mqtt.io/information/configuration.html#configuration

joemadd3n, 2020-09-23:
> How did you set the MQTT server address in the configuration?
> 
> Please ensure you prefix the IP address with "mqtt://" as indicated in the configuration options here: https://www.zigbee2mqtt.io/information/configuration.html#configuration

Yes! This worked, my fault entirely.
Thank you.

closed 2020-09-23 by joemadd3n